realsenseai/realsense-ros #1: Support for F200 - SR300

severin-lemaignan, 2016-03-01:
Great to see official ROS support for the RealSense cameras.

Can we expect soon support for the F200 and SR300 as well?

mkhansen-intel, 2016-03-01:
Yes! Our first release was really only for R200, but we'll add support for F200 and SR300 soon. We'll post an update to the issue when we do.

mkhansen-intel, 2016-03-01:
We currently are only supporting the R200 camera, we'll update the launch files and documentation when the other cameras are supported.

closed 2016-03-01 by mkhansen-intel

severin-lemaignan, 2016-03-02:
Thanks for the feedback.

You still may want to clarify the documentation and leave this issue open (possibly renaming it 'Support for F200-SR300) so that other users know what to expect and can easily monitor the progress.