realsenseai/realsense-ros #3: Nodelet crashes when trying to display point cloud

severin-lemaignan, 2016-03-01:
Happens with a SR300 on Ubuntu 14.04 (with a patched kernel 4.4)

Steps to reproduce:
- start `roscore` and `rviz`
- launch `realsense_camera realsense_r200_nodelet_standalone_preset.launch`
- in rviz, add a ` PointCloud2` visualization and set the topic to `camera/depth/points`
- the nodelet crashes.

Log:
``` 
$ roslaunch realsense_camera realsense_r200_nodelet_standalone_preset.launch 
... logging to /home/lab/.ros/log/b4e40ce0-dfb7-11e5-8a15-00dbdf1f9ec4/roslaunch-PORTEGE-R830-4670.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://PORTEGE-R830:38149/

SUMMARY
========

PARAMETERS
 * /rosdistro: jade
 * /rosversion: 1.11.16

NODES
  /
    RealsenseNodelet (nodelet/nodelet)
    standalone_nodelet (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[standalone_nodelet-1]: started with pid [4688]
process[RealsenseNodelet-2]: started with pid [4689]
[ INFO] [1456841667.896780624]: Initializing nodelet with 4 worker threads.
[ INFO] [1456841668.391947827]: RealSense Camera - Starting camera nodelet.
[ INFO] [1456841669.237153943]: RealSense Camera - Number of cameras connected: 1
[ INFO] [1456841669.237346757]: RealSense Camera - Firmware version: 3.10.10.0
[ INFO] [1456841669.237507359]: RealSense Camera - Name: Intel RealSense SR300
[ INFO] [1456841669.237656812]: RealSense Camera - Serial no: 54FFFFFF9205001823
[ INFO] [1456841669.237785441]: RealSense Camera - Enabling Depth Stream: preset mode
[ INFO] [1456841669.238289684]: RealSense Camera - Enabling Color Stream: preset mode
[ INFO] [1456841669.275615697]: RealSense Camera - Publishing camera transforms
[standalone_nodelet-1] process has died [pid 4688, exit code -11, cmd /opt/ros/jade/lib/nodelet/nodelet manager __name:=standalone_nodelet __log:=/home/lab/.ros/log/b4e40ce0-dfb7-11e5-8a15-00dbdf1f9ec4/standalone_nodelet-1.log].
log file: /home/lab/.ros/log/b4e40ce0-dfb7-11e5-8a15-00dbdf1f9ec4/standalone_nodelet-1*.log
[RealsenseNodelet-2] process has finished cleanly
log file: /home/lab/.ros/log/b4e40ce0-dfb7-11e5-8a15-00dbdf1f9ec4/RealsenseNodelet-2*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

The nodelets logs do not contain anything useful.

severin-lemaignan, 2016-03-01:
Same issue with F200

mkhansen-intel, 2016-03-01:
Thanks for the input, this release was really for R200 only, but we plan to add support for F200 & SR300 in another release shortly. Is this working OK for R200? In any case, we will reproduce and fix this bug. Thank you for bringing it to our attention, we'll post an update to the issue when we've got a fix.

severin-lemaignan, 2016-03-01:
@mkhansen-intel Thanks for the feedback. I really need to work with the SR300/F200 though since my application requires close human-robot interaction.

I've done a bit of `printf`-based debugging (nodelet are painful to debug!!), and on my system, the nodelet crashes between [L731](https://github.com/intel-ros/realsense/blob/indigo-devel/camera/src/realsense_camera_nodelet.cpp#L731) and [L745](https://github.com/intel-ros/realsense/blob/indigo-devel/camera/src/realsense_camera_nodelet.cpp#L745) **when `x=128` and `y=270`**

mkhansen-intel, 2016-03-01:
We've captured your requirement for the F200 & SR300, and will add support for those cameras in another release, until then I'm closing this issue. Thank you for your patience.

closed 2016-03-02 by mkhansen-intel