realsenseai/realsense-ros #5: Install issue/question: missing symlink

2Reeko, 2016-03-06:
Hi,

I come across this error while catkin making.

```
make[2]: *** No rule to make target `/usr/local/lib/librealsense.so', needed by `/home/user/catkin_ws/devel/lib/librealsense_camera_nodelet.so'.  Stop.
make[1]: *** [ros/realsense/camera/CMakeFiles/realsense_camera_nodelet.dir/all] Error 2
```

I added  "/usr/local/lib" to my "LD_LIBRARY_PATH". However, there isn't a librealsense.so file in "/usr/local/lib". 

Can I solve this by putting a symlink to the librealsense folder in usr/local/lib ? Is there a better way to fix this?

Thanks,

Christian

sthoduka, 2016-03-06:
When you run `sudo make install` on [librealsense](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md), it should copy the .so file to /usr/local/lib.

2Reeko, 2016-03-07:
Thanks, I must have only run the make.

closed 2016-03-07 by 2Reeko