realsenseai/realsense-ros #7: Multiple cameras question

prarobo, 2016-03-08:
I have some questions on using the ros wrapper for multiple cameras. Currently I see in your ros code a rs_context object that contains all camera device information. Your source calls the first realsense device (index 0) and pushes the data from the device into ros messages.

It will be convenient to spawn multiple nodelets where each nodelet brings up a single realsense camera. Currently if I try that, the nodelets crash. I believe this is due a single rs_context object controlling all cameras and when a new rs_context object is created (from a different nodelet), it causes some low level permissions issue. Please correct me if I am wrong.

One way I can think of getting around this problem is to just use a single nodelet to get all datastreams from all realsenses using a single rs_context object.  Is there a more elegant way to solve this problem?

I would prefer a solution where each realsense camera can be brought up from a separate nodelet. Please let me know if you want further information or clarifications.

reaganlo, 2016-03-10:
As a user, which is the preferred use case:
1) Stream data from a realsense camera by giving the unique "serial number" of the camera as input (like in the launch file)
OR 
2) The ability for the nodelet to traverse all the connected realsense cameras, fetch their unique serial numbers and stream data from them.

prarobo, 2016-03-10:
I would go with 1.

In my use case, I have multiple realsenses. I would like to have the device IDs of each realsense connected to be repeatable over system reboots. I feel connecting using a unique serial number would help.

liz-murphy, 2016-03-10:
I would also like (1). I've tried to implement this functionality in the PR above, but it is flaky.  As @prarobo said, opening multiple contexts in different processes does not seem to work well.

reaganlo, 2016-03-24:
We are currently working on implementing this feature. The first release with respect to this would be to uniquely identify a camera. Please refer to #18

reaganlo, 2016-03-31:
@liz-murphy @prarobo If you are interested in the progress of this issue, please refer to the closing comments of issue #18

reaganlo, 2016-04-26:
Fixed in release [1.0.3](https://github.com/intel-ros/realsense/releases/tag/1.0.3)

closed 2016-04-26 by reaganlo