realsenseai/realsense-ros #9: Consolidate publish_tf and enable_tf in launch/includes/realsense_r200_nodelet.launch.xml

liz-murphy, 2016-03-10:
This should just be a single variable "publish_tf", right now the behaviour is backwards from what is expected and the inclusion of "enable_tf" is confusing.

I have been using the launch file realsense_r200_rgbd.launch with publish_tf set to true from existing OpenNI compatible launch files and wondering why no tfs are being published from the nodelet.

reaganlo, 2016-03-10:
@liz-murphy , I understand why you could get confused. But let me try to explain the reason why we have two flags "publish_tf" and "enable_tf":

This was done to follow the standards in [freenect.launch](https://github.com/ros-drivers/freenect_stack/blob/master/freenect_launch/launch/freenect.launch), where "publish_tf" is set to true when the transforms are generated **statically** using "**static_transform_publisher**". 

By **default**, the realsense package generates transforms from within the nodelet. 

So when the "publish_tf" is set to true, we wanted to turn-off the default behavior of the realsense package and not generate transforms from within the nodelet, as the expectation/assumption was that the transforms will be generated statically. We have a comment in the realsense_r200_nodelet.launch.xml that states that.     
_"enable_tf" will be set to false only if "publish_tf" is true. Else it is set to true.
         Ensures that the nodelet does not publish tf if the transforms are published statically._

Let me know if this helps.

reaganlo, 2016-03-10:
@liz-murphy, Also, since you are using realsense_r200_nodelet.launch.xml, I would like to mention that you could change the resolution of the depth and color streams in **"manual" mode**. Please refer to the sample launch file [manual launch](https://github.com/intel-ros/realsense/blob/indigo-devel/camera/launch/realsense_r200_nodelet_standalone_manual.launch).

We will update the realsense_r200_nodelet.launch.xml in the next release to show this option, in case users overlook the manual.launch example.

liz-murphy, 2016-03-10:
@reaganlo OK, my bad, I'm sorry. You are not providing a default static transform file though, so it is still odd to set publish_tf=true and have nothing published.
https://github.com/ros-drivers/freenect_stack/blob/edaf577f6db318ca12f2ac767619b2c6a971cff2/freenect_launch/launch/freenect.launch#L129

With respect to resolutions etc - you should be using a combination of dynamic reconfigure and/or ros params to load the configuration. It's just a more standard way of doing things and allows more transparency in remapping. It also means you can get rid of all those constants here https://github.com/intel-ros/realsense/blob/indigo-devel/camera/src/realsense_camera_nodelet.h#L78

I've made a start on that in my fork, needed it for compatability with our existing launch files on the robot. 
https://github.com/liz-murphy/realsense/blob/dev/camera/src/realsense_camera_nodelet.h
https://github.com/liz-murphy/realsense/blob/dev/camera/src/realsense_camera_nodelet.cpp

Note that it is also a lot more convenient to set the resolution by enumerating the device modes in the dynamic reconfigure configuration and fetching the width/height etc in the code via

dev->get_stream_mode(strm, k, width, height, format, framerate);

which stops the user from setting an invalid combination.

liz-murphy, 2016-03-10:
@reaganlo if it helps with development I can send you the launch file we currently use to launch 2 cameras. Let me know if that'd be useful.

reaganlo, 2016-03-10:
@liz-murphy, Yes, your suggestions are all valid. We are working on converting args to params and will have it part of the next release (next week). I was about to do a quick test on your pull request #8 .
I will post my comments on your pull request in that thread.

Yes, it would be great if you could send the launch files that you are using, so that we can understand the requirement better. 

I will review this issue #9 with my team in the morning and decide how to resolve it. Meanwhile, if you feel it could be "closed" based on our discussion (i.e let publish_tf be set to false which would let the nodelet publish the transforms) , please feel free to close it :)

aditya2592, 2019-06-26:
Hi, Does the publish_tf flag exist in the current version ?

doronhi, 2019-06-27:
No. The static transformations are published.

aditya2592, 2019-06-27:
Is there any way to turn the static publishing off? In our case the camera is already in the robot URDF model and transforms are present there

doronhi, 2019-06-27:
We could add a flag in future versions, if it is a common demand. In the meanwhile, could you rename and ignore them?

aditya2592, 2019-06-27:
Are you referring to using tf_prefix? If I use that it also renames the frame_ids of images and point clouds and then the transform between image frame and other frames is lost as it is not present in the tf tree

doronhi, 2019-06-30:
I see now what you mean.
It could take me a while to set a flag to disable publishing tf as there are more issues I'd like to address in the process. In the meanwhile, I think that if you disabled [this line](https://github.com/IntelRealSense/realsense-ros/blob/d3867401225ab819fdae1efdb57909f294e99e3e/realsense2_camera/src/base_realsense_node.cpp#L176) you'll get what you need.

aditya2592, 2019-06-30:
Thanks @doronhi. I will go ahead and comment this line for now.

doronhi, 2019-07-10:
fixed in #843

RealSenseCustomerSupport, 2019-11-07:
----------------------------------------------

@liz-murphy  Any other questions for this ticket? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-11-15:
----------------------------------------------

@liz-murphy  Will close this if no other questions. Thanks!

closed 2019-11-15 by RealSenseCustomerSupport