realsenseai/realsense-ros #10: roslaunch realsense_camera realsense_r200_rgbd.launch (Crash)

vrubies, 2016-03-11:
Hi,

I followed the "Installation Guide" step by step and I can even use qtcreator to run cpp-capture. I see the images just fine. I then proceeded to follow the instructions in the README file within the camera directory. I set my LD_PATH... to usr/local/include etc. As far as I can tell I've done everything correctly.

After calling catkin_make and successfully building the package, I run the instruction roslaunch in the title of this issue, and this is what I get:

[image]

What can I do?

reaganlo, 2016-04-11:
Are you still facing this issue? If yes, could you please provide the following information:

### System Configuration
Please complete Your Configuration detail below.

| Version          | Best Known           | Your Configuration   |
|:---------------- |:---------------------|:---------------------|
| Operating System | Ubuntu 14.04.4 LTS   | ? ? ?                |
| Kernel           | 4.4.0-040400-generic | ?-?-?                |
| ROS              | indigo               | ?                    |
| librealsense     | 0.9.1                | ?.?.?                |
| R200 Firmware    | 1.0.72.06            | ?.?.?.?              |

---
#### How to collect Configuration Data
*This section can be deleted before submission.*

| Version          | Method |
|:---------------- |:------------ |
| Operating System | `grep DISTRIB_DESCRIPTION /etc/*elease*` |
| Kernel           | `uname -r` |
| ROS              | `rosversion -d` |
| librealsense     | `cat <path to librealsense>/librealsense/readme.md | grep release-image | awk -F- '{print $3}'` |
| R200 Firmware    | View the ROS log from running nodelet **OR** `<path to librealsense>/librealsense/bin/cpp-enumerate-devices | grep -i firmware` |

mdhorn, 2016-07-29:
Closing due to now response for > 4 weeks.

closed 2016-07-29 by mdhorn