realsenseai/realsense-ros #12: Black IR view for R200

ibaranov-cp, 2016-03-16:


ddiakopoulos, 2016-03-16:
Can you double check one of the librealsense examples? RGB only and no IR or depth indicates that the UVC driver patching process failed. It would show up in dmesg as an "unknown video format" when the system tries to use the camera.

ibaranov-cp, 2016-03-17:
That is certainly possible.

For my first attempt, I have been trying to make a installer script to ease deployment to ROS robots (with our standard installer setup, please ignore the terrible hard-linked file paths)

I will probably have to put in some more effort to automate the "y" acceptance of some of the sub scripts.

Any comments on possible mis-steps?

`
sudo apt-get update
sudo apt-get upgrade
sudo apt-get install libusb-1.0-0-dev
sudo apt-get install libssl-dev

mkdir /home/administrator/catkin_ws
mkdir /home/administrator/catkin_ws/src
cd /home/administrator/catkin_ws/src
git clone https://github.com/IntelRealSense/librealsense.git

cd librealsense

./scripts/install_glfw3.sh
./scripts/install_dependencies-4.4.sh
./scripts/patch-uvcvideo-4.4.sh v4.4-wily

sudo modprobe uvcvideo

sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/
sudo udevadm control --reload-rules && udevadm trigger

make && sudo make install

sudo cp -r /include/librealsense /usr/local/include

cd ..
git clone https://github.com/intel-ros/realsense.git
cd ..
catkin_make
`

ddiakopoulos, 2016-03-17:
That looks good, but with the behavior you describe, one of the sub-steps of `./scripts/patch-uvcvideo-4.4.sh v4.4-wily` isn't working on the system. Sometimes when this issue happens simply running the script again fixes it.

ibaranov-cp, 2016-03-17:
To note:

```
[13108.297644] usb 2-1: new SuperSpeed USB device number 3 using xhci_hcd
[13108.315558] usb 2-1: New USB device found, idVendor=8086, idProduct=0a80
[13108.315570] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[13108.315579] usb 2-1: Product: Intel RealSense 3D Camera R200
[13108.315586] usb 2-1: Manufacturer: Intel Corp
[13108.315593] usb 2-1: SerialNumber: SN_2211014466
[13108.317899] uvcvideo: Unknown video format 49323159-0000-0010-8000-00aa00389b71
[13108.317913] uvcvideo: Unknown video format 20493859-0000-0010-8000-00aa00389b71
[13108.317931] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[13108.322092] uvcvideo: Unknown video format 2036315a-0000-0010-8000-00aa00389b71
[13108.322111] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[13108.323219] uvcvideo: Unable to create debugfs 2-3 directory.
[13108.325024] uvcvideo: Unknown video format 30315752-0000-0010-8000-00aa00389b71
[13108.325043] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[13108.326218] uvcvideo: Unable to create debugfs 2-3 directory.
[13108.326851] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.4/input/input14

```

ddiakopoulos, 2016-03-17:
@ibaranov-cp yup, the unknown video formats only show up when the uvcvideo.ko module hasn't been patched fully by that script.

ibaranov-cp, 2016-03-17:
Damn, ok will look into....

archielee, 2016-03-17:
Hi, I've been taking over for @ibaranov-cp and I ran the uvc patch script by itself. It seems to succeed up until:
`cp  $CONFIG_LOCATION/.config .`

The error message is:
`cp: will not overwrite just-created './.config' with '/usr/src/linux-headers-4.4.4-040404-generic/.config'`
After this, the rest of the script doesn't seem to execute because I don't see any output afterwards from the logs. The modules never seem to get compiled either. Is this something on our end? Should I manually execute the rest of the steps?

archielee, 2016-03-17:
Manually executed the copying step then ran the rest of the sections and it now works. No idea why the script failed to copy

ddiakopoulos, 2016-03-17:
@archielee your guess is as good as mine! First time I've ever seen that issue.

closed 2016-04-20 by mdhorn

mgserafi, 2017-01-03:
@archielee , how did you manually copy the files and where did you copy them to?

mgserafi, 2017-01-03:
With some print outs, I found that `CONFIG_LOCATION` was set to multiple paths with a `.config` file. After manually setting `CONFIG_LOCATION`, the script works.