realsenseai/realsense-ros #15: F200 support?

gregormelhorn, 2016-03-18:
Hello,

I am really new to ROS and 3D cameras, so forgive me if the issue is easy to solve. 

I managed to install everything according to the docs, but when running one of the launch files, there is an error about infrared2 stream:

`process[RealsenseNodelet-2]: started with pid [9442]
[ INFO] [1458304316.516457675]: Initializing nodelet with 8 worker threads.
[ INFO] [1458304316.584243499]: RealSense Camera - Starting camera nodelet.
[ INFO] [1458304317.185167616]: RealSense Camera - Number of cameras connected: 1
[ INFO] [1458304317.185233790]: RealSense Camera - Firmware version: 2.60.0.0
[ INFO] [1458304317.185246538]: RealSense Camera - Name: Intel RealSense F200
[ INFO] [1458304317.185260261]: RealSense Camera - Serial no: 015150131405
[ INFO] [1458304317.185271646]: RealSense Camera - Enabling Color Stream: preset mode
[ INFO] [1458304317.185402923]: RealSense Camera - Enabling Depth Stream: preset mode
[ INFO] [1458304317.185522817]: RealSense Camera - Enabling Infrared Stream: preset mode
[ INFO] [1458304317.185639040]: RealSense Camera - Enabling Infrared2 Stream: preset mode
[ERROR] [1458304317.185765818]: RealSense Camera - Error calling rs_get_stream_intrinsics ( device:0x1c22cb0, stream:INFRARED2, intrin:0x7ffc92766080 ): 
stream not enabled: INFRARED2 

[FATAL] [1458304317.301574020]: Service call failed!
`

I suppose this is related to differences between the f200 and r200 cameras? What do you suggest to do?

Thanks, 
Gregor

gregormelhorn, 2016-03-18:
it seems to work when commenting out everything that tries to access the infrared2 stream...

rjingar, 2016-03-18:
Current release supports R200 camera only. The error you are getting is because F200 has only one Infrared camera. 
Also, even after commenting out everything related to infrared2 stream, all other functionality and supported camera settings are for R200 camera only.
Future releases will have support added for F200 and SR300.

gregormelhorn, 2016-03-18:
I see your point. Anyway, the camera does work somehow with minor changes, so I will try to work with that and wait for a new release. Should I feel confident enough about quality, I will also provide my changes to you...

rjingar, 2016-03-21:
Great that it worked for you. Though changes enable F200 camera, testing or enabling has not been done for it. Stability or quality can be guaranteed once it is tested and released officially.

linknum23, 2016-03-28:
What is the time frame on this? I am wondering if I should take the time to get this working on older projects not supported by intel for the time being...

linknum23, 2016-03-28:
for reference there is a fork that added support for the SR300 and F200 here: https://github.com/nlyubova/realsense

NCharabaruk, 2016-04-06:
Any updates on when functionality for the F200 will be complete?

atenpas, 2016-04-08:
I'm also interested in functionality for the SR300

linknum23, 2016-04-08:
The fork I linked to works fine for the sr300.

atenpas, 2016-04-08:
I see, thanks. Sorry, I did not read all the comments in here.

severin-lemaignan, 2016-04-11:
For reference, I've rebased nlyubova patch on the current Intel head: https://github.com/severin-lemaignan/realsense
Successfully tested with the SR300.

mdhorn, 2016-07-29:
F200 supported added in release 1.3.0

closed 2016-07-29 by mdhorn