realsenseai/realsense-ros #18: Connect to camera using unique id

reaganlo, 2016-03-24:
We are currently working on this functionality. 
The user will be able to specify a unique id in the launch file using the parameter "serial_no".

More details regarding this will be provided as the development progresses.

reaganlo, 2016-03-31:
This issue has been fixed in Release 1.0.2

# Details
**Note**: The term "camera" here refers to Intel RealSense R200 cameras. It has been tested only for Linux Kernel version 4.4.0-040400-generic, librealsense version: 0.9 and R200 Firmware version: 1.0.72.06

- The nodelet will exit with error if no cameras are detected by the system.
- The nodelet will print the serial_no of all the cameras detected by the system.
- The nodelet will connect to a camera based on the serial_no parameter in the launch file.
- If only one camera is detected by the system, the nodelet will connect to it even if the serial_no is not specified in the launch file.
- The nodelet will exit with error if multiple cameras are detected by the system but no serial_no is specified in the launch file.
- The nodelet will exit with error if an incorrect serial_no is specified in the launch file.

**Note**: Launching multiple cameras simultaneously using serial_no has not be **fully** tested and **must not** be considered as part of this issue. It will be tested and confirmed in our future release. Although you may try by creating launch files similar to [realsense_r200_rgbd.launch](https://github.com/intel-ros/realsense/blob/indigo-devel/camera/launch/realsense_r200_rgbd.launch) and giving a unique value for the "camera" namespace and "serial_no" parameter for each camera. E.g camera = "camera1" and serial_no = "2131007626".

closed 2016-03-31 by reaganlo

mirzashah, 2016-04-01:
@reaganlo just tested out the latest release, configuring by serial number works. Even though you said it's untested, it appears two cameras work simultaneously using the method you suggested . Great job! :)

I did notice though that when running two simultaneously, both cameras retain the parent link of "camera_link". e.g. If you set camera:=camera1, the rgb and depth links will be camera1_rgb_frame and camera2_rgb_frame, but the parent link will be camera_link instead of camera1_link.