realsenseai/realsense-ros #20: Calibration Question

NCharabaruk, 2016-03-31:
Hello,

I am wondering if there are any plans to add a node to perform joint calibration of the RGB and Depth cameras in order to increase the accuracy and allow for the depth images to be registered in the RGB frame.

Thanks

PercATI, 2016-04-06:
Hi @NCharabaruk  ,

The current ROS node do publish the TF transform between the two cameras. You can also find that transform in the camera_info message.
The calibration data is burned in the cameras themselves at the factory. We simply publish it in the ROS standards.

you can try running this node to see the registration:

	<node pkg="nodelet" type="nodelet" name="depth_register"
		args="load depth_image_proc/register standalone_nodelet ">

		<remap from="depth/camera_info" to="/camera/depth/camera_info"/>
		<remap from="depth/image_rect" to="/camera/depth/image_raw"/>
		<remap from="rgb/camera_info" to="/camera/color/camera_info"/>

	</node>
 from here: http://wiki.ros.org/depth_image_proc 

Hope that helps.

Amit

reaganlo, 2016-04-11:
@NCharabaruk Does this answer your question?

NCharabaruk, 2016-04-11:
@reaganlo Somewhat. I'm using an F200 which is currently unsupported by this package, but is supported by another one that doesn't pull out the factory calibration parameters. In fact, I didn't even know that they were in there. I was able to find them by modifying this package to run on the F200, only to pull out the calibration params though.

closed 2016-04-11 by NCharabaruk

Isabelp, 2017-03-16:
Hi,
I am wondering if there is a way to calibrate the F200. I have tried with the same process used to calibrate the Kinect (http://wiki.ros.org/openni_launch/Tutorials/IntrinsicCalibration) but the service set_camera_info doesn't exist and the window to calibrate the ir sensor doesn't work at all.

Thanks

mdhorn, 2017-03-16:
Calibration of the RealSense camera is only supported via the Windows RealSense SDK.
https://software.intel.com/en-us/intel-realsense-sdk/

Isabelp, 2017-03-17:
Hi, 

thanks for the fast answer. 

In the sdk I have found the camera calibration for better tracking accuracy in the object tracking folder, which I could use directly from the Metaio toolbox (https://software.intel.com/sites/landingpage/realsense/camera-sdk/v1.1/documentation/html/doc_ot_the_metaio_toolbox.html). I have also found a way to set the depth values depending on the application (https://software.intel.com/en-us/articles/intel-realsense-sdk-code-samples ). However, I am sorry if I am skipping something very obvious but there is not available a general camera calibration?

Thanks