realsenseai/realsense-ros #22: Start multiple camera nodes from single roslaunch XML

nmertins, 2016-04-07:
### System Configuration

| Version          | Known Good   | Your Configuration |
| ---------------- | ------------:| ------------------:|
| Operating System | ubuntu 14.04 |                ubuntu 14.04 |
| Kernel Version   |     4.4.0-13 |            4.4.0-040400-generic |
| R200 FW          |    1.0.72.06 |            1.0.72.06 |
| librealsense     |        0.9.1 |              0.9.1 |
--
It'd be nice to be able to start up multiple camera nodes from a single .launch file. For example:

```
<launch>
    <include file="$(find realsense_camera)/launch/realsense_r200_rgbd.launch">
        <arg name="camera"      value="rs1" />
        <arg name="serial_no"   value="1234567890" />
    </include>

    <include file="$(find realsense_camera)/launch/realsense_r200_rgbd.launch">
        <arg name="camera"      value="rs2" />
        <arg name="serial_no"   value="0987654321" />
    </include>
</launch>
```
Running the script above results in the following error which brings down the process:
```
[ERROR] [1460047172.302085622]: RealSense Camera - Error calling rs_create_context ( api_version:4 ): 
UVCIOC_CTRL_QUERY:UVC_SET_CUR error 5, Input/output error
```

mdhorn, 2016-04-11:
Official support for Multiple cameras is being release soon.
Launching multiple camera from a single launch file wasn't in our original acceptable criteria, but we've capture this and it will get addressed -- may not be in the first release.

prarobo, 2016-04-26:
@mdhorn Do you have a timeline for this release?

mdhorn, 2016-04-28:
We don't have a hard date for this fix. It is high on the list, but there are a couple of other items that are in front of it.

The issue stems from librealsense assuming all cameras would be started from the same process/thread whereas our ROS module launches each as a separate process via a nodelet.

reaganlo, 2016-05-26:
@nmertins @prarobo This required some changes at the librealsense level, which has been implemented in their latest "master" branch https://github.com/IntelRealSense/librealsense/commit/bff3ab59972e2a7a16031d0970cef4d29471757d

We are waiting for librealsense to make a "release", to update our ROS repo with the related launch file changes. In the mean time, you could do the following:

- Download the latest librealsense "master" branch as mentioned above.
- Copy the file "realsense_r200_multiple_cameras.launch.xml" in the attachment into the "**launch/includes**" folder.
- Copy the file "realsense_r200_multiple_cameras.launch" in the attachment into the "**launch**" folder. Make sure to read the comments in the file and make the changes accordingly. Then launch the file.
 
[multiple_cameras.zip](https://github.com/intel-ros/realsense/files/283902/multiple_cameras.zip)

Thank you for your patience and please let us know if this works.

closed 2016-06-07 by mdhorn