realsenseai/realsense-ros #25: xyzrgb point cloud freezes in rviz after displaying for 1 second

jgoppert, 2016-04-12:
### System Configuration

| Version          | Known Good   | Your Configuration |
| ---------------- | ------------:| ------------------:|
| Operating System | ubuntu 14.04 |                ubuntu 14.04 |
| Kernel Version   |     4.4.0-13 |             4.4.0-13 |
| R200 FW          |    1.0.72.06 |            1.0.72.06 |
| librealsense     |        0.9.1 |              master  f019a  > 0.9.1 |
| ROS                  |     Indigo  |                 Jade       |

### Expected Behavior

Start rgbd node, should display xyzrgb point cloud. (Point cloud with color from rgb image).

### Actual Behavior

xyzrgb point cloud  ( /camera/depth_registered/points ) displays for 1 second before crashing

xyz point cloud displays fine ( /camera/depth/points )

### Steps to Reproduce

Unplug the realsense usb connection and reconnect it (it won't reliably start /camera/depth_registered/points at all otherwise.

```bash
roslaunch realsense_camera realsense_r200_rgbd.launch 
rviz
```
Add /camera/depth_registered/points as PointCloud2

mdhorn, 2016-04-13:
By crash to you mean rviz stops streaming? Please confirm this is what you are seeing.

If yes, we could reproduce this and it will be investigated further.

jgoppert, 2016-04-13:
Yes, it stops streaming.  If I select the other point cloud topic it starts streaming again without rgb, but won't start rgbxyz point cloud stream again.

ffurrer, 2016-04-13:
We experienced a very similar behavior on a Ubuntu machine with an identical configuration as the one above except on Indigo. Sporadically we were able to get a single (or very few) colored point cloud message(s). In rviz, this lead to the behavior described above.
We also tested it on an OSX 10.11 machine running Jade where we got no colored point cloud at all.

mrshpot, 2016-05-10:
@jgoppert Could you check if PR #38 helps in your case?

mdhorn, 2016-05-10:
@jgoppert Have you tried release 1.0.4?  Did that resolve the issue?
If not, can you please add your system configuration information: CPU, Memory, type of storage (disk/emmc/sdcard...)

jgoppert, 2016-05-19:
@mdhorn @mrshpot fixed for me on 1.0.4

mdhorn, 2016-05-20:
Great!

closed 2016-05-20 by mdhorn

jgoppert, 2016-06-20:
@mdhorn This seems to be back on latest master of realsense-ros and librealsense

```bash

 WARN] [1466386439.555341438]: TF2 exception:
Lookup would require extrapolation into the future.  Requested time 1466386439.554010830 but the latest data is at time 1466386439.553981243, when looking up transform from frame [camera_depth_optical_frame] to frame [camera_rgb_optical_frame]

```

betaupsx86, 2016-10-22:
Experiencing this issue in realsense ros 1.5.0 and librealsense 1.11.0. Depth_registered/points stream freezes after a second. No Warns or Errors from Rviz.

mdhorn, 2016-10-24:
@betaupsx86 
Can you please provide the details of your platform: processor, memory, local storage type (SSD, HD, etc)?

I'm attaching a patch to set the message filter to use exact instead of approximate. Please give this a try on your system.

[0001-Set-the-sync-filter-to-exact.patch.txt](https://github.com/intel-ros/realsense/files/549169/0001-Set-the-sync-filter-to-exact.patch.txt)

betaupsx86, 2016-10-25:
Ubuntu 14.04      4.2.0-27-generic
Intel i5 Quad-core  2.4 GHz   6GB RAM
HDD

There seems to be small periodic frame drops in the IR streams. I actually went back to realsense 1.4 and now I'm experiencing the same issue. With that version everything used to work fine.