realsenseai/realsense-ros #26: Error for realsense_camera realsense_r200_nodelet_standalone_preset.launch

sclin-loauas, 2016-04-12:
System Configuration

| Version          | Known Good   | Your Configuration |
| ---------------- | ------------:| ------------------:|
| Operating System | ubuntu 14.04 |                ubuntu 14.04 |
| Kernel Version   |     4.4.0-13 |            4.4.0-040400-generic |
| R200 FW          |    1.0.72.06 |            1.0.72.06 |
| librealsense     |        0.9.1 |              0.9.0|

I can run cpp-capture and build the package using catkin-make. However when I run 
"roslaunch realsense_camera realsense_r200_nodelet_standalone_preset.launch" , an error occurs.

[ INFO] [1460466020.703162344]: Initializing nodelet with 4 worker threads.
[ INFO] [1460466020.790980800]: RealSense Camera - Starting camera nodelet.
**[ERROR] [1460466021.095609328]: RealSense Camera - Error calling rs_create_context ( api_version:4 ): 
uvc_open2(...) returned Access denied 

[FATAL] [1460466021.232074645]: Service call failed!
[standalone_nodelet-2] process has died [pid 7430, exit code 1, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=standalone_nodelet __log:=/home/aicrobo/.ros/log/82e77eb2-00ae-11e6-a8a2-c01885c34cc3/standalone_nodelet-2.log].
log file: /home/aicrobo/.ros/log/82e77eb2-00ae-11e6-a8a2-c01885c34cc3/standalone_nodelet-2*.log
[RealsenseNodelet-3] process has died [pid 7437, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load realsense_camera/RealsenseNodelet standalone_nodelet mode preset enable_depth true enable_color true enable_pointcloud true enable_tf true __name:=RealsenseNodelet __log:=/home/aicrobo/.ros/log/82e77eb2-00ae-11e6-a8a2-c01885c34cc3/RealsenseNodelet-3.log].
log file: /home/aicrobo/.ros/log/82e77eb2-00ae-11e6-a8a2-c01885c34cc3/RealsenseNodelet-3*.log
**

Do I need to run the command as a super user? I tried to add "sudo" before the command but it can not find the command "roslaunch"
What can I do?

reaganlo, 2016-04-13:
We could not reproduce this error. 
Can you please ensure that you are using the Video4Linux backend? Follow the instructions at https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#video4linux-backend

You could also try installing the librealsense version 0.9.1

mdhorn, 2016-04-19:
Try adding your login account to the group `video`.

`sudo adduser $USER video`

You may need to log out and log back into the system to see the new group.

`groups`

In general, the video devices will be owned by `root` and set to group `video`:

```
$ ls -l /dev/video*
crw-rw----+ 1 root video 81, 3 Apr 12 18:02 /dev/video3
crw-rw----+ 1 root video 81, 4 Apr 12 18:02 /dev/video4
crw-rw----+ 1 root video 81, 5 Apr 12 18:02 /dev/video5
```

mdhorn, 2016-05-10:
@sclin-loauas Were you able to solve this permissions problem?

mdhorn, 2016-06-06:
Closing due to no response from submitter.

closed 2016-06-06 by mdhorn