realsenseai/realsense-ros #27: RealSense with OpenNi

lluissalord, 2016-04-14:
I'm trying to work with OpenNi using the RealSense camera. I found in some place that OpenNi doesn't support RealSense using it directly (I don't know if it's true but I try sometime I didn't figure out how to do it).

The thing is that I'm trying to do it through ROS, I launch `roslaunch realsense_camera realsense_r200_rgbd.launch` and then  `roslaunch openni2_launch openni2.launch`. But it gives me the following error:

```
SUMMARY
========

PARAMETERS
 * /camera/camera_nodelet_manager/num_worker_threads: 4
 * /camera/depth_rectify_depth/interpolation: 0
 * /camera/driver/auto_exposure: True
 * /camera/driver/auto_white_balance: True
 * /camera/driver/color_depth_synchronization: False
 * /camera/driver/depth_camera_info_url: 
 * /camera/driver/depth_frame_id: camera_depth_opti...
 * /camera/driver/depth_registration: False
 * /camera/driver/device_id: #1
 * /camera/driver/rgb_camera_info_url: 
 * /camera/driver/rgb_frame_id: camera_rgb_optica...
 * /rosdistro: indigo
 * /rosversion: 1.11.16

NODES
  /camera/
    camera_nodelet_manager (nodelet/nodelet)
    depth_metric (nodelet/nodelet)
    depth_metric_rect (nodelet/nodelet)
    depth_points (nodelet/nodelet)
    depth_rectify_depth (nodelet/nodelet)
    depth_registered_sw_metric_rect (nodelet/nodelet)
    driver (nodelet/nodelet)
    points_xyzrgb_sw_registered (nodelet/nodelet)
    rectify_color (nodelet/nodelet)
    register_depth_rgb (nodelet/nodelet)
  /
    camera_base_link (tf/static_transform_publisher)
    camera_base_link1 (tf/static_transform_publisher)
    camera_base_link2 (tf/static_transform_publisher)
    camera_base_link3 (tf/static_transform_publisher)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[camera/camera_nodelet_manager-1]: started with pid [27714]
process[camera/driver-2]: started with pid [27717]
process[camera/rectify_color-3]: started with pid [27718]
process[camera/depth_rectify_depth-4]: started with pid [27719]
process[camera/depth_metric_rect-5]: started with pid [27727]
process[camera/depth_metric-6]: started with pid [27742]
process[camera/depth_points-7]: started with pid [27765]
process[camera/register_depth_rgb-8]: started with pid [27782]
process[camera/points_xyzrgb_sw_registered-9]: started with pid [27795]
process[camera/depth_registered_sw_metric_rect-10]: started with pid [27812]
[FATAL] [1460642189.985024526]: Service call failed!
process[camera_base_link-11]: started with pid [27829]
process[camera_base_link1-12]: started with pid [27833]
process[camera_base_link2-13]: started with pid [27837]
process[camera_base_link3-14]: started with pid [27841]
[ INFO] [1460642190.084527595]: Initializing nodelet with 4 worker threads.
[camera/rectify_color-3] process has died [pid 27718, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load image_proc/rectify camera_nodelet_manager --no-bond image_mono:=rgb/image_raw image_rect:=rgb/image_rect_color __name:=rectify_color __log:=/home/lluis/.ros/log/c772d3be-0244-11e6-aa51-08edb95c077a/camera-rectify_color-3.log].
log file: /home/lluis/.ros/log/c772d3be-0244-11e6-aa51-08edb95c077a/camera-rectify_color-3*.log
[FATAL] [1460642190.529826850]: Service call failed!
[camera/driver-2] process has died [pid 27717, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load openni2_camera/OpenNI2DriverNodelet camera_nodelet_manager ir:=ir rgb:=rgb depth:=depth depth_registered:=depth_registered rgb/image:=rgb/image_raw depth/image:=depth_registered/image_raw __name:=driver __log:=/home/lluis/.ros/log/c772d3be-0244-11e6-aa51-08edb95c077a/camera-driver-2.log].
log file: /home/lluis/.ros/log/c772d3be-0244-11e6-aa51-08edb95c077a/camera-driver-2*.log
```

Any idea of what I can do to solve that or use OpenNi with the RealSense camera?

Thank you so much!
Lluis

reaganlo, 2016-04-19:
@lluissalord We have not tested our package for OpenNi. We will look into this and see if this can be achieved using some launch file changes.

reaganlo, 2016-04-22:
@lluissalord 
For our RealSense package, "roslaunch realsense_camera realsense_r200_rgbd.launch" is supposed to be a replacement for "roslaunch openni2_launch openni2.launch" and only the former should be launched.

The **realsense_r200_rgbd.launch** file generates most of the "topics" that **openni2.launch** does, except for the ones listed here - [Limitations](https://github.com/intel-ros/realsense/tree/indigo-devel/camera#limitations). Please let us know if there are other "topics" for which you are not getting data.

reaganlo, 2016-05-26:
@lluissalord Please let me know if I was able to answer your question so that I can close this issue.

lluissalord, 2016-05-28:
Yes you can close this issue, because what I wanted was to use openNi with RealSense, but is not an issue of this package.

closed 2016-05-31 by reaganlo