realsenseai/realsense-ros #28: editing default parameters while using own ros-package

PPinsel, 2016-04-18:
### System Configuration
Please complete Your Configuration detail below.

| Version          | Best Known           | Your Configuration   |
|:---------------- |:---------------------|:---------------------|
| Operating System | Ubuntu 14.04.4 LTS   | Ubuntu 14.04.4 LTS  |
| Kernel           | 4.4.0-040400-generic | ?-?-?                |
| ROS              | indigo               | indigo                    |
| librealsense     | 0.9.1                | newest (April-18-2016)                |
| R200 Firmware    | 1.0.72.06            | ?.?.?.?              |

Hello,

I am new to ROS / librealsense and I want to permanently change static parameters like texture_difference_threshold such that my ROS-node gets modified depth points for further image processing. Is there a way to do so?

With the provided librealsense/cpp-config-ui I was able to temporarily change those values, but found no way to save my edits.

Thanks in advance and regards,

PP.

rjingar, 2016-04-18:
To change parameter values statically, you can specify camera parameter and desired value in the launch file. Launching RealsenseNodelet using modified launch file will set given parameter value in camera.
Once you have RealsenseNodelet running, you can also use rqt_reconfigure to change the parameter value dynamically.
Please refer to readme for supported parameters and details on how to change parameters dynamically.

PPinsel, 2016-04-19:
Perfect answer, thank you!

closed 2016-04-19 by PPinsel