realsenseai/realsense-ros #29: R200 Jetson TX1 ROS

solosito, 2016-04-20:
**### System Configuration**
Please complete Your Configuration detail below.

| Version          | Best Known           | Your Configuration   |
|:---------------- |:---------------------|:---------------------|
| Operating System | Ubuntu 14.04.4 LTS   | Ubuntu 14.04.1 LTS |
| Kernel           | 4.4.0-040400-generic | 3.10.67-g458d45c |
| ROS              | indigo               | indigo                   |
| librealsense     | 0.9.1                | 0.9.1                |
| R200 Firmware    | 1.0.72.06            | 1.0.72.06           |

**### Expected Behavior**
Realsense ROS package compiling.

**### Actual Behavior**
Examples from librealsense working properly using LibUVC backend on NVIDIA Jetson TX1, the only problem is the one about unplugging and plugging the realsense.
Realsense ROS package not compiling:

`ubuntu@tegra-ubuntu:~/rs_ws$ catkin_make
Base path: /home/ubuntu/rs_ws
Source space: /home/ubuntu/rs_ws/src
Build space: /home/ubuntu/rs_ws/build
Devel space: /home/ubuntu/rs_ws/devel
Install space: /home/ubuntu/rs_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/ubuntu/rs_ws/build"
####
####
#### Running command: "make -j4 -l4" in "/home/ubuntu/rs_ws/build"
####
[  0%] [ 11%] Built target std_msgs_generate_messages_py
Built target realsense_camera_gencfg
[ 11%] Built target std_msgs_generate_messages_cpp
[ 22%] [ 22%] Built target gtest
Built target std_msgs_generate_messages_lisp
[ 22%] Built target _realsense_camera_generate_messages_check_deps_cameraConfiguration
[ 33%] [ 44%] [ 66%] Built target realsense_camera_generate_messages_cpp
Built target realsense_camera_generate_messages_lisp
Built target realsense_camera_generate_messages_py
make[2]: *** No rule to make target `/usr/lib/arm-linux-gnueabihf/libopencv_videostab.so.2.4.8', needed by `/home/ubuntu/rs_ws/devel/lib/librealsense_camera_nodelet.so'.  Stop.
make[1]: *** [realsense/camera/CMakeFiles/realsense_camera_nodelet.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** No rule to make target `/usr/lib/arm-linux-gnueabihf/libopencv_videostab.so.2.4.8', needed by `/home/ubuntu/rs_ws/devel/lib/realsense_camera/realsense_camera_test_rgbd'.  Stop.
make[1]: *** [realsense/camera/CMakeFiles/realsense_camera_test_rgbd.dir/all] Error 2
make[2]: *** No rule to make target `/usr/lib/arm-linux-gnueabihf/libopencv_videostab.so.2.4.8', needed by `/home/ubuntu/rs_ws/devel/lib/realsense_camera/realsense_camera_test'.  Stop.
make[1]: *** [realsense/camera/CMakeFiles/realsense_camera_test.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed`

It is a problem with opencv due to TX1 is using opencv4tegra. Is there any known fix for it?

**### Steps to Reproduce**

1. Follow installation guide for librealsense using LibUVC backend and avoiding the mssse flag for compilation.
2. Download ROS package for realsense and trying to compile it with catkin_make.
3. Getting the error previously described.

mdhorn, 2016-04-20:
We currently aren't validating the LibUVC backend for librealsense primarily since it has been deprecated on Linux.

But that doesn't to be appear to be your problem. This just looks like you don't have the required OpenCV libraries installed on your system.

A little Googling shows some know issues with ROS and using opencv4tegra. Maybe this link might help you past the problem:
https://devtalk.nvidia.com/default/topic/835118/embedded-systems/incorrect-configuration-in-opencv4tegra-debian-packages-and-solution

solosito, 2016-04-25:
I don't have the same opencv4tegra version, even so I tried but it is not working. I am gonna try to find another solution.

mdhorn, 2016-05-10:
@solosito Did you find a solution to the problem?
Looking to see if there is something to share with other users before closing the issue.

solosito, 2016-05-11:
Yes, the issue is solved. 

NVidia had put the Opencv library in /usr/lib rather than /usr/lib/arm-linux-gnueabihf which is expected by ROS, so the links to the library were wrong. 

If somebody has the same error, see this thread: https://devtalk.nvidia.com/default/topic/900926/jetson-tx1/how-to-install-the-zed-ros-driver-on-the-jtx1/

mdhorn, 2016-05-11:
Great that you solved it.
Thank you for sharing the solution.

closed 2016-05-11 by mdhorn