realsenseai/realsense-ros #30: Registered depth and color

senia907, 2016-04-21:
| Version          | Best Known           | Your Configuration   |
|:---------------- |:---------------------|:---------------------|
| Operating System | Ubuntu 14.04.4 LTS   | Ubuntu 14.04.4 LTS               |
| Kernel           | 4.4.0-040400-generic | 4.4.0-040400-generic                |
| ROS              | indigo               | indigo                    |
| librealsense     | 0.9.1                | 0.9.1                |
| R200 Firmware    | 1.0.72.06            | 1.0.72.06              |

### Expected Behavior
Registered(matched) depth and color image at 640x480.
I'm trying to use rtabmap(mapping package, not so relevant to the problem) in ros, and I want to get a color and a matching depth image at 640x480 resolution.

### Actual Behavior
Using "realsense_r200_nodelet_standalone_manual.launch" (with color&depth width and height set to 640x480), and using the topics 
/camera/color/image_raw
/camera/depth/image_raw
I get the following results:
[image]
[image]
[image]

where the color and depth doesn't match.
Using "realsense_r200_rgbd.launch", which should register(match?) the color&depth gives me a 480x360 resolution at "/camera/depth/image_raw". other topics didn't give me an input, or bad(480x360) resolution.
Only "/camera/depth_registered/sw_registered/image_rect" and "#same#/image_rect_raw" gave me results at the desired resolution(640x480) but at some wired format(not as "/camera/depth/image_raw")

trying to change the default resolution at [realsense_camera_nodelet.h](https://github.com/intel-ros/realsense/blob/indigo-devel/camera/src/realsense_camera_nodelet.h) resulted in a flickering(bad) depth image. Couldn't find other direction to try.

I saw in the [limitation](https://github.com/intel-ros/realsense/tree/indigo-devel/camera#Limitations) section you mention that there isn't a hardware based registartion, but the software registration doesn't suite me either, is there a way to fix it? Or something else to try to get a matching depth and color image at 640x480.

### Steps to Reproduce
Mentioned launch files and "rosrun image_view image_view image:=#topic#"

reaganlo, 2016-04-22:
@senia907 
Depth Registration is not done within the R200 nodelet. You could use the nodelets provided in rgbd_launch to do it. 

In order to do that at your desired resolution (640x480), you can try the following:
- Replace the **realsense_r200_nodelet.launch.xml** file in the launch/includes folder with [realsense_r200_nodelet.launch.xml.txt](https://github.com/intel-ros/realsense/files/230977/realsense_r200_nodelet.launch.xml.txt) (make sure to remove the .txt extension which was added because git does not allow .xml attachments).
This file has been modified to include the "**manual**" mode, and the **resolutions** & **fps** for the depth and color streams.
- Then launch **realsense_r200_rgbd.launch** like you did before.
- Then view the **/camera/depth_registered/sw_registered/image_rect** and **/camera/depth_registered/sw_registered/image_rect_raw** topics using rViz. Make sure to un-check the **Normalize Range** option.

If you still see alignment issues and/or bad data, please let us know and we will look into this further.

senia907, 2016-04-24:
Hi,
Thanks for the replay, I changed the realsense_r200_nodelet.launch.xml file with the one you added, and launched realsense_r200_rgbd.launch, yet I got a bad(white+checkered) image at  /camera/depth_registered/sw_registered/image_rect:
[image]
 and a bad(checkered) image at #same#/image_rect_raw:
[image]
where I want the format of a regular depth image(like in /camera/depth/image_raw):
[image]
I'm using image_view package to view the images, and launching Rviz with the normalize range option disabled, where I think is connected to my problem, but I saw similler results.
Either way, if it's in Rviz It doesn't help me much as I need a registered depth stream with /camera/depth/image_raw format.
Thanks.

seniakalma, 2016-05-04:
Hi,
Is there a solution available? 
I see this problem as making the ROS driver completely unreliable.

senia907, 2016-05-04:
Hi,
Is there a solution available? 
I see this problem as making the ROS driver completely unreliable.

ddiakopoulos, 2016-05-04:
@senia907 since you would have also installed librealsense as part of this driver, maybe it would make sense to experiment with librealsense and use its rectified streams directly (see one of the several examples, like cpp-alignimage)

senia907, 2016-05-15:
Hi,
Thanks for the replay, yet cpp-alignimage is just a view, and isn't capable of changing/moving one of the streams. All other example didn't seem useful too.

linknum23, 2016-06-07:
@senia907 I have added the changes suggested to my [fork](https://github.com/linknum23/realsense). Please try using it with the realsense_r200_rgbd.launch file included. I hope it fixes your issues.

senia907, 2016-08-09:
Tried your fork, It didn't fix it.

linknum23, 2016-08-09:
So the color and depth are still out of alignment using the realsense_r200_rgbd.launch file?

senia907, 2016-08-10:
Yes, they are still out of alignment.

mkhansen-intel, 2016-10-31:
@senia907  - have you run with our newest release 1.6.0 and do you still see this issue? We've made many updates since this issue was first filed that may have helped.

bheisele, 2016-11-02:
having similar problem (r200_nodelet_rgbd.launch):
checkered artifacts in registered depth image: /camera/depth_registered/sw_registered/image_rect_raw 
on Ubuntu 16.04, 4.4.0-45-generic 
Installed precompiled version for ros-kinetic-relasense-camera
librealsense version: 1.11.0
realsense_camera: 1.5.0

mindThomas, 2017-04-14:
Can't the depth stream only be run up to a resolution of 628x468 pixels?

mindThomas, 2017-04-18:
Checkered artifacts are not actual artifacts but rather a result from the image registration. The image registration both transforms (scales, moves and rotates) the depth image into the RGB image and hence also changes the intrinsics of the image. As the resolution of the depth image is lower than the RGB image (when running in 480x360 mode) the checkered depth image is simply a result of missing pixels when the image is scaled.
See also: https://github.com/ros-perception/image_pipeline/issues/265

PS. I can confirm that both low resolution (480x360) and high resolution (640x480) works. Though both resolutions have some clear edges due to the disparity matching/generation inside the R200 camera.

icarpis, 2018-01-31:
Fixed by #261

closed 2018-01-31 by icarpis