realsenseai/realsense-ros #33: Make folder for driver consistent with package name ("realsense_camera")

skohlbr, 2016-04-24:
It is a (nearly) ubiquitously used convention within the ROS community to name a folder where a package resides the same as the package name in the ```package.xml``` file. Especially users new to ROS might be confused by the driver being placed in the "camera" folder instead of "realsense_camera" (which is the actual package name according to the [package.xml](https://github.com/intel-ros/realsense/blob/indigo-devel/camera/package.xml#L3).

mkhansen-intel, 2016-04-25:
Thanks for the input. We'll discuss internally and decide whether to proceed with this change.

ghindman, 2016-05-16:
We will take this under advisement but do not plan to implement at this time.

mdhorn, 2016-05-25:
This change will be in the the next release.

Next release is pending on a new release of https://github.com/IntelRealSense/librealsense/ due out soon.

mdhorn, 2016-06-06:
Librealsense v0.9.2 was released.
ROS Realsense package will be releasing shortly with this fix; should be version 1.1.0.

mdhorn, 2016-06-07:
Fixed in Release 1.1.0

closed 2016-06-07 by mdhorn