realsenseai/realsense-ros #36: ROS Installation of Realsense package

ecmnet, 2016-05-06:
I try to install the ROS-Librealsense package on OS X. ROS is installed properly and working well. Also librealsense is working;  cpp-capture also works fine.

When trying to run catkin_make on the ROS package, I get the following error:
```
Scanning dependencies of target _realsense_camera_generate_messages_check_deps_cameraConfiguration
Scanning dependencies of target realsense_camera_gencfg
[ 28%] Generating dynamic reconfigure files from cfg/camera_params.cfg: /Users/ecmnet/PixHawk/ros/devel/include/realsense_camera/camera_paramsConfig.h /Users/ecmnet/PixHawk/ros/devel/lib/python2.7/site-packages/realsense_camera/cfg/camera_paramsConfig.py
**/Users/ecmnet/PixHawk/ros/build/camera/setup_custom_pythonpath.sh: line 4: /Users/ecmnet/PixHawk/ros/src/camera/cfg/camera_params.cfg: Permission denied**
/Users/ecmnet/PixHawk/ros/build/camera/setup_custom_pythonpath.sh: line 4: exec: /Users/ecmnet/PixHawk/ros/src/camera/cfg/camera_params.cfg: cannot execute: Undefined error: 0
make[2]: *** [/Users/ecmnet/PixHawk/ros/devel/include/realsense_camera/camera_paramsConfig.h] Error 126
make[1]: *** [camera/CMakeFiles/realsense_camera_gencfg.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

Any advise how to overcome that issue? Thank you

mkhansen-intel, 2016-05-06:
It looks like you have a shell script that is trying to execute the .cfg file python script and it is getting permission denied. Either your shell script is not being executed with the right permissions or the cfg file is not set as executable (rwx). Check your permissions.

Also, we're not currently testing our realsense camera package with OS X. So if this works let us know.

ecmnet, 2016-05-07:
@mkhansen-intel Thank you for your hint. After doing two additional modifications (i am using PCL-Version 1.8 instead of 1.7, removing linker options in CMakeLists.txt), the package could be built without errors. The nodelet is running perfectly on jade/OSX now.

closed 2016-05-07 by ecmnet

kubark42, 2016-10-06:
@ecmnet, I have a similar installation problem. Do you think you could quickly elaborate on your answer? What in specific were the two additional modifications?