realsenseai/realsense-ros #37: error roslaunch realsense_camera realsense_r200_nodelet_standalone_preset.launch

WeJay, 2016-05-09:
### System Configuration
Please complete Your Configuration detail below.

| Version          | Best Known           | Your Configuration   |
|:---------------- |:---------------------|:---------------------|
| Operating System | Ubuntu 14.04.4 LTS   | Ubuntu 14.04.4 LTS                |
| Kernel           | 4.4.0-040400-generic | 4.4.0-040400-generic |
| Backend          | video4linux          | Video4Linux backend                  |
| ROS              | indigo               | indigo                   |
| ROS RealSense    | Latest Release       | Latest Release               |
| librealsense     | 0.9.1                |  0.9.1                |
| R200 Firmware    | 1.0.72.06            | 1.0.72.06              |

---
HI! I'm Wejay
I have same problem like issue #26,I'm not really good at linux,please tell me how to solve this 
problem.
And I can not run cpp-capture by command line,but running within QT creator works.
It shows:
RealSense error calling rs_create_context(api_version:4):
uvc_open2(...) returned Access denied
In [https://github.com/IntelRealSense/librealsense/issues/81](url),I really can not understand how to fix this problem.
[image]

mdhorn, 2016-05-09:
Yes, this looks exactly like #26.

Did you attempt the solution noted?

Try adding your login account to the group `video`.

`sudo adduser $USER video`

You may need to log out and log back into the system to see the new group.

`groups`

In general, the video devices will be owned by `root` and set to group `video`:

```
$ ls -l /dev/video*
crw-rw----+ 1 root video 81, 3 Apr 12 18:02 /dev/video3
crw-rw----+ 1 root video 81, 4 Apr 12 18:02 /dev/video4
crw-rw----+ 1 root video 81, 5 Apr 12 18:02 /dev/video5
```

WeJay, 2016-05-10:
I have attempted the solution noted. But I can not still solve my problem. 

after I  do ,
$ ls -l /dev/video*
crw-rw----+ 1 root video 81, 0  5月 10 16:45 /dev/video0
crw-rw----+ 1 root video 81, 1  5月 10 16:45 /dev/video1
crw-rw----+ 1 root video 81, 2  5月 10 16:45 /dev/video2
crw-rw----+ 1 root video 81, 3  5月 10 16:45 /dev/video3

mdhorn, 2016-05-10:
Did you try logging out and logging back in after making the group change?
The group membership only takes effect on login.

WeJay, 2016-05-12:
I tried your steps. Then, I can't still solve. 
I can not launch cpp-capture.launch.
My result show
[image]

but running within QT creator works. show 

[image]

mdhorn, 2016-05-12:
The fact that cpp-capture is failing in rs_create_context would imply that something is holding or not release the device.

In this issue you stated that you are using Video4Linux backend and NOT libuvc; can you please confirm that. It looks more like your problem is similar to https://github.com/IntelRealSense/librealsense/issues/81 where libuvc was in use. 

Can you confirm you are aren't using libuvb (which we don't support):
`dpkg -l | grep libuvc`
And that you have a current and/or patch kernel?
`uname -a`

WeJay, 2016-05-16:
sorry , I'm not really good at linux. 
Thank for your patient to teach me step by step.
[image]

I  install by below 
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#video4linux-backend

I install both Video4Linux backend and LibUVC backend.

As problem IntelRealSense/librealsense#81, I need remove LibUVC backend?

How can i do ?

mdhorn, 2016-05-16:
Okay, there is the problem; the directions are an either/or, not both. When you did the second install following the directions for libuvc, you replaced the librealsense library with the libuvc version which we don't support.

Repeat the installation steps for "Video4Linux backend".

WeJay, 2016-05-16:
When I repeat the installation steps for "Video4Linux backend", it shows they exist. Then,I can not install again.

Should I change the file name , v4.4-wily , to v4.5-wily ?

Or how can I solve the problem?

Thank you.

mdhorn, 2016-05-16:
The kernel portion should already be installed.

Just change directory (cd) to where you you put librealsense source
`cd` *\<librealsense\>*
`make clean`
`make && sudo make install`

WeJay, 2016-05-19:
Hi , I have three problem.

> Problem1
I run `sudo dmesg | tail -n 50` after I repeat the installation steps for "Video4Linux backend" . 
It shows "failed to excute"
[image]
How do I solve?
However , I can run cpp-capture.

> Problem 2 
I run `roslaunch turtlebot_bringup `minimal.launch`.
Then I meet this problem.
[image]
[image]
This turtlebot_state is error.

> Problem3 
I run `roslaunch realsense_camera realsense_r200_navigation.launch`.
[image]
The device R200 can light.
But I run `roslaunch realsense_navigation gmapping.launch`.
[image]
It shows 0 device connected.

Sorry , help me solve problems.

Thank you.

mdhorn, 2016-06-06:
It appears you got the RealSense ROS node working.

For the navigation example, did you follow all of the directions in the [README.md](https://github.com/intel-ros/realsense/blob/indigo-devel/camera/navigation/turtlebot/README.md)?

closed 2016-06-06 by mdhorn