realsenseai/realsense-ros #40: Change resolution of R200 for color and depth image

CruJon, 2016-05-17:
### System Configuration
Please complete Your Configuration detail below.

| Version          | Best Known           | Your Configuration   |
|:---------------- |:---------------------|:---------------------|
| Operating System | Ubuntu 14.04.4 LTS   | 14.04.4                |
| Kernel           | 4.4.0-040400-generic | 4.4.0-040400-generic                |
| Backend          | video4linux          | video4linux                    |
| ROS              | indigo               | indigo
| ROS RealSense    | Latest Release       | Latest Release                |
| librealsense     | 0.9.1                | 0.9.1                  |
| R200 Firmware    | 1.0.72.06            | ?.?.?.?              |

Hi there, 

I would like to start the color and depth stream with a different resolution than default values (which are 640x480). My desired resolutions for both color and depth stream are 1920x1080.

I tried to change default values in realsense_camera_nodelet.h file but that didn't work.
As stated in the documentation, there are several static parameters for these values.

Can anyone help me out on this?

Thanks in advance

reaganlo, 2016-05-17:
You may refer to this launch file as an example:
https://github.com/intel-ros/realsense/blob/indigo-devel/camera/launch/realsense_r200_nodelet_standalone_manual.launch

mdhorn, 2016-05-17:
Note that the depth stream is not available at 1920x1080.
I'm digging for better spec information on the R200, but see https://software.intel.com/en-us/articles/realsense-r200-camera
Available Resolutions
@60FPS, depth at 320x240,, color can be 640x480
@60FPS, depth at 480x360,  color can be 320x240 or 640x480
@30FPS, depth at 320x240,, color can be 640x480, 1280x720 or 1920x1080
@30FPS, depth at 480x360, color can be 320x240, 640x480, 1280x720, or 1920x1080

CruJon, 2016-05-17:
Thanks for your fast replies!

When I try to increase color resolution to anything higher than 640x480, following error appears:

`[FATAL] [1463514790.049318799]: Service call failed!
[standalone_nodelet-2] process has died [pid 13313, exit code -11, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=standalone_nodelet __log:=/home/jon/.ros/log/fa784562-1c68-11e6-ab32-b0c090467a11/standalone_nodelet-2.log].
log file: /home/jon/.ros/log/fa784562-1c68-11e6-ab32-b0c090467a11/standalone_nodelet-2*.log
[RealsenseNodelet-3] process has died [pid 13325, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load realsense_camera/RealsenseNodelet standalone_nodelet __name:=RealsenseNodelet __log:=/home/jon/.ros/log/fa784562-1c68-11e6-ab32-b0c090467a11/RealsenseNodelet-3.log].
log file: /home/jon/.ros/log/fa784562-1c68-11e6-ab32-b0c090467a11/RealsenseNodelet-3*.log`

I tried to change the resolution in realsense_camera_nodelet.h as well as in preset.launch (as shown in manual.launch). But neither worked out. 

Of the mentioned log file in the error code above there is only one log file existing. In there it says: 

`[0m[ INFO] [1463514789.021226322]: Loading nodelet /RealsenseNodelet of type realsense_camera/RealsenseNodelet to manager standalone_nodelet with the following remappings:[0m
[0m[ INFO] [1463514789.025962007]: waitForService: Service [/standalone_nodelet/load_nodelet] has not been advertised, waiting...[0m
[0m[ INFO] [1463514789.046851364]: waitForService: Service [/standalone_nodelet/load_nodelet] is now available.[0m`

Not sure if this is helpful though...

Do you have any idea what this error means?
Didn't find anything online...

Thanks again in advance!

reaganlo, 2016-07-06:
@CruJon  The depth stream is not available at 1920x1080. But you should be able to get the color stream at that resolution. Are you still getting any errors for the color stream at HD resolution?

l3s777, 2016-08-08:
hey everyone!
into the same topic, do you think it is possible to adjust/improve the quality of RGB/Depth with the values of COLOR_AUTO_EXPOSURE, HUE & SHARPNESS , etc ?
thanks!!!

reaganlo, 2016-08-09:
@l3s777 Once you launch the camera, you can adjust the dynamic parameters by running the command `$ rosrun rqt_reconfigure rqt_reconfigure`
Refer to [Dynamic Parameters] (
https://github.com/intel-ros/realsense/blob/indigo-devel/realsense_camera/README.md#dynamic-parameters) supported for different types of RealSense cameras.

l3s777, 2016-08-09:
@reaganlo  thanks for the reply... but I am not using ros, only librealsense and r200.
is there any other way ?

Thanks

reaganlo, 2016-08-09:
If you install and build librealsense from source, you can run the **cpp-configure-ui** sample program in the **`<librealsense root folder>/bin`**
For more details, I would suggest that you post the question in the [librealsense repo] (https://github.com/IntelRealSense/librealsense/issues)

closed 2016-08-09 by reaganlo

l3s777, 2016-08-10:
hi @reaganlo  I have already checked the cpp-configure-ui example, and tried to check how to fix the quality in depth but it seems to not help at all and I didn't find more info about this at the repo of librealsense.

thanks for ur help!

mindThomas, 2017-04-12:
Why is it that both color and depth stream can not be 640x480 at the same time?
Why is the depth stream limited to 480x360?

ilhamovitch, 2018-08-11:
Hello everyone I have a small question about the different resolution of color and depth images :
I want to use the depth image to segment the color image using the information from the nearest object and the fact that the 2 images have different resolutions I had a problem in segmentation I will be grateful if you have an idea or solution for this problem. All information will be useful and thank you in advance