realsenseai/realsense-ros #41: what is the depth images scale value

liyuming1978, 2016-05-20:
http://vision.in.tum.de/data/datasets/rgbd-dataset/file_formats 
has said  "The depth images are scaled by a factor of 5000, i.e., a pixel value of 5000 in the depth image corresponds to a distance of 1 meter from the camera, 10000 to 2 meter distance, etc. A pixel value of 0 means missing value/no data"

I just want to know what is the value for R200 
? = 1 meter for R200 depth camera.

linknum23, 2016-05-20:
This value is actually configurable in the underlying librealsense driver. The default for the R200 camera, which is not changed in this ros driver, is 1mm resolution. So a value of 1000 corresponds to 1 meter. Here is where to find this information: [https://github.com/IntelRealSense/librealsense/blob/master/doc/projection.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/projection.md).
FYI: In ROS the depth is expected to be in one of 2 formats: 32-bit float in meters or 16-bit unsigned integers in millimeter resolution. This ros realsense driver currently publishes depth in 16-bit 1 millimeter resolution.

closed 2016-06-06 by mdhorn

jlysnantel, 2019-03-14:
Is there a topic that return the depth_scale for the current camera ? I am using the D435

doronhi, 2019-03-14:
The D435 publishes depth in "16-bit unsigned integers in millimeter resolution."

jlysnantel, 2019-03-15:
Even in the the topic /camera/aligned_depth_to_color/image_raw? When I echo the topic I only got values between 0 and 255.

iscsrl, 2020-07-08:
As @jlysnantel I've the same need to get the depth of a single pixel. Is there any way I can do it through any Ros topic?

doronhi, 2020-07-08:
@jlysnantel , sorry for failing to return to you.
I believe you already figured thing out by now or moved on but still..
Check out the function "imageDepthCallback" in the file: "realsense/realsense2_camera/scripts/show_center_depth.py"
There is an example, in python, for subscribing and printing the depth value.
If you add, in line 16: `print cv_image.dtype` you'll see that the image type is `uint16`. So the values you get in the range of 0-255 just mean that you need to cast the buffer differently.

The format is defined in [REP-118](https://www.ros.org/reps/rep-0118.html), under the "OpenNI Raw Representation" section.

iscsrl, 2020-07-08:
@doronhi I was already using this solution but I'm not able to make the following line work with python3:

`cv_image = self.bridge.imgmsg_to_cv2(data, data.encoding)`

It only works correctly on python, but running it with python3 (python 3.6 version) I get the following error:

```
[ERROR] [1594211659.759482]: bad callback: <function convert_depth_image at 0x7f6079571e18>
Traceback (most recent call last):
    ...
from cv_bridge.boost.cv_bridge_boost import getCvType
ImportError: dynamic module does not define module export function (PyInit_cv_bridge_boost)
```
It looks like it cannot import `from cv_bridge.boost.cv_bridge_boost import getCvType`.
Any tips to solve this?

Shrutii07, 2021-05-28:
> This value is actually configurable in the underlying librealsense driver. The default for the R200 camera, which is not changed in this ros driver, is 1mm resolution. So a value of 1000 corresponds [...]

The link is not working. Could you please direct me the correct link?
Even I am looking to extract single depth value in mm for a depth img from R200 camera using ros.