realsenseai/realsense-ros #42: Invalid roslaunch syntax

Oushesh, 2016-05-24:
### System Configuration
Please complete Your Configuration detail below.

| Version          | Best Known           | Your Configuration   |
|:---------------- |:---------------------|:---------------------|
| Operating System | Ubuntu 14.04.4 LTS   | ?                 |
| Kernel           | 4.4.0-040400-generic | ?-?-?                |
| Backend          | video4linux          | ?                    |
| ROS              | indigo               | ?                    |
| ROS RealSense    | Latest Release       | ?.?.?                |
| librealsense     | 0.9.1                | ?.?.?                |
| R200 Firmware    | 1.0.72.06            | ?.?.?.?              |

---
#### How to collect Configuration Data
*This section can be deleted before submission.*

| Version          | Method |
|:---------------- |:------------ |
| Operating System | `grep DISTRIB_DESCRIPTION /etc/*elease*` |
| Kernel           | `uname -r` |
| Backend          | `ls /sys/class | grep video4linux` |
| ROS              | `rosversion -d` |
| ROS RealSense    | `rosversion realsense_camera` |
| librealsense     | `cat <path to librealsense>/librealsense/readme.md | grep release-image | awk -F- '{print $3}'` |
| R200 Firmware    | View the ROS log from running nodelet **OR** `<path to librealsense>/librealsense/bin/cpp-enumerate-devices | grep -i firmware` |

---

### Expected Behavior
launching the the launch files when called with roslaunch in the folder where the launch files are found

### Actual Behavior
while processing /home/oushesh/catkin_ws_r200/src/realsense_camera/launch/includes/realsense_r200_nodelet.launch.xml: Invalid roslaunch XML syntax: [Errno 2] No such file or directory: u'/home/oushesh/catkin_ws_r200/src/realsense_camera/launch/includes/realsense_r200_nodelet.launch.xml'

### Steps to Reproduce

Any idea why this is so? 

I tried also a different git :  
https://github.com/BlazingForests/realsense_camera/

with this one its working fine. only the I do not get RGB picture. Depth picture and Point Clouds I got but was quite noisy.

So I wanted to try this new git since its more recent. 

Thanks for any help guys.

reaganlo, 2016-05-26:
The error shows that you dont have the following file - https://github.com/intel-ros/realsense/blob/indigo-devel/camera/launch/includes/realsense_r200_nodelet.launch.xml
Can you please check if this file is present in your workspace. It should have been part of the package download. 

Also ensure that you do the following steps:
- Run "rosdep install realsense_camera" to install package dependencies
- Compile the "realsense_camera" package by executing the "catkin_make" command.
- Then try launching the https://github.com/intel-ros/realsense/blob/indigo-devel/camera/launch/realsense_r200_rgbd.launch file

Oushesh, 2016-05-26:
Hi man,
I actually I checked and I have the realsense_r200_nodelet.launch.xml in
the directory /realsense/camera/launch/includes...

and running "rosdep install realsense_camera"  says  #All required rosdeps
installed successfully

I realised the bash path in my ./bashrc file was something else.. so I
renamed my workspace accordingly and initialised the workspace and created
a new package with catkin_create_package with name realsense, cloned the
contents from your repository on Github and overworte the realsense.

Run Catkin_make again. Everything's fine. no errors. then I go   until the
launch folder and with roslaunch I do the rgbd.launch....

This is what I get:

started roslaunch server http://oushesh-550P5C-550P7C:46416/
>
> SUMMARY
> ========
>
> PARAMETERS
>  * /camera/camera_nodelet/base_frame_id: camera_link
>  * /camera/camera_nodelet/color_fps: 30
>  * /camera/camera_nodelet/color_frame_id: camera_rgb_frame
>  * /camera/camera_nodelet/color_height: 480
>  * /camera/camera_nodelet/color_optical_frame_id: camera_rgb_optica...
>  * /camera/camera_nodelet/color_width: 640
>  * /camera/camera_nodelet/depth_fps: 30
>  * /camera/camera_nodelet/depth_frame_id: camera_depth_frame
>  * /camera/camera_nodelet/depth_height: 360
>  * /camera/camera_nodelet/depth_optical_frame_id: camera_depth_opti...
>  * /camera/camera_nodelet/depth_width: 480
>  * /camera/camera_nodelet/enable_color: True
>  * /camera/camera_nodelet/enable_depth: True
>  * /camera/camera_nodelet/enable_pointcloud: False
>  * /camera/camera_nodelet/enable_tf: True
>  * /camera/camera_nodelet/ir2_frame_id: camera_ir2_frame
>  * /camera/camera_nodelet/ir_frame_id: camera_ir_frame
>  * /camera/camera_nodelet/mode: manual
>  * /camera/camera_nodelet/serial_no:
>  * /camera/camera_nodelet_manager/num_worker_threads: 4
>  * /camera/depth_rectify_depth/interpolation: 0
>  * /camera/disparity_depth/max_range: 4.0
>  * /camera/disparity_depth/min_range: 0.5
>  * /camera/disparity_registered_sw/max_range: 4.0
>  * /camera/disparity_registered_sw/min_range: 0.5
>  * /rosdistro: indigo
>  * /rosversion: 1.11.19
>
> NODES
>   /camera/
>     camera_nodelet (nodelet/nodelet)
>     camera_nodelet_manager (nodelet/nodelet)
>     debayer (nodelet/nodelet)
>     depth_metric (nodelet/nodelet)
>     depth_metric_rect (nodelet/nodelet)
>     depth_points (nodelet/nodelet)
>     depth_rectify_depth (nodelet/nodelet)
>     depth_registered_sw_metric_rect (nodelet/nodelet)
>     disparity_depth (nodelet/nodelet)
>     disparity_registered_sw (nodelet/nodelet)
>     points_xyzrgb_sw_registered (nodelet/nodelet)
>     rectify_color (nodelet/nodelet)
>     rectify_ir (nodelet/nodelet)
>     rectify_mono (nodelet/nodelet)
>     register_depth_rgb (nodelet/nodelet)
>
> auto-starting new master
> process[master]: started with pid [30296]
> ROS_MASTER_URI=http://localhost:11311
>
> setting /run_id to 31649f2e-2341-11e6-874c-e8039ab1eed7
> process[rosout-1]: started with pid [30309]
> started core service [/rosout]
> process[camera/camera_nodelet_manager-2]: started with pid [30326]
> process[camera/camera_nodelet-3]: started with pid [30327]
> process[camera/debayer-4]: started with pid [30328]
> process[camera/rectify_mono-5]: started with pid [30329]
> process[camera/rectify_color-6]: started with pid [30340]
> process[camera/rectify_ir-7]: started with pid [30347]
> process[camera/depth_rectify_depth-8]: started with pid [30361]
> process[camera/depth_metric_rect-9]: started with pid [30368]
> process[camera/depth_metric-10]: started with pid [30381]
> process[camera/depth_points-11]: started with pid [30389]
> process[camera/register_depth_rgb-12]: started with pid [30397]
> process[camera/points_xyzrgb_sw_registered-13]: started with pid [30407]
> process[camera/depth_registered_sw_metric_rect-14]: started with pid
> [30417]
> process[camera/disparity_depth-15]: started with pid [30427]
> process[camera/disparity_registered_sw-16]: started with pid [30434]
> [ INFO] [1464267360.582433279]: Initializing nodelet with 4 worker threads.
> [ERROR] [1464267364.873147377]: Failed to load nodelet
> [/camera/camera_nodelet] of type [realsense_camera/RealsenseNodelet] even
> after refreshing the cache: Could not find library corresponding to plugin
> realsense_camera/RealsenseNodelet. Make sure the plugin description XML
> file has the correct name of the library and that the library actually
> exists.
> [ERROR] [1464267364.873303701]: The error before refreshing the cache was:
> Could not find library corresponding to plugin
> realsense_camera/RealsenseNodelet. Make sure the plugin description XML
> file has the correct name of the library and that the library actually
> exists.
> [FATAL] [1464267364.873604928]: Service call failed!
> [camera/camera_nodelet-3] process has died [pid 30327, exit code 255, cmd
> /opt/ros/indigo/lib/nodelet/nodelet load realsense_camera/RealsenseNodelet
> camera_nodelet_manager camera/depth/image_raw:=depth/image_raw
> camera/color/image_raw:=rgb/image_raw
> camera/infrared1/image_raw:=ir/image_raw
> camera/infrared2/image_raw:=ir2/image_raw camera/depth/points:=depth/points
> camera/depth/camera_info:=depth/camera_info
> camera/color/camera_info:=rgb/camera_info
> camera/infrared1/camera_info:=ir/camera_info
> camera/infrared2/camera_info:=ir2/camera_info __name:=camera_nodelet
> __log:=/home/oushesh/.ros/log/31649f2e-2341-11e6-874c-e8039ab1eed7/camera-camera_nodelet-3.log].
> log file:
> /home/oushesh/.ros/log/31649f2e-2341-11e6-874c-e8039ab1eed7/camera-camera_nodelet-3*.log
>
>
If you did any particular installation, could write me down the steps. I
could try it out.

From a different repository I get the r200 running but not the RGB
picture.. Thats why I am keen to try this stack.

Thanks mate!!!

reaganlo, 2016-06-06:
@Oushesh Looks like the initial issue you had has been resolved.

Regarding the latest issue, the only way I could reproduce was by manually deleting the "catkin_ws/devel/lib/librealsense_camera_nodelet.so" file. Can you please check if this file is getting generated in this folder after you do the catkin_make.

Also, I see that you are doing the "catkin_create_pkg" for realsense. You don't have to do that step. You can directly clone our repo into your "catkin_ws/src/" folder and do a "catkin_make" from your catkin_ws

Oushesh, 2016-06-09:
Hi man. Thanks for your reply. Actually I did again the way you said. I
cloned the link directly into the catkin_ws/src and made catkin_make.

But I do not see the file librealsense_camera_nodelet.so in the folder
"catkin_ws/devel/lib/??

How did you resolve this??

reaganlo, 2016-07-06:
@Oushesh Were you able to resolve this issue? The "catkin_make" command should place the library in the "devel/lib" folder based on the contents of the "CMakeLists.txt" file. If its still not working, try deleting the "build" and "devel" folders inside your catkin_ws (rm -rf build devel) and try catkin_make again.

reaganlo, 2016-08-09:
Closing due to no response.

closed 2016-08-09 by reaganlo

zubair123khan, 2017-03-08:
i am facing errors in my xml file ,, [ERROR] [1488985784.397939712]: Failed to load nodelet [/camera/person] of type [realsense_person/PersonNodelet] even after refreshing the cache: Could not find library corresponding to plugin realsense_person/PersonNodelet. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1488985784.397991545]: The error before refreshing the cache was: Could not find library corresponding to plugin realsense_person/PersonNodelet. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[FATAL] [1488985784.398151586]: Failed to load nodelet '/camera/person` of type `realsense_person/PersonNodelet` to manager `manager'
[camera/person-4] process has died [pid 4532, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load realsense_person/PersonNodelet manager depth:=depth color:=rgb person:=person __name:=person __log:=/home/zubair/.ros/log/41ec2720-0411-11e7-8d2b-5800e315aa80/camera-person-4.log].
log file: /home/zubair/.ros/log/41ec2720-0411-11e7-8d2b-5800e315aa80/camera-person-4*.log

please help

zubair123khan, 2017-03-09:
not able to launch realsense_person tracking ,, getting so many errors, please help

mdhorn, 2017-03-09:
Person tracking is not part of this repository.

The pervious repository was removed as it required proprietary library which are not publicly available. The support for Person tracking is TBD once the team that owns the product has an open source solution.

For future issues related to RealSense Camera package, please open a new issue rather than commenting on a closed issue. Closed issues are generally ignored.

zubair123khan, 2017-03-10:
do you know any other person tracking library, for R200 realsense camera ?