realsenseai/realsense-ros #47: Remove native node support for PointCloud

mdhorn, 2016-06-07:
### System Configuration
Please complete Your Configuration detail below. Refer to the BKC at [Configuration](../realsense_camera/README.md#configuration).

| Version          | Your Configuration   |
|:---------------- |:---------------------|
| Operating System | Ubuntu 16.04 LTS                |
| Kernel           | 4.4.0-22-generic                |
| Backend          |  video4linux                   |
| ROS              | kinetic               |
| ROS RealSense    | 1.1.0              |
| librealsense     | v0.9.2               |
| R200 Firmware    | 1.0.72.06             |

### Expected Behavior
No frame loss on Color, Depth, and IR streams.

### Actual Behavior
Running streams with higher frame rates and/or resolution can result in dropped frames.

The publishing loop for the stream is slowed when native PointCloud generation is enabled in the ROS RealSense node. If kept, this should be placed into a separate thread for publishing the PointCould.

Instead of having the functionality native in the Node, most user would use the existing ROS packages to generate any needed PointClouds.

Recommendation: Remove the native PointCloud from the ROS RealSense node as it is redundant and creating unnecessary overhead.

rljordan, 2017-02-22:
Will not fix.

closed 2017-02-22 by rljordan

skohlbr, 2017-08-12:
Given that the rgbd-nodelet based point cloud has severe registration error that hasn't been fixed since more than a year, I think that for many applications some dropped frames would be acceptable, if the cloud is actually registered correctly. Alternatively, fixing #201 (and related issues linked there) would be preferable.