realsenseai/realsense-ros #55: xyzrgb point cloud freezes in rviz after displaying for 15 seconds

jgoppert, 2016-06-21:
This is a regression on issue #25

### System Configuration
Please complete Your Configuration detail below. Refer to the BKC at [Configuration](../realsense_camera/README.md#configuration).

| Version          | Your Configuration   |
|:---------------- |:---------------------|
| Operating System | ? ? ?                |
| Kernel           | ?-?-?                |
| Backend          | ?                    |
| ROS              | ?                    |
| ROS RealSense    | ?.?.?                |
| librealsense     | ?.?.?                |
| R200 Firmware    | ?.?.?.?              |

---
#### How to collect Configuration Data
*This section can be deleted before submission.*

| Version          | Method |
|:---------------- |:------------ |
| Operating System | `grep DISTRIB_DESCRIPTION /etc/*elease*` |
| Kernel           | `uname -r` |
| Backend          | `ls /sys/class | grep video4linux` |
| ROS              | `rosversion -d` |
| ROS RealSense    | `rosversion realsense_camera` |
| librealsense     | `cat <path to librealsense>/librealsense/readme.md | grep release-image | awk -F- '{print $3}'` |
| R200 Firmware    | View the ROS log from running nodelet **OR** `<path to librealsense>/librealsense/bin/cpp-enumerate-devices | grep -i firmware` |

---

### Expected Behavior

### Actual Behavior

### Steps to Reproduce

mdhorn, 2016-06-28:
Can we get you current configuration details?

Please complete Your Configuration detail below. Refer to the BKC at [Configuration](../realsense_camera/README.md#configuration).

| Version          | Your Configuration   |
|:---------------- |:---------------------|
| Operating System | ? ? ?                |
| Kernel           | ?-?-?                |
| Backend          | ?                    |
| ROS              | ?                    |
| ROS RealSense    | ?.?.?                |
| librealsense     | ?.?.?                |
| R200 Firmware    | ?.?.?.?              |
| HW CPU model and number | ? |
| RAM | ? |

jgoppert, 2016-06-28:
I can't reproduce this anymore. Closing.

closed 2016-06-28 by jgoppert