realsenseai/realsense-ros #57: problem cpp-capture example : rs.warn: Subdevice 1 bad magic number 0x0

won13y, 2016-06-23:
I tried to connect the camera to a USB3 port and run the "cpp-capture

" sample program in the 

"librealsense/bin" folder. 

it works as following figures:

[image]

COLOR image is correctly shown, 

but detph / infrared1/infrared 2 image are not enabled, and are shown as black

    rs.warn: Subdevice 1 bad magic number 0x0

ddiakopoulos, 2016-06-25:
This is an installation issue on your Ubuntu system. Please refer to the issue you created on the librealsense repo for resolution (you need to follow the installation.md more closely): https://github.com/IntelRealSense/librealsense/issues/183

mdhorn, 2016-06-27:
Appears to be tracked in librealsense.
Closing.

closed 2016-06-27 by mdhorn