realsenseai/realsense-ros #58: proportional-to-distance mean error of depth measurements

imbachb, 2016-06-23:
### System Configuration
| Version          | Your Configuration   |
|:---------------- |:---------------------|
| Operating System | Ubuntu 14.04.4 LTS               |
| Kernel           | 4.4.0-040400-generic               |
| Backend          | video4linux                    |
| ROS              | indigo                    |
| ROS RealSense    | 1.0.4                |
| librealsense     | 0.9.1                |
| R200 Firmware    | 1.0.72.06              |

I'm currently trying to use the R200 for 3D map generation. Something that I observed is that the (or my) R200 seems to have a mean error in depth measurements  that is dependent on the measured distance. The following maps were generated while moving the quadrocopter slowly towards a wall. In the first image you can clearly see how the wall shifts forward when the quadrotor gets nearer (i.e. mean error gets smaller). In the second image I used a quadratic function of the measured distance to subtract the mean error.

[image] [image]

I'm wondering whether this is a problem of my R200 or whether this is expected behaviour.
Depth image resolution is 320x240@30fps. I don't modify any of the calibration settins.

ghindman, 2016-06-28:
What ROS package were you using to generate the map data, and is this issue reproducible outside of ROS?  It's difficult to tell if this is a ROS issue, or a librealsense issue with the data provided.

imbachb, 2016-07-05:
What would be the simplest way to do this outside of ROS on Ubuntu? Is there an example program or something like that that outputs the depth?

I hand the pointclouds from the realsense package over to the octomap package. The octomap cointains a modified stereo camera inverse sensor model to account for the measurement error distribution.
The error however was not only detected when using octomap. It's already present in the pointcloud generated by the realsense package.

tripzero, 2016-10-31:
@imbachb can you ask this question to upstream librealsense?  I believe we are just forwarding the mean error data from librealsense.  They could probably give you better feedback on whether or not what you are seeing is expected behavior or not.

closed 2017-02-22 by rljordan