realsenseai/realsense-ros #63: Cannot install realsense_camera

jb892, 2016-07-04:
### System Configuration
Please complete Your Configuration detail below. Refer to the BKC at [Configuration](../realsense_camera/README.md#configuration).

| Version          | Your Configuration   |
|:---------------- |:---------------------|
| Operating System |   Ubuntu 16.04 LTS     |
| Kernel           |  4.4.0-040400-generic            |
| Backend          |                    |
| ROS              | kinetic       |
| ROS RealSense    |   Cannot locate [realsense_camera]   |
| librealsense     |      0.9.1         |
| R200 Firmware    |   1.0.72.06      |

### Expected Behavior
Install realsense_camera

### Actual Behavior
$ rosdep install realsense_camera

ERROR: Rosdep cannot find all required resources to answer your query
Missing resource realsense_camera
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/opt/ros/kinetic/share
ROS path [2]=/opt/ros/kinetic/stacks

### Steps to Reproduce

mdhorn, 2016-07-04:
realsense_camera is not a release package, that command appears to be incorrect in the README.md directions
$ `rosdep install --from-paths` _< location where you cloned realsense repo >_

jb892, 2016-07-04:
Hi mdhorn, 
Thanks for your reply! However, I got this error:
<---------------------------------------------------------------------------------------------------------------------->
$ rosdep install --from-paths /home/jack/catkin_ws/src/realsense_camera 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-kinetic-librealsense
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install ros-kinetic-librealsense] failed
<---------------------------------------------------------------------------------------------------------------------->
How to solve this?

mdhorn, 2016-07-04:
For the latest release, this should work:

$ `rosdep install --skip-keys=librealsense --from-paths`_< location where you cloned realsense repo >_

Please confirm and we will update the README.md.

jb892, 2016-07-05:
Cheers mate! It does solved this problem
<---------------------------------------------------------------------->
#All required rosdeps installed successfully
<---------------------------------------------------------------------->

However, when I try to compile this package, I got these errors:
<---------------------------------------------------------------------->
$ catkin_make
...
[ 95%] Built target realsense_camera_nodelet
[ 97%] Linking CXX executable /home/jack/catkin_ws/devel/lib/realsense_camera/tests_rgbd_topics
CMakeFiles/tests_rgbd_topics.dir/test/rgbd_topics.cpp.o: In function `getParams()':
rgbd_topics.cpp:(.text+0x1992): undefined reference to `ros::NodeHandle::NodeHandle(std::string const&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&)'
rgbd_topics.cpp:(.text+0x19ce): undefined reference to `ros::NodeHandle::hasParam(std::string const&) const'
rgbd_topics.cpp:(.text+0x19e7): undefined reference to `ros::NodeHandle::getParam(std::string const&, std::string&) const'
rgbd_topics.cpp:(.text+0x1a2d): undefined reference to `ros::NodeHandle::hasParam(std::string const&) const'
rgbd_topics.cpp:(.text+0x1a46): undefined reference to `ros::NodeHandle::getParam(std::string const&, std::string&) const'
rgbd_topics.cpp:(.text+0x1a8d): undefined reference to `ros::NodeHandle::hasParam(std::string const&) const'
rgbd_topics.cpp:(.text+0x1aa6): undefined reference to `ros::NodeHandle::getParam(std::string const&, std::string&) const'
rgbd_topics.cpp:(.text+0x1ae8): undefined reference to `ros::NodeHandle::hasParam(std::string const&) const'
rgbd_topics.cpp:(.text+0x1b01): undefined reference to `ros::NodeHandle::getParam(std::string const&, std::string&) const'
rgbd_topics.cpp:(.text+0x1b43): undefined reference to `ros::NodeHandle::hasParam(std::string const&) const'
rgbd_topics.cpp:(.text+0x1b5c): undefined reference to `ros::NodeHandle::getParam(std::string const&, std::string&) const'
CMakeFiles/tests_rgbd_topics.dir/test/rgbd_topics.cpp.o: In function `ros::SubscriptionCallbackHelperT<boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&, void>::deserialize(ros::SubscriptionCallbackHelperDeserializeParams const&)':
rgbd_topics.cpp:(.text._ZN3ros27SubscriptionCallbackHelperTIRKN5boost10shared_ptrIKN11sensor_msgs12PointCloud2_ISaIvEEEEEvE11deserializeERKNS_43SubscriptionCallbackHelperDeserializeParamsE[_ZN3ros27SubscriptionCallbackHelperTIRKN5boost10shared_ptrIKN11sensor_msgs12PointCloud2_ISaIvEEEEEvE11deserializeERKNS_43SubscriptionCallbackHelperDeserializeParamsE]+0x800): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
CMakeFiles/tests_rgbd_topics.dir/test/rgbd_topics.cpp.o: In function `ros::SubscriptionCallbackHelperT<boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&, void>::deserialize(ros::SubscriptionCallbackHelperDeserializeParams const&)':
rgbd_topics.cpp:(.text._ZN3ros27SubscriptionCallbackHelperTIRKN5boost10shared_ptrIKN11sensor_msgs6Image_ISaIvEEEEEvE11deserializeERKNS_43SubscriptionCallbackHelperDeserializeParamsE[_ZN3ros27SubscriptionCallbackHelperTIRKN5boost10shared_ptrIKN11sensor_msgs6Image_ISaIvEEEEEvE11deserializeERKNS_43SubscriptionCallbackHelperDeserializeParamsE]+0x653): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
CMakeFiles/tests_rgbd_topics.dir/test/rgbd_topics.cpp.o: In function `main':
rgbd_topics.cpp:(.text.startup+0x5f): undefined reference to `ros::init(int&, char**, std::string const&, unsigned int)'
rgbd_topics.cpp:(.text.startup+0x108): undefined reference to `ros::NodeHandle::NodeHandle(std::string const&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&)'
rgbd_topics.cpp:(.text.startup+0xb65): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
rgbd_topics.cpp:(.text.startup+0xbf3): undefined reference to `ros::console::print(ros::console::FilterBase*, void*, ros::console::levels::Level, std::basic_stringstream<char, std::char_traits<char>, std::allocator<char> > const&, char const*, int, char const*)'
CMakeFiles/tests_rgbd_topics.dir/test/rgbd_topics.cpp.o: In function `ros::SubscriptionCallbackHelperT<boost::shared_ptr<sensor_msgs::CameraInfo_<std::allocator<void> > const> const&, void>::deserialize(ros::SubscriptionCallbackHelperDeserializeParams const&)':
rgbd_topics.cpp:(.text._ZN3ros27SubscriptionCallbackHelperTIRKN5boost10shared_ptrIKN11sensor_msgs11CameraInfo_ISaIvEEEEEvE11deserializeERKNS_43SubscriptionCallbackHelperDeserializeParamsE[_ZN3ros27SubscriptionCallbackHelperTIRKN5boost10shared_ptrIKN11sensor_msgs11CameraInfo_ISaIvEEEEEvE11deserializeERKNS_43SubscriptionCallbackHelperDeserializeParamsE]+0x8f7): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
collect2: error: ld returned 1 exit status
realsense_camera/CMakeFiles/tests_rgbd_topics.dir/build.make:361: recipe for target '/home/jack/catkin_ws/devel/lib/realsense_camera/tests_rgbd_topics' failed
make[2]: *** [/home/jack/catkin_ws/devel/lib/realsense_camera/tests_rgbd_topics] Error 1
[... cut ...]
realsense_camera/CMakeFiles/tests_camera_core.dir/build.make:361: recipe for target '/home/jack/catkin_ws/devel/lib/realsense_camera/tests_camera_core' failed
make[2]: *** [/home/jack/catkin_ws/devel/lib/realsense_camera/tests_camera_core] Error 1
CMakeFiles/Makefile2:1866: recipe for target 'realsense_camera/CMakeFiles/tests_camera_core.dir/all' failed
make[1]: *** [realsense_camera/CMakeFiles/tests_camera_core.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
<----------------------------------------------------------------------------------->

mdhorn, 2016-07-06:
It is hard to tell from the information you include where the build is failing....I'm pretty sure there is a failure above where you included. You might try attaching the full build log.

Also, try build in a single thread so that the build will stop on the first failure.
`cd /home/jack/catkin_ws`
`catkin_make -j1`

jb892, 2016-07-06:
## Hi man, here is the full build log:
<-------------------------------------------------------------------------------------------------------------->
jack@jack-All-Series:~/catkin_ws$ catkin_make -j1 install
Base path: /home/jack/catkin_ws
Source space: /home/jack/catkin_ws/src
Build space: /home/jack/catkin_ws/build
Devel space: /home/jack/catkin_ws/devel
Install space: /home/jack/catkin_ws/install
####
#### Running command: "cmake /home/jack/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/jack/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/jack/catkin_ws/install -G Unix Makefiles" in "/home/jack/catkin_ws/build"
####
CMake Error: The current CMakeCache.txt directory /home/jack/catkin_ws/build/CMakeCache.txt is different than the directory /home/jack/catkin_ws_ps/build where CMakeCache.txt was created. This may result in binaries being created in the wrong place. If you are not sure, reedit the CMakeCache.txt
-- Using CATKIN_DEVEL_PREFIX: /home/jack/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/kinetic
-- This workspace overlays: /opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/jack/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.1
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - realsense_camera
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'realsense_camera'
-- ==> add_subdirectory(realsense_camera)
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:90 (find_package):
  Could not find a package configuration file provided by "librealsense" with
  any of the following names:

    librealsenseConfig.cmake
    librealsense-config.cmake

  Add the installation prefix of "librealsense" to CMAKE_PREFIX_PATH or set
  "librealsense_DIR" to a directory containing one of the above files.  If
  "librealsense" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  realsense_camera/CMakeLists.txt:20 (find_package)

-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- realsense_camera: 0 messages, 1 services
-- Configuring incomplete, errors occurred!
See also "/home/jack/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/jack/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
jack@jack-All-Series:~/catkin_ws$ source devel/setup.bash
jack@jack-All-Series:~/catkin_ws$ catkin_make -j1 install
Base path: /home/jack/catkin_ws
Source space: /home/jack/catkin_ws/src
Build space: /home/jack/catkin_ws/build
Devel space: /home/jack/catkin_ws/devel
Install space: /home/jack/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/jack/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/jack/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/jack/catkin_ws/devel;/opt/ros/kinetic
-- This workspace overlays: /home/jack/catkin_ws/devel;/opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/jack/catkin_ws/build/test_results
[... 226 lines cut ...]
#### Running command: "make cmake_check_build_system" in "/home/jack/catkin_ws/build"
####
####
#### Running command: "make -j1" in "/home/jack/catkin_ws/build"
####
[ 50%] Linking CXX executable /home/jack/catkin_ws/devel/lib/my_pcl_tutorial/example
CMakeFiles/example.dir/src/example.cpp.o: In function `main':
example.cpp:(.text+0x116): undefined reference to `ros::init(int&, char**, std::string const&, unsigned int)'
example.cpp:(.text+0x172): undefined reference to `ros::NodeHandle::NodeHandle(std::string const&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&)'
CMakeFiles/example.dir/src/example.cpp.o: In function `void ros::Publisher::publish<sensor_msgs::PointCloud2_<std::allocator<void> > >(sensor_msgs::PointCloud2_<std::allocator<void> > const&) const':
example.cpp:(.text._ZNK3ros9Publisher7publishIN11sensor_msgs12PointCloud2_ISaIvEEEEEvRKT_[_ZNK3ros9Publisher7publishIN11sensor_msgs12PointCloud2_ISaIvEEEEEvRKT_]+0xa0): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
example.cpp:(.text._ZNK3ros9Publisher7publishIN11sensor_msgs12PointCloud2_ISaIvEEEEEvRKT_[_ZNK3ros9Publisher7publishIN11sensor_msgs12PointCloud2_ISaIvEEEEEvRKT_]+0x1a3): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
example.cpp:(.text._ZNK3ros9Publisher7publishIN11sensor_msgs12PointCloud2_ISaIvEEEEEvRKT_[_ZNK3ros9Publisher7publishIN11sensor_msgs12PointCloud2_ISaIvEEEEEvRKT_]+0x29e): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
[... cut ...]
my_pcl_tutorial/CMakeFiles/example.dir/build.make:314: recipe for target '/home/jack/catkin_ws/devel/lib/my_pcl_tutorial/example' failed
make[2]: *** [/home/jack/catkin_ws/devel/lib/my_pcl_tutorial/example] Error 1
CMakeFiles/Makefile2:1776: recipe for target 'my_pcl_tutorial/CMakeFiles/example.dir/all' failed
make[1]: *** [my_pcl_tutorial/CMakeFiles/example.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j1" failed
<-------------------------------------------------------------------------------------------------------------->

mdhorn, 2016-07-06:
Are you sourcing the develop setup file?

0) `cd /home/jack/catkin_ws`
1) Remove build and devel folders: `rm -r devel/ build/`
2) Source: source devel/setup.bash
3) Compile using catkin_make --pkg realsense_camera
4) Source again, step 2)

jb892, 2016-07-07:
You have removed devel/ folder in step 1). How can you source the setup file in devel/ in step 2)?

mdhorn, 2016-07-07:
0) cd /home/jack/catkin_ws
1) Remove build and devel folders: rm -r devel/ build/
2) Compile using catkin_make --pkg realsense_camera
3) Source: source devel/setup.bash

jb892, 2016-07-07:
<-------------------------------------------------------------------------------------------------------------->
jack@jack-All-Series:~/catkin_ws$ rm -r devel/ build/
jack@jack-All-Series:~/catkin_ws$ catkin_make --pkg realsense_camera
Base path: /home/jack/catkin_ws
Source space: /home/jack/catkin_ws/src
Build space: /home/jack/catkin_ws/build
Devel space: /home/jack/catkin_ws/devel
Install space: /home/jack/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/jack/catkin_ws/build"
####
####
#### Running command: "make -j8 -l8" in "/home/jack/catkin_ws/build/realsense_camera"
####
[  0%] Built target tf_generate_messages_nodejs
[  0%] Built target pcl_msgs_generate_messages_nodejs
[ 11%] Built target gtest
[ 11%] Built target std_msgs_generate_messages_cpp
[ 11%] Built target topic_tools_generate_messages_lisp
[ 11%] Built target topic_tools_generate_messages_nodejs
[ 11%] Built target pcl_msgs_generate_messages_py
[ 11%] Built target _realsense_camera_generate_messages_check_deps_cameraConfiguration
[ 11%] Built target pcl_msgs_generate_messages_lisp
[ 11%] Built target pcl_msgs_generate_messages_cpp
[ 11%] Built target actionlib_msgs_generate_messages_py
[ 11%] Built target tf2_msgs_generate_messages_cpp
[ 11%] Built target tf2_msgs_generate_messages_lisp
[ 11%] Built target topic_tools_generate_messages_py
[ 11%] Built target tf2_msgs_generate_messages_py
[ 11%] Built target actionlib_msgs_generate_messages_nodejs
[ 11%] Built target actionlib_generate_messages_py
[ 11%] Built target actionlib_msgs_generate_messages_lisp
[ 11%] Built target actionlib_msgs_generate_messages_eus
[ 11%] Built target actionlib_generate_messages_lisp
[ 11%] Built target actionlib_generate_messages_eus
[ 11%] Built target tf_generate_messages_py
[ 11%] Built target rosgraph_msgs_generate_messages_nodejs
[ 11%] Built target rosgraph_msgs_generate_messages_lisp
[ 11%] Built target roscpp_generate_messages_nodejs
[ 11%] Built target tf_generate_messages_eus
[ 11%] Built target rosgraph_msgs_generate_messages_cpp
[ 11%] Built target pcl_msgs_generate_messages_eus
[ 11%] Built target rosgraph_msgs_generate_messages_eus
[ 11%] Built target dynamic_reconfigure_generate_messages_py
[ 11%] Built target tf_generate_messages_cpp
[ 11%] Built target std_msgs_generate_messages_py
[ 11%] Built target std_msgs_generate_messages_eus
[ 11%] Built target roscpp_generate_messages_lisp
[ 11%] Built target dynamic_reconfigure_generate_messages_lisp
[ 11%] Built target tf2_msgs_generate_messages_eus
[ 11%] Built target nodelet_generate_messages_cpp
[ 11%] Built target dynamic_reconfigure_generate_messages_eus
[ 11%] Built target dynamic_reconfigure_generate_messages_cpp
[ 11%] Built target dynamic_reconfigure_generate_messages_nodejs
[ 11%] Built target geometry_msgs_generate_messages_eus
[ 11%] Built target nodelet_generate_messages_py
[ 11%] Built target roscpp_generate_messages_cpp
[ 11%] Built target std_msgs_generate_messages_lisp
[ 11%] Built target rosgraph_msgs_generate_messages_py
[ 11%] Built target tf_generate_messages_lisp
[ 11%] Built target dynamic_reconfigure_gencfg
[ 11%] Built target bond_generate_messages_py
[ 11%] Built target nodelet_generate_messages_lisp
[ 11%] Built target geometry_msgs_generate_messages_lisp
[ 11%] Built target nodelet_generate_messages_nodejs
[ 11%] Built target bond_generate_messages_cpp
[ 11%] Built target bond_generate_messages_eus
[ 11%] Built target roscpp_generate_messages_eus
[ 11%] Built target topic_tools_generate_messages_eus
[ 11%] Built target topic_tools_generate_messages_cpp
[... 87 lines cut ...]
rgbd_topics.cpp:(.text+0x1aa6): undefined reference to `ros::NodeHandle::getParam(std::string const&, std::string&) const'
rgbd_topics.cpp:(.text+0x1ae8): undefined reference to `ros::NodeHandle::hasParam(std::string const&) const'
rgbd_topics.cpp:(.text+0x1b01): undefined reference to `ros::NodeHandle::getParam(std::string const&, std::string&) const'
rgbd_topics.cpp:(.text+0x1b43): undefined reference to `ros::NodeHandle::hasParam(std::string const&) const'
rgbd_topics.cpp:(.text+0x1b5c): undefined reference to `ros::NodeHandle::getParam(std::string const&, std::string&) const'
CMakeFiles/tests_rgbd_topics.dir/test/rgbd_topics.cpp.o: In function `ros::SubscriptionCallbackHelperT<boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&, void>::deserialize(ros::SubscriptionCallbackHelperDeserializeParams const&)':
rgbd_topics.cpp:(.text._ZN3ros27SubscriptionCallbackHelperTIRKN5boost10shared_ptrIKN11sensor_msgs12PointCloud2_ISaIvEEEEEvE11deserializeERKNS_43SubscriptionCallbackHelperDeserializeParamsE[_ZN3ros27SubscriptionCallbackHelperTIRKN5boost10shared_ptrIKN11sensor_msgs12PointCloud2_ISaIvEEEEEvE11deserializeERKNS_43SubscriptionCallbackHelperDeserializeParamsE]+0x800): undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
CMakeFiles/tests_rgbd_topics.dir/test/rgbd_topics.cpp.o: In function `ros::SubscriptionCallbackHelperT<boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&, void>::deserialize(ros::SubscriptionCallbackHelperDeserializeParams const&)':
rgbd_topics.cpp:(.text._ZN3ros27SubscriptionCallbackHelperTIRKN5boost10shared_ptrIKN11sensor_msgs6Image_ISaIvEEEEEvE11deserializeERKNS_43SubscriptionCallbackHelperDeserializeParamsE[_ZN3ros27SubscriptionCallbackHelperTIRKN5boost10shared_ptrIKN11sensor_msgs6Image_ISaIvEEEEEvE11de [... cut ...]
realsense_camera/CMakeFiles/tests_rgbd_topics.dir/build.make:361: recipe for target '/home/jack/catkin_ws/devel/lib/realsense_camera/tests_rgbd_topics' failed
make[2]: *** [/home/jack/catkin_ws/devel/lib/realsense_camera/tests_rgbd_topics] Error 1
CMakeFiles/Makefile2:493: recipe for target 'realsense_camera/CMakeFiles/tests_rgbd_topics.dir/all' failed
make[1]: *** [realsense_camera/CMakeFiles/tests_rgbd_topics.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
<-------------------------------------------------------------------------------------------------------------->

mdhorn, 2016-07-07:
Looks like you are missing the cv-bridge package:
`sudo apt-get install ros-$ROS_DISTRO-cv-bridge`

What initial ROS package did you install from the [installation directions for ROS](http://wiki.ros.org/kinetic/Installation/Ubuntu)?
desktop?  `sudo apt-get install ros-kinetic-desktop`
base?  `sudo apt-get install ros-kinetic-ros-base`

jb892, 2016-07-07:
I think I have installed the cv-bridge package. I installed Desktop-Full.
```
jack@jack-All-Series:~/catkin_ws$ sudo apt-get install ros-$ROS_DISTRO-cv-bridge[sudo] password for jack: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
ros-kinetic-cv-bridge is already the newest version (1.12.0-1xenial-20160603-161345-0700).
ros-kinetic-cv-bridge set to manually installed.
The following packages were automatically installed and are no longer required:
  account-plugin-twitter checkbox-ng dh-apparmor friends friends-dispatcher
  friends-facebook friends-twitter g++-4.8 gir1.2-ebook-1.2
  gir1.2-ebookcontacts-1.2 gir1.2-edataserver-1.2 gir1.2-gnomebluetooth-1.0
  gir1.2-messagingmenu-1.0 gir1.2-networkmanager-1.0 gnome-icon-theme-symbolic
  gstreamer0.10-nice gstreamer0.10-plugins-good gstreamer0.10-pulseaudio
  gstreamer0.10-x gstreamer1.0-clutter gtk3-engines-unico iproute libbind9-90
  libboost-date-time1.54.0 libboost-system1.54.0 libcamel-1.2-45 libcdr-0.0-0
  libcmis-0.4-4 libcolamd2.8.0 libcolord1 libcolorhug1 libcrypt-passwdmd5-perl
  libdns100 libebackend-1.2-7 libebook-1.2-14 libebook-contacts-1.2-0
  libedata-book-1.2-20 libedataserver-1.2-18 libegl1-mesa-drivers
  libegl1-mesa-lts-vivid libelfg0 libexiv2-12 libfarstream-0.1-0 libfriends0
  libgbm1-lts-vivid libgdata13 libgee2 libgl1-mesa-dri-lts-vivid
  libgl1-mesa-glx-lts-vivid libglapi-mesa-lts-vivid libgles1-mesa-lts-vivid
  libgles2-mesa-lts-vivid libglew1.10 libglewmx1.10 libgnome-bluetooth11
  libgnome-control-center1 libgnome-desktop-3-7 libgphoto2-port10 libgrip0
  libgtop2-7 libicu52 libimobiledevice4 libisc95 libisccc90 libisccfg90
  libisl10 libllvm3.6 liblouis2 liblwres90 libmbim-glib0 libminiupnpc8
  libmspub-0.0-0 libnl-route-3-200 libntdb1 libopenobex2 liborcus-0.6-0
  libparted0debian1 libplist1 libpocketsphinx1 libpoppler44 libprotobuf8
  libqmi-glib0 libqpdf13 libqt5qml-graphicaleffects libqt5sensors5
  libqt5webkit5-qmlwebkitplugin libraw9 librhythmbox-core8 librtmp0
  libsphinxbase1 libsystemd-daemon0 libsystemd-journal0 libsystemd-login0
  libt1-5 libthumbnailer0 libunityvoice1 libupower-glib1 libusbmuxd2
  libvisio-0.0-0 libvpx1 libwnck-common libwnck22 libwpd-0.9-9 libwpg-0.2-2
  libwps-0.2-2 libxatracker2-lts-vivid libxcb-util0 libxtables10
  linux-headers-3.19.0-25 linux-headers-3.19.0-25-generic
  linux-headers-generic-lts-vivid linux-image-3.19.0-25-generic
  linux-image-3.19.0-64-generic linux-image-extra-3.19.0-25-generic
  linux-image-extra-3.19.0-64-generic linux-image-generic-lts-vivid
  linux-signed-image-3.19.0-64-generic linux-signed-image-generic-lts-vivid
  obex-data-server python-commandnotfound python-dbus-dev python-gdbm
  python-gnomekeyring python-ibus python-libxml2 python-ntdb python-pexpect
  python-ptyprocess python-renderpm python-reportlab python-reportlab-accel
  python-requests python-smbc python3-checkbox-ng qml-module-qtquick-dialogs
  qml-module-qtquick-localstorage qml-module-qtquick-privatewidgets
  qml-module-ubuntu-ui-extras-browser qtdeclarative5-dialogs-plugin
  qtdeclarative5-localstorage-plugin qtdeclarative5-privatewidgets-plugin
  qtdeclarative5-qtfeedback-plugin
  qtdeclarative5-ubuntu-ui-extras-browser-plugin
  qtdeclarative5-ubuntu-ui-extras-browser-plugin-assets
  qtdeclarative5-window-plugin rhythmbox-mozilla sphinx-voxforge-hmm-en
  sphinx-voxforge-lm-en telepathy-indicator ubuntu-extras-keyring
  unity-lens-friends unity-scope-audacious unity-scope-clementine
  unity-scope-gmusicbrowser unity-scope-gourmet unity-scope-guayadeque
  unity-scope-musique unity-voice-service xfonts-mathml
  xserver-xorg-dev-lts-vivid xserver-xorg-input-evdev-lts-vivid
  xserver-xorg-input-mouse-lts-vivid xserver-xorg-input-synaptics-lts-vivid
  xserver-xorg-input-vmmouse-lts-vivid xserver-xorg-input-wacom-lts-vivid
  xserver-xorg-video-ati-lts-vivid xserver-xorg-video-cirrus-lts-vivid
  xserver-xorg-video-fbdev-lts-vivid xserver-xorg-video-intel-lts-vivid
  xserver-xorg-video-mach64-lts-vivid xserver-xorg-video-mga-lts-vivid
  xserver-xorg-video-neomagic-lts-vivid xserver-xorg-video-nouveau-lts-vivid
  xserver-xorg-video-openchrome-lts-vivid xserver-xorg-video-r128-lts-vivid
  xserver-xorg-video-radeon-lts-vivid xserver-xorg-video-savage-lts-vivid
  xserver-xorg-video-siliconmotion-lts-vivid
  xserver-xorg-video-sisusb-lts-vivid xserver-xorg-video-tdfx-lts-vivid
  xserver-xorg-video-trident-lts-vivid xserver-xorg-video-vesa-lts-vivid
  xserver-xorg-video-vmware-lts-vivid
Use 'sudo apt autoremove' to remove them.
0 to upgrade, 0 to newly install, 0 to remove and 0 not to upgrade.

```

mdhorn, 2016-07-07:
And you are sourcing /opt/ros/kinetic/setup.bash in your startup file?

echo $ROS_PACKAGE_PATH

jb892, 2016-07-07:
I think I followed the ROS installation guide carefully and I have done sourcing /opt/ros/kinetic/setup.bash already.
```
jack@jack-All-Series:~/catkin_ws$ echo $ROS_PACKAGE_PATH 
/home/jack/catkin_ws/src:/opt/ros/kinetic/share:/opt/ros/kinetic/stacks
```

mdhorn, 2016-07-07:
Well, you can try:
`catkin_make_isolated --pkg realsense_camera`

if that fails, I'm afraid I'll have to send you over to http://answers.ros.org/ for assistance.

jb892, 2016-07-07:
Sadly, it does not work. And most importantly, my previously working packages don't work now. So frustrated. But anyway, thanks very much man! Thanks for your patience and kindness! 
And, what should I do next to get assistance from http://answers.ros.org/ ?

```
jack@jack-All-Series:~/catkin_ws$ sudo apt-get install ros-$ROS_DISTRO-cv-bridgeReading package lists... Done
Building dependency tree       
Reading state information... Done
ros-kinetic-cv-bridge is already the newest version (1.12.0-1xenial-20160603-161345-0700).
0 to upgrade, 0 to newly install, 0 to remove and 0 not to upgrade.
jack@jack-All-Series:~/catkin_ws$ echo $ROS_PACKAGE_PATH 
/home/jack/catkin_ws/src:/opt/ros/kinetic/share:/opt/ros/kinetic/stacks
jack@jack-All-Series:~/catkin_ws$ ^C
jack@jack-All-Series:~/catkin_ws$ catkin_make_isolated --pkg realsense_camera
Base path: /home/jack/catkin_ws
Source space: /home/jack/catkin_ws/src
Build space: /home/jack/catkin_ws/build_isolated
Devel space: /home/jack/catkin_ws/devel_isolated
Install space: /home/jack/catkin_ws/install_isolated
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
~~  traversing 1 packages in topological order:
~~  - realsense_camera
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
The packages or cmake arguments have changed, forcing cmake invocation

==> Processing catkin package: 'realsense_camera'
==> Creating build directory: 'build_isolated/realsense_camera'
==> cmake /home/jack/catkin_ws/src/realsense_camera -DCATKIN_DEVEL_PREFIX=/home/jack/catkin_ws/devel_isolated/realsense_camera -DCMAKE_INSTALL_PREFIX=/home/jack/catkin_ws/install_isolated -G Unix Makefiles in '/home/jack/catkin_ws/build_isolated/realsense_camera'
-- The C compiler identification is GNU 4.9.3
-- The CXX compiler identification is GNU 4.9.3
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/jack/catkin_ws/devel_isolated/realsense_camera
-- Using CMAKE_PREFIX_PATH: /home/jack/catkin_ws/devel;/opt/ros/kinetic
-- This workspace overlays: /home/jack/catkin_ws/devel;/opt/ros/kinetic
-- Found PythonInterp: /usr/bin/python (found version "2.7.11") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/jack/catkin_ws/build_isolated/realsense_camera/test_results
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.1
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:90 (find_package):
  Could not find a package configuration file provided by "librealsense" with
  any of the following names:

    librealsenseConfig.cmake
    librealsense-config.cmake

[... 264 lines cut ...]
camera_core.cpp:(.text.startup+0x2fa): undefined reference to `image_transport::ImageTransport::subscribeCamera(std::string const&, unsigned int, boost::function<void (boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&, boost::shared_ptr<sensor_msgs::CameraInfo_<std::allocator<void> > const> const&)> const&, boost::shared_ptr<void> const&, image_transport::TransportHints const&)'
camera_core.cpp:(.text.startup+0x428): undefined reference to `ros::NodeHandle::NodeHandle(std::string const&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&)'
camera_core.cpp:(.text.startup+0x4eb): undefined reference to `image_transport::ImageTransport::subscribeCamera(std::string const&, unsigned int, boost::function<void (boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&, boost::shared_ptr<sensor_msgs::CameraInfo_<std::allocator<void> > const> const&)> const&, boost::shared_ptr<void> const&, image_transport::TransportHints const&)'
camera_core.cpp:(.text.startup+0x613): undefined reference to `ros::NodeHandle::NodeHandle(std::string const&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&)'
camera_core.cpp:(.text.startup+0x6de): undefined reference to `image_transport::ImageTransport::subscribeCamera(std::string const&, unsigned int, boost::function<void (boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&, boost::shared_ptr<sensor_msgs::CameraInfo_<std::allocator<void> > const> const&)> const&, boost::shared_ptr<void> const&, image_transport::TransportHints const&)'
camera_core.cpp:(.text.startup+0x824): undefined reference to `ros::NodeHandle::NodeHandle(std::string const&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std:: [... cut ...]
make[2]: *** [/home/jack/catkin_ws/devel_isolated/realsense_camera/lib/realsense_camera/tests_camera_core] Error 1
CMakeFiles/Makefile2:2949: recipe for target 'CMakeFiles/tests_camera_core.dir/all' failed
make[1]: *** [CMakeFiles/tests_camera_core.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
<== Failed to process package 'realsense_camera': 
  Command '['make', '-j8', '-l8']' returned non-zero exit status 2

Reproduce this error by running:
==> cd /home/jack/catkin_ws/build_isolated/realsense_camera && make -j8 -l8

Command failed, exiting.

```

mdhorn, 2016-07-07:
I would search for your issue on http://answers.ros.org
http://answers.ros.org/question/234322/linker-error-on-ros-kinetic-ubuntu-1604/

Did you by chance upgrade from 14.04 to 16.04?

Also, what is your version of gcc?
`gcc --version`

jb892, 2016-07-07:
My ubuntu version: 16.04 LTS

gcc (Ubuntu 4.9.3-13ubuntu2) 4.9.3

mdhorn, 2016-07-07:
Based on your answer, I'll assume you did upgrade a system that was 14.04 --> 16.04.

If you read the answer which I previously linked
http://answers.ros.org/question/234322/linker-error-on-ros-kinetic-ubuntu-1604/

You have the wrong version of gcc and need to upgrade it to the default of 5.3.x.

jb892, 2016-07-07:
Thanks man! I just installed clang and replace g++ to clang. It does solve my problem! Thanks a lot! Really appreciate it!

mdhorn, 2016-07-08:
Fixed by PR 64
https://github.com/intel-ros/realsense/pull/64

closed 2016-07-08 by mdhorn

scandrew, 2016-08-18:
I have an identical issue. I am fairly new to ROS. Went through most of the tutorials. But after following the installation steps of librealsense (installing QTCreator, Video4Linux patch...etc.), something broke in regards to catkin. At first I thought it couldn't just make the realsense_camera package, but then I realised even when the camera package is removed from catkin_ws, and just trying to catkin_make my old beginner_tutorials package, that worked without issues prior to this - causes errors. 

I tried substituting gcc for clang, and that built the beginner_tutorials package without errors, but not the realsense_camera package. Still that would have been a temporary solution. I would like to fix catkin.
I tried reinstalling the entire ROS package and doing it all from scratch, but that did not seem to make any difference. 
Should I just reinstall the entire Ubuntu OS, and start from scratch? I would really rather like to try to find a solution instead. In case this happens again in the future.

mdhorn, 2016-08-18:
@scandrew Unfortunately, this Issue listed several different problems. To get help, please do not comment on a previously closed Issue; instead, open a new issue. If you think your problem relates to another closed issue, you can reference that issue.

When you open the new request, please complete the templet to describe your environment, show the command(s) your using, and your results so we have some idea of the nature of your problem.

Where I have seen people have problems is completing all of the sections of Step 3; in your case, you should have only completed '(OR) Kernel in16.04.xx' part of step 3.
Check your version of gcc;
`gcc --version`
`gcc (Ubuntu 5.4.0-6ubuntu1~16.04.2) 5.4.0 20160609`

If you don't have gcc > 5.3, then you'll need to fix that.
`sudo apt-get install gcc-5`
`sudo update-alternatives --config gcc`

scandrew, 2016-08-21:
Thank you for the guidance! I will open a new issue. I think you are right, since I couldn't get my realsense camera to work, I probably went through more steps in patching the kernel than I needed. I have since installed it properly on the other 2 computers, and everything works fine there.
I just have one question for you, since we are on this topic. Is there a way to revert these extra steps, that are now causing the problems, without fully reinstalling Ubuntu? It would be nice if I could clean it, or reinstall a few modules to try fix the problem, otherwise I will be forced to just reinstall the OS and start from scratch.
Thanks,
Andrew

mdhorn, 2016-08-23:
@scandrew If you followed all of the listed instructions, then you probably have a wrong kernel and a the wrong version of gcc.  The kernel should not be a problem, but the version of gcc will be. Basically, just reverse what the script did:

```
sudo update-alternatives --remove-all gcc
sudo apt-get --reinstall install gcc g++
sudo apt-get purge gcc-4.9 g++-4.9
sudo add-apt-repository -r ppa:ubuntu-toolchain-r/test
```

Those commands should restore the default ubuntu 16.04.x version of gcc/g++.