realsenseai/realsense-ros #69: Not able to launch camera due to Error calling rs_enable_stream

miteshkotak, 2016-07-13:
Even  after reinstalling it many times 

when i try to launch realsense camera with the following command 
roslaunch realsense_camera r200_nodelet_default.launch

i get the following error

NODES
  /
    camera (nodelet/nodelet)
    nodelet_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [22679]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to bd721756-48c3-11e6-8aa1-78d00420a772
process[rosout-1]: started with pid [22692]
started core service [/rosout]
process[nodelet_manager-2]: started with pid [22707]
process[camera-3]: started with pid [22710]
[ INFO] [1468391622.192277938]: Initializing nodelet with 8 worker threads.
[ INFO] [1468391623.083983247]: /camera - Detected the following cameras:
				- Serial No: 014150160808; Firmware: 2.60.0.0; Name: Intel RealSense F200; USB Port ID: 2-1
[ INFO] [1468391623.087121962]: /camera - Connecting to camera with Serial No: 014150160808 USB Port ID: 2-1
[ INFO] [1468391623.089646350]: /camera - Enabling Color stream: manual mode
[ INFO] [1468391623.092237652]: /camera - Enabling Depth stream: manual mode
[ INFO] [1468391623.094707818]: /camera - Enabling IR stream: manual mode
[ INFO] [1468391623.097742781]: /camera - Enabling IR2 stream: manual mode

[ERROR] [1468391623.098017013]: /camera - Error calling rs_enable_stream ( device:0x21f2908, stream:INFRARED2, width:640, height:480, format:Y8, framerate:30 ): 
unsupported stream 

[FATAL] [1468391623.272767279]: Failed to load nodelet '/camera` of type `realsense_camera/R200Nodelet` to manager `nodelet_manager'
[nodelet_manager-2] process has died [pid 22707, exit code 1, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=nodelet_manager __log:=/home/acfr/.ros/log/bd721756-48c3-11e6-8aa1-78d00420a772/nodelet_manager-2.log].
log file: /home/acfr/.ros/log/bd721756-48c3-11e6-8aa1-78d00420a772/nodelet_manager-2*.log
[camera-3] process has died [pid 22710, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load realsense_camera/R200Nodelet nodelet_manager camera/depth/image_raw:=camera/depth/image_raw camera/color/image_raw:=camera/color/image_raw camera/ir/image_raw:=camera/ir/image_raw camera/ir2/image_raw:=camera/ir2/image_raw camera/depth/points:=camera/depth/points camera/depth/camera_info:=camera/depth/camera_info camera/color/camera_info:=camera/color/camera_info camera/ir/camera_info:=camera/ir/camera_info camera/ir2/camera_info:=camera/ir2/camera_info camera/get_settings:=camera/get_settings __name:=camera __log:=/home/acfr/.ros/log/bd721756-48c3-11e6-8aa1-78d00420a772/camera-3.log].
log file: /home/acfr/.ros/log/bd721756-48c3-11e6-8aa1-78d00420a772/camera-3*.log**

### System Configuration
Please complete Your Configuration detail below. Refer to the BKC at [Configuration](../realsense_camera/README.md#configuration).

| Version          | Your Configuration   |
|:---------------- |:---------------------|
| Operating System |ubutnu 14.04.4               |
| Kernel           | 4.2.0                |
| Backend          | ?                    |
| ROS              | indigo                   |
| ROS RealSense    | F200                |
| librealsense     | 0.9.1                |
| R200 Firmware    | 2.71              |

---
#### How to collect Configuration Data
*This section can be deleted before submission.*

| Version          | Method |
|:---------------- |:------------ |
| Operating System | `grep DISTRIB_DESCRIPTION /etc/*elease*` |
| Kernel           | `uname -r` |
| Backend          | `ls /sys/class | grep video4linux` |
| ROS              | `rosversion -d` |
| ROS RealSense    | `rosversion realsense_camera` |
| librealsense     | `cat <path to librealsense>/librealsense/readme.md | grep release-image | awk -F- '{print $3}'` |
| R200 Firmware    | View the ROS log from running nodelet **OR** `<path to librealsense>/librealsense/bin/cpp-enumerate-devices | grep -i firmware` |

---

### Expected Behavior

### Actual Behavior

### Steps to Reproduce

rjingar, 2016-07-13:
Current release supports R200 camera only. The error you are getting is because F200 has only one Infrared camera. Nodelet functionalities and supported camera settings are for R200 camera only.

Future releases will have support added for F200 and SR300.

miteshkotak, 2016-07-13:
ok thanks  rjingar

closed 2016-07-13 by miteshkotak