realsenseai/realsense-ros #78: Error: run r200_nodelet_rgbd.lauch

jb892, 2016-07-28:
### System Configuration

| Version          | Your Configuration   |
|:---------------- |:---------------------|
| Operating System |   Ubuntu 16.04 LTS     |
| Kernel           |  4.4.0-040400-generic            |
| Backend          |                    |
| ROS              | kinetic       |
| ROS RealSense    |   Cannot locate [realsense_camera]   |
| librealsense     |      0.9.1         |
| R200 Firmware    |   1.0.72.06      |

Hi! I just got these errors when I was trying to run r200_nodelet_rgbd.lauch.
Please help! Thanks!

```
jack@jack-All-Series:~/catkin_ws$ roslaunch realsense_camera r200_nodelet_r
r200_nodelet_resolution.test  r200_nodelet_rgbd.test
r200_nodelet_rgbd.launch      
jack@jack-All-Series:~/catkin_ws$ roslaunch realsense_camera r200_nodelet_rgbd.launch 
... logging to /home/jack/.ros/log/f7e33cf8-5444-11e6-86bf-086266340963/roslaunch-jack-All-Series-12936.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://jack-All-Series:41003/

SUMMARY
========

PARAMETERS
 * /camera/camera/base_frame_id: camera_link
 * /camera/camera/color_fps: 30
 * /camera/camera/color_frame_id: camera_rgb_frame
 * /camera/camera/color_height: 480
 * /camera/camera/color_optical_frame_id: camera_rgb_optica...
 * /camera/camera/color_width: 640
 * /camera/camera/depth_fps: 30
 * /camera/camera/depth_frame_id: camera_depth_frame
 * /camera/camera/depth_height: 360
 * /camera/camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/camera/depth_width: 480
 * /camera/camera/enable_color: True
 * /camera/camera/enable_depth: True
 * /camera/camera/enable_pointcloud: False
 * /camera/camera/enable_tf: True
 * /camera/camera/ir2_frame_id: camera_ir2_frame
 * /camera/camera/ir_frame_id: camera_ir_frame
 * /camera/camera/mode: manual
 * /camera/camera/serial_no: 
 * /camera/camera/usb_port_id: 
 * /camera/camera_nodelet_manager/num_worker_threads: 4
 * /camera/depth_rectify_depth/interpolation: 0
 * /camera/disparity_depth/max_range: 4.0
 * /camera/disparity_depth/min_range: 0.5
 * /camera/disparity_registered_sw/max_range: 4.0
 * /camera/disparity_registered_sw/min_range: 0.5
 * /camera/points_xyzrgb_sw_registered/queue_size: 100
 * /rosdistro: kinetic
 * /rosversion: 1.12.2

NODES
  /camera/
    camera (nodelet/nodelet)
    camera_nodelet_manager (nodelet/nodelet)
    debayer (nodelet/nodelet)
    depth_metric (nodelet/nodelet)
    depth_metric_rect (nodelet/nodelet)
    depth_points (nodelet/nodelet)
    depth_rectify_depth (nodelet/nodelet)
    depth_registered_sw_metric_rect (nodelet/nodelet)
    disparity_depth (nodelet/nodelet)
    disparity_registered_sw (nodelet/nodelet)
    points_xyzrgb_sw_registered (nodelet/nodelet)
    rectify_color (nodelet/nodelet)
    rectify_ir (nodelet/nodelet)
    rectify_mono (nodelet/nodelet)
    register_depth_rgb (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[camera/camera_nodelet_manager-1]: started with pid [12954]
process[camera/camera-2]: started with pid [12955]
process[camera/debayer-3]: started with pid [12956]
process[camera/rectify_mono-4]: started with pid [12957]
process[camera/rectify_color-5]: started with pid [12974]
process[camera/rectify_ir-6]: started with pid [12983]
process[camera/depth_rectify_depth-7]: started with pid [13003]
[ INFO] [1469709228.472179466]: Initializing nodelet with 4 worker threads.
process[camera/depth_metric_rect-8]: started with pid [13025]
process[camera/depth_metric-9]: started with pid [13051]
process[camera/depth_points-10]: started with pid [13060]
process[camera/register_depth_rgb-11]: started with pid [13081]
process[camera/points_xyzrgb_sw_registered-12]: started with pid [13097]
process[camera/depth_registered_sw_metric_rect-13]: started with pid [13114]
process[camera/disparity_depth-14]: started with pid [13117]
process[camera/disparity_registered_sw-15]: started with pid [13127]
[ERROR] [1469709229.063014833]: Failed to load nodelet [/camera/camera] of type [realsense_camera/R200Nodelet] even after refreshing the cache: Failed to load library /home/jack/catkin_ws/devel/lib//librealsense_camera_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /home/jack/catkin_ws/devel/lib//librealsense_camera_nodelet.so: undefined symbol: _ZN12class_loader20class_loader_private30getCurrentlyLoadingLibraryNameEv)
[ERROR] [1469709229.063055053]: The error before refreshing the cache was: Failed to load library /home/jack/catkin_ws/devel/lib//librealsense_camera_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /home/jack/catkin_ws/devel/lib//librealsense_camera_nodelet.so: undefined symbol: _ZN12class_loader20class_loader_private30getCurrentlyLoadingLibraryNameEv)
[FATAL] [1469709229.063154743]: Service call failed!
[camera/camera-2] process has died [pid 12955, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load realsense_camera/R200Nodelet camera_nodelet_manager camera/depth/image_raw:=depth/image_raw camera/color/image_raw:=rgb/image_raw camera/ir/image_raw:=ir/image_raw camera/ir2/image_raw:=ir2/image_raw camera/depth/points:=depth/points camera/depth/camera_info:=depth/camera_info camera/color/camera_info:=rgb/camera_info camera/ir/camera_info:=ir/camera_info camera/ir2/camera_info:=ir2/camera_info __name:=camera __log:=/home/jack/.ros/log/f7e33cf8-5444-11e6-86bf-086266340963/camera-camera-2.log].
log file: /home/jack/.ros/log/f7e33cf8-5444-11e6-86bf-086266340963/camera-camera-2*.log

```

mdhorn, 2016-07-29:
Did you follow the [librealsesne installation instructions](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md)?

Are you using the recommended Video4Linux backend? None list listed.

The version of librealsense you listed is too old; you need to upgraded to v0.9.2.
This is likely the reason for the undefined library symbol.
In your librealsense github local repo, update to version v0.9.2:
`git pull`
`git checkout -B branch_v0.9.2 v0.9.2`
`make && sudo make install`

There will be production packages released soon which will avoid this setup overhead.

LorenzMeier, 2016-08-02:
@mdhorn Just ran into the same issue.

LorenzMeier, 2016-08-02:
@mdhorn Your suggested fix does not fix this. Do you have an Ubuntu 16.04 system with ROS Kinetic that you can test on? A lot of the documentation is generally also outdated or outdated for 16.04.

LorenzMeier, 2016-08-02:
Error in my case:

```sh
[ERROR] [1470149601.091958711]: Failed to load nodelet [/camera] of type [realsense_camera/R200Nodelet] even after refreshing the cache: Failed to load library /home/pixhawk/catkin_ws/devel/lib//librealsense_camera_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /home/pixhawk/catkin_ws/devel/lib//librealsense_camera_nodelet.so: undefined symbol: _ZNK3ros10NodeHandle8hasParamERKSs)
[ERROR] [1470149601.092001623]: The error before refreshing the cache was: Failed to load library /home/pixhawk/catkin_ws/devel/lib//librealsense_camera_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /home/pixhawk/catkin_ws/devel/lib//librealsense_camera_nodelet.so: undefined symbol: _ZNK3ros10NodeHandle8hasParamERKSs)
[FATAL] [1470149601.092118211]: Service call failed!
[camera-3] process has died [pid 21593, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load realsense_camera/R200Nodelet nodelet_manager camera/depth/image_raw:=camera/depth/image_raw camera/color/image_raw:=camera/color/image_raw camera/ir/image_raw:=camera/ir/image_raw camera/ir2/image_raw:=camera/ir2/image_raw camera/depth/points:=camera/depth/points camera/depth/camera_info:=camera/depth/camera_info camera/color/camera_info:=camera/color/camera_info camera/ir/camera_info:=camera/ir/camera_info camera/ir2/camera_info:=camera/ir2/camera_info camera/get_settings:=camera/get_settings __name:=camera __log:=/home/pixhawk/.ros/log/da858370-58c0-11e6-b2f8-4485007315ea/camera-3.log].
```

LorenzMeier, 2016-08-02:
I noticed that the Intel install instructions for librealsense mess with the GCC version installed on the system and its not entirely clear which version is required. I'm investigating if this is the core reason the package breaks.

mdhorn, 2016-08-02:
Are are building the code in Continuous Integration for both Indigo and Kinetic as well as build for both in the ROS Build Farm. We also do full regression testing under both Indigo and Kinetic before an official release is made.

I'm open to improving the documentation if you see some part that isn't covering both 14.04 for Indigo and 16.04 for Kinetic.

In the librealsense directions, you should be using the compiler upgrade steps if you are on Kinetic, just for Indigo under 14.04.
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#video4linux-backend
In Step 3 you would only follow the instructions for 16.04 "Kernel in 16.04.xx"

If you are still having this problem, can you please provide all of the requested data for your configuration

| Version          | Method |
|:---------------- |:------------ |
| Operating System | `grep DISTRIB_DESCRIPTION /etc/*elease*` |
| Kernel           | `uname -r` |
| Backend          | `ls /sys/class | grep video4linux` |
| ROS              | `rosversion -d` |
| ROS RealSense    | `rosversion realsense_camera` |
| librealsense     | `cat <path to librealsense>/librealsense/readme.md | grep release-image | awk -F- '{print $3}'` |
| R200 Firmware    | View the ROS log from running nodelet **OR** `<path to librealsense>/librealsense/bin/cpp-enumerate-devices | grep -i firmware` |

And the commands you used to both build and then launch the node.
There has to be something difference we aren't seeing when it builds for many people and the build farm, so helping identifying this difference will be good.

jb892, 2016-08-03:
Hi @mdhorn ,

Thanks for your reply! I tried the command you provided to update my librealsense. However, I failed to update. Could you please give me a more detailed setup guide? Thanks very much! 
Bo.
```
jack@jack-All-Series:~/Documents/librealsense$ git pull
Already up-to-date.
jack@jack-All-Series:~/Documents/librealsense$ git checkout -B branch_v0.9.2 v0.9.2
M	librealsense.qtcreator/c-tutorial-1-depth.pro
M	librealsense.qtcreator/cpp-capture.pro
M	librealsense.qtcreator/cpp-pointcloud.pro
Switched to a new branch 'branch_v0.9.2'
jack@jack-All-Series:~/Documents/librealsense$ make && sudo make install
mkdir -p obj/libuvc
mkdir -p lib
mkdir -p bin
g++ src/uvc-libuvc.cpp -std=c++11 -fPIC -pedantic -Ofast -Wno-missing-field-initializers -Wno-switch -Wno-multichar -DRS_USE_V4L2_BACKEND `pkg-config --cflags --libs libusb-1.0` -mssse3 -c -o obj/uvc-libuvc.o
/bin/sh: 1: g++: not found
Makefile:98: recipe for target 'obj/uvc-libuvc.o' failed
make: *** [obj/uvc-libuvc.o] Error 127
```

mdhorn, 2016-08-03:
@jb892 The current error shows g++ not found.
I'm not sure how you previously built the librealsense and the camera module without g++, but please run these steps:
``sudo apt-get update && sudo apt-get upgrade``
``sudo apt-get install libusb-1.0-0-dev libglfw3-dev g++``

Did you already follow the required librealsense instructions?
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

Thomas00010111, 2016-09-02:
Hi @mdhorn,
I am running into the same problem on my PC, on my laptop I can use my r200 camera with ros. Here is information you asked for:

robot@robot-S5520SC:~/SourceCode/librealsense$ git clone https://github.com/IntelRealSense/librealsense.git
cd librealsense
git pull
git checkout -B branch_v0.9.2 v0.9.2
Then steps from master/doc/installation.md. ( I had t add this line to ./scripts/patch-uvcvideo-4.4.sh: CONFIG_LOCATION=/usr/src/linux-headers-$(uname -r) )

robot@robot-S5520SC:~/SourceCode/librealsense$ grep DISTRIB_DESCRIPTION /etc/*elease*
/etc/lsb-release:DISTRIB_DESCRIPTION="Ubuntu 14.04.5 LTS"
robot@robot-S5520SC:~/SourceCode/librealsense$ uname -r
4.4.0-040400-generic
robot@robot-S5520SC:~/SourceCode/librealsense$ ls /sys/class | grep video4linux
video4linux
robot@robot-S5520SC:~/SourceCode/librealsense$ rosversion -d
indigo
robot@robot-S5520SC:~/SourceCode/librealsense$ rosversion realsense_camera
1.4.0
robot@robot-S5520SC:~/SourceCode/librealsense$ cat /home/robot/SourceCode/librealsense/readme.md | grep release-image | awk -F- '{print $3}'

0.9.1
robot@robot-S5520SC:~/SourceCode/librealsense$ cat /home/robot/SourceCode/librealsense/bin/cpp-enumerate-devices | grep -i firmware
Binary file (standard input) matches

And the error I get when calling roslaunch realsense_camera r200_nodelet_default.launch is:
[ERROR] [1472774016.851459441]: Failed to load nodelet [/camera] of type [realsense_camera/R200Nodelet] even after refreshing the cache: MultiLibraryClassLoader: Could not create object of class type realsense_camera::R200Nodelet as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[ERROR] [1472774016.851543889]: The error before refreshing the cache was: Failed to load library /home/robot/catkin_ws/devel/lib//librealsense_camera_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /home/robot/catkin_ws/devel/lib//librealsense_camera_nodelet.so: undefined symbol: rs_is_stream_enabled)
[FATAL] [1472774016.851864496]: Failed to load nodelet '/camera` of type `realsense_camera/R200Nodelet` to manager `nodelet_manager'
[camera-3] process has died [pid 24367, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load realsense_camera/R200Nodelet nodelet_manager camera/depth/image_raw:=camera/depth/image_raw camera/color/image_raw:=camera/color/image_raw camera/ir/image_raw:=camera/ir/image_raw camera/ir2/image_raw:=camera/ir2/image_raw camera/depth/points:=camera/depth/points camera/depth/camera_info:=camera/depth/camera_info camera/color/camera_info:=camera/color/camera_info camera/ir/camera_info:=camera/ir/camera_info camera/ir2/camera_info:=camera/ir2/camera_info camera/get_settings:=camera/get_settings __name:=camera __log:=/home/robot/.ros/log/4c1b2dbe-709f-11e6-b4a9-001517fd6d9c/camera-3.log].
log file: /home/robot/.ros/log/4c1b2dbe-709f-11e6-b4a9-001517fd6d9c/camera-3*.log

Regards,
Thomas

mdhorn, 2016-09-02:
@Thomas00010111  
There is a conflict between those versions of librealsense and realsense_camera.

Since you've followed the steps for updating the kernel for librealsense, please try using just the prebuilt, release ROS packages.

Open a new terminal, and do not source your local catkin workspace. Check that you do not attempt to source your workspace in your shell startup files (i.e. .bashrc or .profile).

Only source the system setup file:
`source /opt/ros/indigo/setup.bash`
Verify this by checking the ROS_PACKAGE_PATH
`echo $ROS_PACKAGE_PATH`
`    /opt/ros/indigo/share:/opt/ros/indigo/stacks`

Install the released RealSense camera package:
`sudo apt-get update`
`sudo apt-get install ros-indigo-realsense-camera`
This will install both realsense camera and librealsense into /opt/ros/indigo

Now verify that you have the released versions installed:
`rosversion librealsense`
`0.9.2`
`rosversion realsense_camera`
`1.3.0`

If you need the latest version of realsense_camera (version 1.4.0 for the SR300 support), then you can a) build from source, or b) use the release candidate from the [Shadow Repository|(http://wiki.ros.org/ShadowRepository).

If you build from sources, be sure NOT to have librealsense in your catkin workspace or you could face the version conflict again.
If you use the Shadow Repository, be sure to remove it when you are done as you will get unexpected results in the future in getting release candidate packages for all of ROS.

Thomas00010111, 2016-09-02:
Hi @mdhorn,
Thank you very much for your fast reply! It worked!
I guess I can know delete the realsense directory from my catkin_ws as well as the librealsense directory (plus /usr/local/lib/lirbrealsense.so)?

Thanks,
Thomas

mdhorn, 2016-09-02:
@Thomas00010111  Yes, you do not need the catkin workspace version of librealsense, nor do you need the /usr/local/lib/librealsense.so, /usr/local/include/librealsense -- they can be purged.

closed 2016-09-02 by mdhorn

Thomas00010111, 2016-09-02:
I think I solved it by deleting /usr/locale/lib/librealsense.so.

Thanks,
Thomas