realsenseai/realsense-ros #82: Cannot see pointcloud in 1.3.0

robertxchen66, 2016-08-04:
| Version          | Your Configuration   |
|:---------------- |:---------------------|
| Operating System | Ubuntu 14.04.4 LTS              |
| Kernel           | 4.4.0-040400-generic                |
| Backend          | video4linux                 |
| ROS              | indigo                    |
| ROS RealSense    | 1.3.0; 1.0.3               |
| librealsense     | ?.?.?                |
| R200 Firmware    | 1.0.72.06          |

---

I installed 1.3.0 version of ROS realsense and tried to run rviz, I can see all four images, but I cannot see the pointcloud. There was no error in the rviz. But under the PointCloud2->Status, the item Transform [...] was missing. 
When I reverted to 1.0.3 version of ROS realsense, everything was OK. I can see the pointcloud, and the Transform[...] item was there.
 Can someone help to tell me what was wrong?

reaganlo, 2016-08-05:
@robertxchen66 Due to performance issues, native pointcloud generation has been disabled by default. You have 2 options:

1. If you want generate the native pointcloud, set the "enable_pointcloud" parameter to true in your launch file and view the topic "/camera/depth/points". Refer to [r200_nodelet_modify_params.launch] (https://github.com/intel-ros/realsense/blob/1.3.0/realsense_camera/launch/r200_nodelet_modify_params.launch) to see how its done.
2. If you want to generate pointcloud using depth_image_proc, use [r200_nodelet_rgbd.launch] (https://github.com/intel-ros/realsense/blob/1.3.0/realsense_camera/launch/r200_nodelet_rgbd.launch) and view the topic "/camera/depth_registered/points".

robertxchen66, 2016-08-05:
Thanks. I will try that.

what kind of performance issue are you talking about?

reaganlo, 2016-08-05:
@robertxchen66 Sometimes, at higher resolutions, the native pointcloud drops the fps of other streams like depth, color etc. We are working on alternate solutions/design. 

Please let us know if the above mentioned options worked for you and update the status of the issue accordingly. Thanks.

robertxchen66, 2016-08-06:
@reaganlo I ran the r200_nodelet_navigation.launch, and I can see the pointcloud.
I use a Gen 6 NUC with 16GB of memory. CPU utilization is around 60%. Not terrible. Memory usage about 1.7GB.

reaganlo, 2016-08-09:
Great. Thanks for the update. Closing the issue.

closed 2016-08-09 by reaganlo