realsenseai/realsense-ros #84: Any way to increase the TF frame rate for the camera topics?

gabrielarpino, 2016-08-07:
Hi, 

Just posting here a quick question on how to change the TF frame rate for the camera topics (increase the frame rate), because I get the following Warnings when running Realsense R200 with RTAB:

[ WARN] [1470216120.812503232, 1470155297.576390646]: rtabmapviz: Could not get transform from odom to camera_depth_optical_frame after 0.200000 seconds (for stamp=1470155297.151934)!

Let me know if there is any parameter i could change or service I could call that might fix this problem by increasing the TF frame publication rate of the camera. Thanks!

Gabe

mdhorn, 2016-08-11:
Sorry for the delayed answer.

We don't have a way to change the rate on the TF, but have fixed this issue in the latest code on the branch ('indigo-devel') by changing this to a Static Transform. The data isn't dynamic, and we probably should have just used a Static TF from the beginning.
https://github.com/intel-ros/realsense/pull/79

You can build the code yourself from the 'tip' of indigo-devel for indigo or kinetic, or I'm attaching a Debian packages built from the latest code as we won't a have a new released package for a week or more. We need to fix at least one more open Issue before making an official release.
[debian_pkgs.zip](https://github.com/intel-ros/realsense/files/413834/debian_pkgs.zip)
`unzip debian_pkgs.zip`
`sudo dpkg -i ros-kinetic-realsense-camera_1.3.99-0xenial_amd64.deb`

Let me know if this resolve your problem.

mdhorn, 2016-08-20:
Static transform will be the default with the release of 1.4.0 next week.

closed 2016-09-21 by mdhorn