realsenseai/realsense-ros #85: Provide a mechanism for stopping the camera

tulku, 2016-08-08:
Hi!

For long running applications, it would be nice to offer an option to turn of the device when it's not being in use (for example, the robot is charging and does not need to pay attention for obstacles). This can help increase the lifetime of the device and also reduce power consumption.

librealsense provides the disable_stream() methods, which can be access via the stream flags in the ros driver, and it's also possible to turn of the projector. However, it would be nice to have a service that can turn the camera on/off, using the start() / stop() calls and also (if needed) turn of the projector and streams.

### System Configuration
Please complete Your Configuration detail below. Refer to the BKC at [Configuration](../realsense_camera/README.md#configuration).

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      |  ubuntu 16.04                |
| Kernel                | 4.4.0                |
| Backend               | v4l                    |
| ROS                   | kinetic                    |
| ROS RealSense         | latest                |
| librealsense          | latest                |
| Camera Type-Firmware  | r200-1.0.72.06            |

ghindman, 2016-08-09:
Hi @tulku we'll need to give this some thought since we would need to ensure that all subscribers were ready for the camera to be turned off.  We'll add this to our backlog.

rjingar, 2016-08-31:
Hi @tulku 
Here are some initial thoughts on how we are planning to provide the services to start and stop the camera.

Services to turns On/Off the camera. 
/camera/set_power <true | false>
bool power_on         # true - to power on camera, false - to power off camera
---
bool success          # false - only if camera has subscribers, and attempted power off

/camera/force_power <true | false>    # to forcefully turn off the camera regardless of subscribers
bool power_on         # true - to power on camera, false - to power off camera
---
# No Result

Ex: 
•Start the Camera :
Rosservice call /camera/set_power true
•To stop the camera only if number of subscribers are zero:
rosservice call /camera/power_on false
•To forcefully stop the camera regardless of subscribers: 
rosservice call /camera/force_power false

These services won't change configuration for the streams. So if camera is restarted (stopped and started again) using above services, nodelet retains stream/projector configs (enable/disable etc.) and camera resumes with the same stream/projector config.

Service to check the camera status:
/camera/is_powered
# No Arguments
---
bool is_powered       # true - camera is powered, false - camera is not powered

Please let us know your comments on this or of you have any other suggestions.

tulku, 2016-08-31:
Hi! The described interface sounds very good. Just one question, on node startup, the camera will start powered on, right?

rjingar, 2016-08-31:
Yes. On node startup, camera will start powered on.

mdhorn, 2016-09-22:
Currently available in the tip of the branch.
Will be included in the upcoming 1.5.0 release.

mdhorn, 2016-09-29:
Fixed by release 1.5.0
https://github.com/intel-ros/realsense/releases/tag/1.5.0

closed 2016-09-29 by mdhorn

CameronDevine, 2018-11-02:
Are these services still supported on release 2.0.0?