realsenseai/realsense-ros #88: Hardware requirement for multiple realsenses

Severn-Vergil, 2016-08-13:
My notebook has one USB 3.0 hub referring the result of lsusb, can it work with multiple realsenses? I hope that the bandwidth is enough for that. Thank you advance for your reply!

lsusb result:
Bus 002 Device 002: ID 8087:8000 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8008 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 005: ID 04f2:b3d8 Chicony Electronics Co., Ltd 
Bus 003 Device 006: ID 04ca:2006 Lite-On Technology Corp. Broadcom BCM43142A0 Bluetooth Device
Bus 003 Device 054: ID 1d57:0008 Xenta 
Bus 003 Device 053: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 003 Device 052: ID 1a40:0801 Terminus Technology Inc. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

reaganlo, 2016-08-19:
@Severn-Vergil Yes, you can use multiple realsense cameras connected to a USB 3.0 hub.
Are you using the **realsense ROS package** or just using **librealsense**?
For librealsense related questions please post them at https://github.com/IntelRealSense/librealsense

ScottJohnson2718, 2016-10-11:
I have a laptop connected to a powered USB3 hub with two R200s connected to the powered hub.  It works well in the librealsense binary sample called cpp-multicam.  But with the ROS setup, the second camera stops publishing depth to /camera2/depth/raw_image after a few seconds as seen from RViz and from "rostopic hz /camera2/depth/raw_image". 
Bandwidth:
Camera 1:
Color : 55 MB/s
Depth : 21 MB/s
ir : 10 MB/s
ir2 : 10 MB/s
Camera 2:
Color : 55 MB/s
Depth : (not showing up)
ir : (not showing up)
ir2 : (not showing up)

If I unplug the cameras and try again I get depth from camera2 for a short time.

I am fairly new to ROS so I am wondering how I can intelligently alter my configuration to reduce bandwidth to see if that helps.

reaganlo, 2016-10-31:
@ScottJohnson2718 @Severn-Vergil Could you please provide your hardware specifications (CPU and Memory)? Also, are you using the launch file similar to the one we have provide [Multiple Camera launch](https://github.com/intel-ros/realsense/blob/indigo-devel/realsense_camera/launch/r200_nodelet_multiple_cameras.launch)

reaganlo, 2017-02-13:
No response. So closing the issue.

closed 2017-02-13 by reaganlo