realsenseai/realsense-ros #92: Multiple cameras can only be started from a single launch file for F200/SR300

mdhorn, 2016-08-19:
### System Configuration

| Version               | Method |
|:--------------------- |:------------ |
| Operating System      | Ubuntu 14.04.5 LTS or  Ubuntu 16.04.1 LTS |
| Kernel                | 4.4.0-34-generic |
| Backend               | video4linux |
| ROS                   | indigo or kinetic |
| ROS RealSense         | 1.3.0 and 1.4.0 |
| librealsense          | 0.9.2 |
| Camera Type-Firmware  | F200-2.60.0.0 and SR300-3.10.10.0|

---

### Expected Behavior
Multiple cameras can be started from independent launch files.

### Actual Behavior
For SR300 or F200, multiple cameras can only be started from a single launch file.

Create 2 launch files with valid serial number of each connected SR300 (or F200) camera. NOTE: serial numbers must be used as usb-device-id are current non-functional for F200/SR300.
The 1st file launched successfully, but the second launch file results in the following:
[ERROR] [1470647138.860273648]: /camera2/camera2 - Error calling rs_create_context ( api_version:4 ):
libusb_claim_interface(...) returned LIBUSB_ERROR_BUSY

### Steps to Reproduce
```
roslaunch realsense_camera sr300_nodelet_camera1.launch
roslaunch realsense_camera sr300_nodelet_camera2.launch
```

severin-lemaignan, 2017-02-22:
I'm hitting this bug with 2 SR300 cameras started from one single launch file that *include* twice the ` sr300_nodelet_rgbd.launch`  launch file (with 2 different serial no).

What do you mean exactly when you say that the workaround is to use a single launch file? Do you mean one launch file that include twice `nodelet_rgbd.launch.xml`?

mdhorn, 2017-02-22:
@severin-lemaignan Yes, you can refer to the R200 Multi-camera launch file as an example:
https://github.com/intel-ros/realsense/blob/indigo-devel/realsense_camera/launch/r200_nodelet_multiple_cameras.launch

rljordan, 2017-02-22:
Will not fix.

closed 2017-02-22 by rljordan