realsenseai/realsense-ros #101: F200 fails to be be launched

ljklonepiece, 2016-09-08:
### System Configuration
Please complete Your Configuration detail below. Refer to the BKC at [Configuration](../realsense_camera/README.md#configuration).

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      |  Ubuntu 14.04 LTS               |
| Kernel                | 4.4.0-040400-generic              |
| Backend               | video4linux                    |
| ROS                   | Indigo                    |
| ROS RealSense         |        1.4.0         |
| librealsense          | 0.9.2               |
| Camera Type-Firmware  | 2.6.0           |

### Expected Behavior
expect the driver works and can see the depth image with rviz

### Actual Behavior
[FATAL] [1473317150.170529596]: Failed to load nodelet '/camera` of type `realsense_camera/F200Nodelet` to manager `nodelet_manager'
[nodelet_manager-1] process has died [pid 14595, exit code -11, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=nodelet_manager __log:=/home/juekun/.ros/log/62e4c552-758e-11e6-926b-f0921cf33d62/nodelet_manager-1.log].

### Steps to Reproduce
roslaunch realsense_camera f200_nodelet_default.launch

### additional info
point clouds can be observed using ./cpp-tutorial-3-pointcloud from librealsense/bin/

reaganlo, 2016-10-28:
@ljklonepiece Were you able to get this to work? We were not able to reproduce this issue.

Try deleting the "devel" and "build" folder in your catkin_ws. Then do a "catkin_make clean" and build the package again using "catkin_make". Also, please ensure that your LD_LIBRARY_PATH is pointing to the location where the librealsense.so file is installed.

reaganlo, 2017-02-13:
No response. Closing the issue.

closed 2017-02-13 by reaganlo