realsenseai/realsense-ros #102: Possible bug in decision when to get new color image

NikolasE, 2016-09-08:
I'm trying to setup an F200 and while I still have driver problems, I had a look at the code and found l. 713 in base_nodelet.cpp:

`        if (pointcloud_publisher_.getNumSubscribers() > 0 &&
            rs_is_stream_enabled(rs_device_, RS_STREAM_DEPTH, 0) == 1 && enable_pointcloud_ == true &&
            (duplicate_depth_color_ == false)) // Skip publishing PointCloud if Depth and/or Color frame was duplicate
        {
          if (camera_publisher_[RS_STREAM_DEPTH].getNumSubscribers() <= 0)
          {
            getStreamData(RS_STREAM_DEPTH);
          }
          if (camera_publisher_[RS_STREAM_COLOR].getNumSubscribers() <= 0)
          {
            getStreamData(RS_STREAM_COLOR);
          }
          publishPCTopic();
        }`

getStreamData(RS_STREAM_COLOR) fills image_[RS_STREAM_COLOR] which is read in publishPCTopic to create a colored pointcloud. So a new color image should always be captured even. I especially don't understand the '<= 0' part. Shouldn't an image get pulled if we have at least on subscriber (for the color image OR the colored pointcloud)?

rjingar, 2016-09-12:
Hi @NikolasE 

Nodelet publishes topics/streams only if there is a subscriber. So if there is no subscriber to the color stream, getStreamData(RS_STREAM_COLOR) will not be called. In this case if user subscribes to pointcloud topic, image_[RS_STREAM_COLOR] needs to be filled by calling getStreamData if no color data is already available. So '<=0' check in l.713 covers the case when there is no subscriber to color stream but a subscriber to pointcloud.

rjingar, 2016-10-28:
Closing due to no response.

closed 2016-10-28 by rjingar