realsenseai/realsense-ros #103: power on/off service for multiple cameras using the same nodelet manager.

y27choi, 2016-09-09:
Hi,

Just a quick question about using service call to power on/off the cameras. How does the service call work when there are multiple cameras running on the same nodelet manager?

It looks like the both camera try to advertise to /camera/force_power. Is it possible to have the service call per nodelet?

rjingar, 2016-09-09:
Hi @y27choi 

Good catch. remapping needs to be added in "nodelet.launch.xml" file for set_power, force_power and is_powered services.
I have created pullrequest (#104) to add multi-camera support for these services. 
For now, you can get required changes from the pullrequest or from "rjingar:set-power-multi-cam-support".

Thanks!

mdhorn, 2016-10-28:
Fix in release 1.5.0
https://github.com/intel-ros/realsense/releases/tag/1.5.0

closed 2016-10-28 by mdhorn