realsenseai/realsense-ros #105: Simultaneaous ROS and UVC webcam access

doisyg, 2016-09-12:
Hello,
I was wondering if there is a way to access the realsense color stream as a UVC webcam (for native support in browser) in the same time as using the ROS realsense node.
It works when the ROS node is off, but when it is on, of course both process can't access it isimultaneously. Any clue how to bypass this. Maybe there is a way of taking the image_raw ros topic and republishing it somehow?

mdhorn, 2016-09-13:
The device is locked to one process at a time. When the ROS node is running, it has control of the video device.
You'll need to write a ROS node that subscribes to the image topic and then sends it to a browser. 

@jondong was working on an image plugin that makes the image available in java for just such a use case; but not sure of the status.

Also, searching Answer.ros.org turned up this solution:
http://answers.ros.org/question/158481/display-ros-image-on-webpage-using-roslibjs/

closed 2016-09-13 by mdhorn

jondong, 2016-09-13:
@doisyg I was working on a solution to publish color image data through VP8 encoded data stream within WebM container, which can be injected directly into a SourceBuffer object associates with a VideoElement. The solution is based on [rosnodejs](http://wiki.ros.org/rosnodejs) and [NW.js](http://nwjs.io/), and I think it can also be used in the browser context with the support of [browserify](http://browserify.org/). The project will be published soon after some internal processes, please stay tuned.

doisyg, 2016-09-14:
@jondong that's great, I am very curious about this project

doisyg, 2016-09-15:
Still in search of a way of running the ROS node for the R200 while accessing the color stream as a UVC video, I tried to set the node param 'enable_color' to false as I am only interested to use the depth data in ROS, but this still locks the color stream. @mdhorn, would it be possible to free the access to the color stream when it is not used by ROS?

Alternatively, I considered using v4l2loopback to duplicate the /dev/videoX associted with the color stream but this would imply having a way of telling librealsense which device to use. @mdhorn, any pointers to where this is done in the source code before I dig in it?

mdhorn, 2016-09-15:
@doisyg The device is locked in the librealsense library -- when using the camera, all of the devices are allocated to the process.
As I stated before, if you intend to use this ROS module then your solution should be native to ROS. Create a ROS node that subscribes to the existing color topic and make use of the image from there. This is exactly what the plug-in that @jondong is creating will do.

I also provide an example of this by searching Answer.ros.org:
http://answers.ros.org/question/158481/display-ros-image-on-webpage-using-roslibjs/

If you are using ROS, then stay within the ROS architecture for your solution; otherwise you'll be investing too much efforts in workarounds.

doisyg, 2016-10-18:
Reporting here because I have been struggling with this for a while but now have a working solution and wish to share.
I simply used the ros-virtual-cam package : https://github.com/mayfieldrobotics/ros-virtual-cam 
Working settings:

1. Start the r200 in ros: `roslaunch realsense_camera r200_nodelet_default.launch`
2. Create a virtual device: `sudo modprobe v4l2loopback devices=1 card_label="Ros CAM"`
3. Send the output of the rgb cam to the created virtual device, in my case /dev/video4, should be /dev/video3 if you have no other cam connected to your computer. Note the YUVV pixel format, important to be able to open the stream in a browser.`rosrun virtual_cam stream _device:=/dev/video4 _width:=640 _height:=480 _fourcc:=YUYV image:=camera/color/image_raw`
4. Check the result in VLC or you favorite browser (with https://apprtc.appspot.com/ for instance)

mdhorn, 2016-10-26:
@doisyg Nice solution using the native ROS. 
Thank you for putting this information here for others.