realsenseai/realsense-ros #106: [Feature request] Expose the apply_depth_control_preset function

I0x0I, 2016-09-13:
Hi everyone, as mentioned in #50, using the default settings of R200 may produce a noisy depth image, the [ds-private.cpp](https://github.com/IntelRealSense/librealsense/blob/22155fdd2a4f20c2736285a753ce8b1059031e10/src/ds-private.cpp) shows the default settings are actually optimized for outdoor scenario.

>         const dc_params dc_params::presets[] = {
            {5, 5, 192,  1,  512, 6, 24, 27,  7,   24}, // (DEFAULT) Default settings on chip. Similiar to the medium setting and best for outdoors.
            {5, 5,   0,  0, 1023, 0,  0,  0,  0, 2047}, // (OFF) Disable almost all hardware-based outlier removal
            {5, 5, 115,  1,  512, 6, 18, 25,  3,   24}, // (LOW) Provide a depthmap with a lower number of outliers removed, which has minimal false negatives.
            {5, 5, 185,  5,  505, 6, 35, 45, 45,   14}, // (MEDIUM) Provide a depthmap with a medium number of outliers removed, which has balanced approach.
            {5, 5, 175, 24,  430, 6, 48, 47, 24,   12}, // (OPTIMIZED) Provide a depthmap with a medium/high number of outliers removed. Derived from an optimization function.
            {5, 5, 235, 27,  420, 8, 80, 70, 90,   12}, // (HIGH) Provide a depthmap with a higher number of outliers removed, which has minimal false positives.
        };

By using the `apply_depth_control_preset` function from the [rs.hpp](https://github.com/IntelRealSense/librealsense/blob/22155fdd2a4f20c2736285a753ce8b1059031e10/include/librealsense/rs.hpp) to set the R200 works in HIGH mode, I can clearly see a reduce of noise points in the depth image.
So exposing the `apply_depth_control_preset` function in the ROS node is likely to solve the problem ;-)

doisyg, 2016-09-13:
That would be amazing!
I've never used librealsense, just the ROS implementation. Without exposing the 'apply_depth_control_preset' function in the ROS node, is there a way of changing this mode and making the change permanent to use it then in ROS?

I0x0I, 2016-09-13:
For different versions of librealsense, fixs might not be the same.
As for the latest version, a possible quick fix can be like

1.  Include `librealsense/rsutil.h` in [base_nodelet.h](https://github.com/intel-ros/realsense/blob/059d268efe7d3a260aa1ce0a6cba991b50ab386e/realsense_camera/include/realsense_camera/base_nodelet.h)

2. Add `rs_apply_depth_control_preset(rs_device_, 5)` in [base_nodelet.cpp](https://github.com/intel-ros/realsense/blob/059d268efe7d3a260aa1ce0a6cba991b50ab386e/realsense_camera/src/base_nodelet.cpp) line 478

I haven't try out this quick fix yet since I use an older version of librealsense and this fix is only for the R200.

doisyg, 2016-09-14:
Thanks a lot @I0x0I , this fix works great and I have a much cleaner pointcloud now

reaganlo, 2016-09-14:
@I0x0I Thanks for your suggestion. We are working on providing a ROS API for this functionality via rqt_reconfigure. We will keep you posted.

mdhorn, 2016-09-22:
This will be include in the upcoming 1.5.0 release.

mdhorn, 2016-09-29:
Fix in 1.5.0
https://github.com/intel-ros/realsense/releases/tag/1.5.0

closed 2016-09-29 by mdhorn

doisyg, 2016-10-18:
Is possible to view a working launch file which use the new depth control preset option?

mdhorn, 2016-10-18:
I'm attaching a sample file to set a different depth control preset: r200_nodelet_dc_preset.launch

However, in testing this I found a bug in the 1.5.0 implementation and documented the bug as Issue #132 

A fix for this should be in the 'indigo-devel' branch soon and it will be in the next release 1.6.0.

jinfagang, 2018-12-21:
@mdhorn  How to open this option in code?

mdhorn, 2018-12-27:
@jinfagang I haven't looked at this for a few years, but you should be able to change the parameter with the API http://wiki.ros.org/ROS/Parameter%20Server%20API