realsenseai/realsense-ros #116: Clarify the role of the enable_color parameter

doisyg, 2016-09-16:
### Expected Behavior
There should be a way of geting the color stream published on camera/color/image_raw  without using color on the generated pointcloud (for one who is only interested in depth data, this free a lot of computing power). 

### Actual Behavior
When enable_color is set to false (in order to generate a colorless pointcloud which uses less cpu), there is no way of accessing the color stream, either from or outside ROS.
When enable_color is set to true, there is no way of generating a colorless pointcloud.

doisyg, 2016-10-18:
Any update on this?

reaganlo, 2016-10-28:
@doisyg We do not recommend using the native pointcloud of the nodelet. That is why, by default, we have "enable_pointcloud" set to "false".

Instead, we suggest that you use the rgbd_launch package to generate pointclouds.
In order to achieve your "Expected Behavior":

1. Create a launch file similar to the [r200_nodelet_rgbd.launch](https://github.com/intel-ros/realsense/blob/indigo-devel/realsense_camera/launch/r200_nodelet_rgbd.launch). Make sure you update the "camera_type" based on the camera that you are using.
2. Change the default values of the following arguments from "true" to "**false**":
**depth_registered_processing**, **disparity_processing**, **disparity_registered_processing**, **hw_registered_processing**, **sw_registered_processing**.
3. Launch the modified launch file.
4. For color image, subscribe to **/camera/rgb/image_raw** topic
5. For colorless pointcloud, subscribe to **/camera/depth/points** topic

Attaching a sample launch file for the R200 camera.
[r200_nodelet_rgbd.launch.txt](https://github.com/intel-ros/realsense/files/559388/r200_nodelet_rgbd.launch.txt)

doisyg, 2016-11-19:
Great, it is much clearer now. Thanks for the sample file.

closed 2016-11-19 by doisyg