realsenseai/realsense-ros #118: Manual Calibration of R200

Parnia, 2016-09-17:
When running SLAM (like RTAB-Map) using R200 data, I see some small movements of the robot in the map but the robot is not actually moving. It seems like it's due to wrong calibration because I didn't have this problem before using xtion and it's calibration files. I calibrated r200 with the calibration node of ROS, but I don't know how to use it on realsense nodelet. I tried this on r200_nodelet_rgbd.launch:
```
 <arg name="rgb_camera_info_url" default="/home/parnia/Desktop/realSense_rgb.yaml" />
 <arg name="ir_camera_info_url" default="/home/pania/Desktop/realSense_depth.yaml" />
```
but it seems like it won't load anything.
how can I use my own calibration parameters?

mdhorn, 2016-10-04:
The ROS node currently does not support this feature, but we have added an internal story to track this enhancement.

ghindman, 2016-10-05:
@Parnia what is the functional issue that you were trying to fix by applying the image pipeline manual calibration?  We've not had anyone request this functionality in the past.

closed 2017-02-22 by rljordan