realsenseai/realsense-ros #119: Error launching gmapping with R200 camera

betaupsx86, 2016-09-19:
### System Configuration
Please complete Your Configuration detail below. Refer to the BKC at [Configuration](../realsense_camera/README.md#configuration).

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | Ubuntu 14..04 LTS              |
| Kernel                | 4.2.0-42-generic                |
| Backend               | video4linux                    |
| ROS                   | indigo                 |
| ROS RealSense         | 1.30                |
| librealsense          | 1.9.7-blue                |
| Camera Type-Firmware  | R200 1.0.72.06            |

---
#### How to collect Configuration Data
*This section can be deleted before submission.*

| Version               | Method |
|:--------------------- |:------------ |
| Operating System      | `grep DISTRIB_DESCRIPTION /etc/*elease*` |
| Kernel                | `uname -r` |
| Backend               | `ls /sys/class | grep video4linux` |
| ROS                   | `rosversion -d` |
| ROS RealSense         | `rosversion realsense_camera` |
| librealsense          | `cat <path to librealsense>/librealsense/readme.md | grep release-image | awk -F- '{print $3}'` |
| Camera Type-Firmware  | View the ROS log from running nodelet |

---

### Expected Behavior
I'm trying to initialize gmapping using a Realsense R200 camera. I've followed all the installation steps as posted on the Readme.md files. All R200 nodelet launch files work perfectly and I get pretty good data from the camera which I can easily visualize in Rviz. However, I'm unable to run gmapping and I think it has to do with some naming issues.

### Actual Behavior
These are screenshots from terminals running:
roslaunch realsense_camera r200_nodelet_navigation.launch (left)
roslaunch realsense_navigation gmapping.launch (right)

[image]

[image]

As you can see an extra /camera is being appended to all params for gmapping.launch. I'm guessing this results on two different nodelet manager names or some other naming issues and so things aren’t loaded correctly? I also noted in issue #37 that the prefix is RealsenseNodelet which seems to be correct from looking at the launch files on the realsense camera package, but it is different from what I have.

### Steps to Reproduce
roslaunch turtlebot_bringup minimal.launch
roslaunch realsense_camera r200_nodelet_navigation.launch
roslaunch realsense_navigation gmapping.launch

I can't figure out how to solve this problem

anelliot, 2016-09-21:
These instructions are now deprecated and in the process of being removed as we are in the process of adding r200 support to the turtlebot navigation package. These changes are almost complete and the r200 navigation package will be removed shortly.

betaupsx86, 2016-09-22:
Yes I noticed when I saw the turtlebot navigation while installing from source. I didn’t substitute the  r200.launch.xml for the one in realsense navigation and so another r200 nodelet was being launched when starting gmapping. I used the r200.launch.xml from realsense navigation and all is working great now. 

Is this fixed now since r200 is being added to the turtlebot navigation package?:
 
I suppose we wont need to start the nodelets separately once r200 is integrated completely

Totally unrelated, but would I be able to run the turtlebot follower by launching the r200 rgbd nodelet and then the follower since rgb_processing seems to be required?

tripzero, 2016-09-29:
Using the new source I was able to run gmapper-demo as described in the turtlebot_navigation tutorial on the wiki[1] and did not encounter the errors described in the report.

[1] - http://wiki.ros.org/turtlebot_navigation/Tutorials/indigo/Build%20a%20map%20with%20SLAM

kevincwells, 2016-10-05:
@betaupsx86, have you been able to retry this with the default turtlebot packages now that they have native R200 support?  Please keep in mind that you need to be pulling from source, as these commits are not yet in the released packages yet.

betaupsx86, 2016-10-05:
I was able to get it all working back then; acml, gmapping and the turtlebot follower example. The problem was that an extra camera nodelet was being started along the launch chain which conflicted with the already running r200 nodelet from the instructions. I modified some xml files and the issue was resolved. I will download the turtlebot sources again to see if everythings runs smoothly now that they integrated the r200. Thanks

betaupsx86, 2016-10-05:
Totally unrelated, but has anyone successfully used any person tracking packages with the r200.

kevincwells, 2016-10-26:
@betaupsx86 I don't believe we have anyone focusing on using the R200 with any of the existent person tracking packages out there at this time, though we have some internal solutions in the pipeline. If you end up having any issues getting them to work, please file a new issue -- we'd love the feedback.

closed 2016-10-26 by kevincwells