realsenseai/realsense-ros #120: rosrun tf view_frames shows no tf being published

sinhau, 2016-09-20:
### System Configuration
Please complete Your Configuration detail below. Refer to the BKC at [Configuration](../realsense_camera/README.md#configuration).

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | Ubuntu 16.04.1 LTS |
| Kernel                | 4.4.0-38-generic |
| Backend               | Video4Linux |
| ROS                   | ROS Kinetic |
| ROS RealSense         | 1.4.0 |
| librealsense          | 0.9.1 (although I used git checkout tags/stable so I would think this would say 0.9.2) |
| Camera Type-Firmware  | R200 1.0.72.06 |

I'm launching two R200 cameras using the r200_nodelet_multiple_cameras.launch.  I modified the launch file as shown below to change the `enable_tf` parameter to true.  When I run the launch file, both cameras launch and I am getting all the data.  The ROS logs even shows that cameras should be publishing transforms (see terminal output below)  However, when I run `rosrun tf view_frames`, the output frames.pdf file says no tf data received. For some reason no tf is being published. Moreover, when I run `rostopic echo /tf_static`, I only see camera2 tf's.  

### r200_nodelet_multiple_cameras.launch
```
<!-- Sample launch file for using multiple RealSense R200 cameras -->
<launch>
  <arg name="manager" value="nodelet_manager" /> <!-- Single nodelet manager for all cameras -->
  <node pkg="nodelet" type="nodelet" name="$(arg manager)" args="manager" output="screen"/>

  <!-- MY MODS -->
  <param name="camera1/enable_pointcloud" type="bool" value="false" />
  <param name="camera1/enable_tf"         type="bool" value="true" />
  <param name="camera2/enable_pointcloud" type="bool" value="false" />
  <param name="camera2/enable_tf"         type="bool" value="true" />
  <!-- ======= -->

  <!-- User must include the .xml file for each camera with unique values for "camera" and 
       either the "serial_no" or "usb_port_id" of the camera.
       "camera" should be a user friendly string that follows the ROS Names convention. -->
  <include file="$(find realsense_camera)/launch/includes/nodelet.launch.xml">
    <arg name="manager"      value="$(arg manager)" />
    <arg name="camera"       value="camera1" />
    <arg name="camera_type"  value="R200" /> <!-- Type of camera -->
    <arg name="serial_no"    value="2431002398" /> <!-- Note: Replace with actual serial number -->
    <arg name="usb_port_id"  value="" /> <!-- USB "Bus#-Port#" -->
  </include>

  <include file="$(find realsense_camera)/launch/includes/nodelet.launch.xml">
    <arg name="manager"      value="$(arg manager)" />
    <arg name="camera"       value="camera2" />
    <arg name="camera_type"  default="R200" /> <!-- Type of camera -->
    <arg name="serial_no"    value="2431018202" /> <!-- Note: Replace with actual serial number -->
    <arg name="usb_port_id"  value="" /> <!-- USB "Bus#-Port#" -->
  </include>
</launch>
```

### Terminal output after running the above launch file
```
SUMMARY
========

PARAMETERS
 * /camera1/base_frame_id: camera1_link
 * /camera1/camera_type: R200
 * /camera1/color_frame_id: camera1_rgb_frame
 * /camera1/color_optical_frame_id: camera1_rgb_optic...
 * /camera1/depth_frame_id: camera1_depth_frame
 * /camera1/depth_optical_frame_id: camera1_depth_opt...
 * /camera1/enable_pointcloud: False
 * /camera1/enable_tf: True
 * /camera1/ir2_frame_id: camera1_ir2_frame
 * /camera1/ir_frame_id: camera1_ir_frame
 * /camera1/serial_no: 2431002398
 * /camera1/usb_port_id: 
 * /camera2/base_frame_id: camera2_link
 * /camera2/camera_type: R200
 * /camera2/color_frame_id: camera2_rgb_frame
 * /camera2/color_optical_frame_id: camera2_rgb_optic...
 * /camera2/depth_frame_id: camera2_depth_frame
 * /camera2/depth_optical_frame_id: camera2_depth_opt...
 * /camera2/enable_pointcloud: False
 * /camera2/enable_tf: True
 * /camera2/ir2_frame_id: camera2_ir2_frame
 * /camera2/ir_frame_id: camera2_ir_frame
 * /camera2/serial_no: 2431018202
 * /camera2/usb_port_id: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.2

NODES
  /
    camera1 (nodelet/nodelet)
    camera2 (nodelet/nodelet)
    nodelet_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [9707]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 21dbc410-7ec2-11e6-8a74-001fc69bc102
process[rosout-1]: started with pid [9720]
started core service [/rosout]
process[nodelet_manager-2]: started with pid [9731]
process[camera1-3]: started with pid [9738]
process[camera2-4]: started with pid [9739]
[ INFO] [1474328294.635700141]: Initializing nodelet with 4 worker threads.
[ INFO] [1474328294.916099353]: /camera1 - Detected the following cameras:
				- Serial No: 2431002398, USB Port ID: 2-3, Name: Intel RealSense R200, Firmware: 1.0.72.06
				- Serial No: 2431018202, USB Port ID: 2-4, Name: Intel RealSense R200, Firmware: 1.0.72.06
[ INFO] [1474328294.916204645]: /camera1 - Connecting to camera with Serial No: 2431002398, USB Port ID: 2-3
[ INFO] [1474328295.215844540]: /camera1 - Setting static camera options
[ INFO] [1474328295.223619262]: /camera1 - Enabling Color in preset mode
[ INFO] [1474328295.224040801]: /camera1 - Enabling Depth in preset mode
[ INFO] [1474328295.224279460]: /camera1 - Enabling IR in preset mode
[ INFO] [1474328295.224421585]: /camera1 - Enabling IR2 in preset mode
[ INFO] [1474328295.224537239]: /camera1 - Starting camera
[ INFO] [1474328295.256159378]: /camera1 - Publishing camera transforms
[ INFO] [1474328295.256235771]: /camera1 - Setting dynamic camera options
[ INFO] [1474328296.077065450]: /camera2 - Detected the following cameras:
				- Serial No: 2431002398, USB Port ID: 2-3, Name: Intel RealSense R200, Firmware: 1.0.72.06
				- Serial No: 2431018202, USB Port ID: 2-4, Name: Intel RealSense R200, Firmware: 1.0.72.06
[ INFO] [1474328296.077198963]: /camera2 - Connecting to camera with Serial No: 2431018202, USB Port ID: 2-4
[ INFO] [1474328296.358673233]: /camera2 - Setting static camera options
[ INFO] [1474328296.365748123]: /camera2 - Enabling Color in preset mode
[ INFO] [1474328296.366053036]: /camera2 - Enabling Depth in preset mode
[ INFO] [1474328296.366201549]: /camera2 - Enabling IR in preset mode
[ INFO] [1474328296.366304975]: /camera2 - Enabling IR2 in preset mode
[ INFO] [1474328296.366402753]: /camera2 - Starting camera
[ INFO] [1474328296.398035491]: /camera2 - Publishing camera transforms
[ INFO] [1474328296.398113191]: /camera2 - Setting dynamic camera options
```

sinhau, 2016-09-26:
Hi @reaganlo, were you able to reproduce this issue?  Just checking in to see if there are any updates on this.

reaganlo, 2016-09-27:
Hi @sinhau , I was able to reproduce the error. We will look into this and provide you an update.

As a **workaround for multiple cameras**, could you please do the following:

- Create default launch files for each of the camera. I am attaching 2 samples.
E.g r200_nodelet_default_1.launch and r200_nodelet_default_2.launch.
- Edit each of the launch files to have unique values for the arguments "manager", "camera" and "serial_no/usb_port_id".
- Launch them in separate terminals.
- Use **rostopic echo /tf_static** to view the transforms.  Note: **rosrun tf view_frames** does not work for static transforms.

[r200_nodelet_default_1.launch.txt](https://github.com/intel-ros/realsense/files/494444/r200_nodelet_default_1.launch.txt)
[r200_nodelet_default_2.launch.txt](https://github.com/intel-ros/realsense/files/494445/r200_nodelet_default_2.launch.txt)

reaganlo, 2016-09-27:
@sinhau Yes, this can happen when you try to launch 2 cameras simultaneous because it leads 2 separate processes trying to access the librealsense "context" at the same time. Just like how you noticed, I would suggest you launch them one after the other.

tulku, 2016-10-26:
It would be good for the drivers to not fail when accessing the "context" simultaneously.

@reaganlo are there plans on fixes this?

mkhansen-intel, 2016-11-18:
@sinhau and @tulku - I looked into this further and the root cause can be traced to this issue in the roscomm libraries (roscpp):[https://github.com/ros/ros_comm/issues/146]( https://github.com/ros/ros_comm/issues/146)

Since that is unlikely to be fixed per their maintainers, the work-around of running from 2 launch files is the only option right now.

tulku, 2016-11-18:
Hi! I agree that the solution is to have multiple launchfiles, however that option is not currently working. As soon as you try to start too nodes (via a launchfile, so they start at the same time) one of the nodes crashes. This is probably a different bug, I can open it tomorrow.

sinhau, 2016-11-18:
@tulku , yes I'm seeing the same issue.  Looks like this bug was introduced in one of the newer releases because launching multiple cameras via separate launch files was working for me on 1.4.0.  Thanks for opening this issue!

tulku, 2016-11-21:
@mkhansen-intel I've created the issue that should be fixed before being able to use multiple nodelet managers. https://github.com/intel-ros/realsense/issues/156

mkhansen-intel, 2016-11-30:
Since the underlying issue is in roscpp and is not scheduled to be fixed, we're looking into a fix for this one that would fork a new process before calling the static transform publisher. We'll update this email with results.

AndyZelenak, 2016-12-01:
fyi I'm also interested in having tf's for multiple cameras, so I'll be watching this.

mdhorn, 2016-12-23:
Fixed in release 1.7.0
Use Dynamic Transforms by setting both 'enable_tf' and 'enable_tf_dynamic' to true.
https://github.com/intel-ros/realsense/releases/tag/1.7.0

closed 2016-12-23 by mdhorn