realsenseai/realsense-ros #123: TK1 Freeze upon realsense ROSnode startup

mallocassi, 2016-09-24:
### System Configuration
Please complete Your Configuration detail below. Refer to the BKC at [Configuration](../realsense_camera/README.md#configuration).

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      |  Ubuntu 14.04.5 LTS                |
| Kernel                |  3.10.40-grinch-21.2.1             |
| Backend               |  video4linux               |
| ROS                   |  indigo                  |
| ROS RealSense         | 1.4.0                |
| librealsense          | 0.9.1          |
| Camera Type-Firmware  |  1.0.72.06       |

---
#### How to collect Configuration Data
*This section can be deleted before submission.*

| Version               | Method |
|:--------------------- |:------------ |
| Operating System      | `grep DISTRIB_DESCRIPTION /etc/*elease*` |
| Kernel                | `uname -r` |
| Backend               | `ls /sys/class | grep video4linux` |
| ROS                   | `rosversion -d` |
| ROS RealSense         | `rosversion realsense_camera` |
| librealsense          | `cat <path to librealsense>/librealsense/readme.md | grep release-image | awk -F- '{print $3}'` |
| Camera Type-Firmware  | View the ROS log from running nodelet |

---

### Expected Behavior

When we plug the Realsense R200 in to our TK1, running the "r200_nodelet_default" node, we expect it to run properly.

### Actual Behavior

When we plug the Realsense R200 in to our TK1, running the "r200_nodelet_default" node, it freezes, does not show any further messages, and does not show any errors after this message:

[ INFO] [1474748112.582733021]: /camera - Starting camera

We have to always press the Reset button on the TK1 after this happens.
Any idea why the Realsense freezes the node/TK1?

Here's what we've tried:
To make sure Realsense gets enough power, we plugged it in to a USB hub, which has its own power adapter.
To make sure it's not a USB speed issue, we disconnected all other things to the USB hub, which is USB 3.0
We have the Grinch kernel installed.
To make sure it's not an HDMI issue, we removed the cable and are SSHing into the TK1.
Power draw is ~10-15% before running ROS, and ~20% running ROS and the node, regardless of whether the camera is plugged in.

Update: Running "r200_nodelet_rgbd.launch" with the Realsense plugged in makes all cores go to 100% and freezing.

### Steps to Reproduce

Plug in TK1. Plug in realsense camera into powered USB hub, connected to TK1.
SSH into TK1, run ROS: 'roscore'
launch the ROS node for the R200 camera: 'roslaunch realsense_camera r200_nodelet_default.launch'

tripzero, 2016-09-29:
You may need a more recent kernel to get the r200 to work.  A good test is to run librealsense/bin/cpp-enumerate.  This will tell you if you have uvcvideo, udev, and all the right drivers setup correctly.

mdhorn, 2016-09-29:
We do not have one of these boards to validate, but a few possible issue I see from the configuration information submitted.
1. You need to be using librealsense v0.9.2 - 0.9.1 is not compatible with realsense_camera 1.4.0.
2. The 3.10.40 does not have support for the R200 data formats. The [librealsense installation directions](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) provide patching information for 3.19.x and the 4.4.x. You'll need to patch your kernel first.

Once you have the kernel patched and the newer librealsense, you should build the [librealsense code examples](https://github.com/IntelRealSense/librealsense/tree/master/examples) and use the cpp-enumerate-devices example to ensure that the camera is visible and accessible. You could also use the librealsense example code cpp-headless to ensure you can capture a single frame from each stream. We will be adding a tool to the ROS realsense_camera package with similar functionality to help with debug in the near future.

Once you know you can capture a frame from each stream, then I would try the ROS node again.

mdhorn, 2016-10-28:
Under 14.04.5 you should be using the latest Xenial kernel 4.4.x.
Please try to use the prebuilt ROS Debian packages as the latest releases handle all of the udev rules and kernel patching; the only [prerequisite is enable downloading of kernel sources](http://wiki.ros.org/librealsense#Installation_Prerequisites).

Closing due to no response.

closed 2016-10-28 by mdhorn