realsenseai/realsense-ros #125: dose realsense camea support v4l2 command (for ex, VIDIOC_S_FMT)

liyuming1978, 2016-10-09:
I need do video-streaming from realsense color camera, (it's /dev/video0), if I get the data from ROS, the latency is too high, it's better to get directly from v4l2, 
but, I can not call  VIDIOC_S_FMT or other v4l2 command for these cameras.
I want to know, is it possible to encoder directly from camera interface?

mdhorn, 2016-10-17:
Are you working in ROS or some other application for using the camera?

If you are working in ROS, then we need to address the latency concerns.

If you working directly with the camera for non-ROS applications, then you just need to be using librealsense: https://github.com/IntelRealSense/librealsense/
which is the library realsense-camera relies upon.

mdhorn, 2016-10-28:
@liyuming1978 Let me know if you are working in ROS and if we need to follow up on this.

closed 2017-02-15 by mdhorn