realsenseai/realsense-ros #127: Multiple R200s on one laptop.  Depth on second camera stops after a few seconds.

ScottJohnson2718, 2016-10-11:
### System Configuration

| Version               | Your Configuration   |
|:--------------------- |:---------------------|
| Operating System      | Ubuntu 14.04.5 LTS  |
| Kernel                | 4.4.0-040400-generic  |
| Backend               | video4linux  |
| ROS                   | Indigo                   |
| ROS RealSense         | 1.4.0  |
| librealsense          | 0.9.1  |
| Camera Type-Firmware  | 1.0.72.06  |

---

### Expected Behavior
Depth from both cameras continues to be published

### Actual Behavior
Depth from camera2 stops publishing after a few seconds.

### Steps to Reproduce

-Connect two R200s via a powered USB3 hub.
-Connect the powered hub to a USB3 port on a laptop running ROS as configured above. 
-Run librealsense/bin/cpp-multicam.  Observe that there are two color streams and two depth streams.  Close cpp-multicam.  This shows the driver is fine.
-Run librealsense/bin/cpp-enumerate-devices.  Note the serial number and usb port ids of the two cameras.
-Edit realsense/launch/r200_nodelet_mulitple_cameras.launch so that it has the serial numbers and usb_port_ids of the two cameras.
-roslaunch realsense_camera r200_nodelet_multiple_cameras.launch
-use "rostopic bw camera1/depth/image_raw" to see that camera 1 is publishing depth.
-use "rostopic bw camera2/depth/image_raw" to see that camera 2 is publishing depth.  Wait.  -Repeat rostopic commands.  See that camera 2 is not publishing depth.  CTRL-C the launch window.
-Edit the launch file to turn off the color channel.  I used:
`<param name="enable_color"            type="bool" value="false" />`
-Repeat launching the cameras.
-Use rostopic to see that both depth channels are now working.

mdhorn, 2016-10-20:
@ScottJohnson2718 We have tested running multiple camera at length; I run reverified running two r200 for well over an hour, but this was without a hub.

We have seen problems with various USB3 hubs, both unpowered and powered. Unfortunately not all hubs are created equally and there doesn't appear to be any clear certification.  At this moment I can't recommend any specific hub, but I would suggest running without a hub if possible, or try another brand of hub.

Separate issue is the configuration. The latest version of ubuntu 14.04.5 should be defaulting to using the 'linux-image-generic-lts-xenial' which will give you a 4.4.0-42 kernel. There should no longer be a need to install the generic 4.4 kernel listed on the librealsense setup directions.
Also, the minimum version of librealsense supported with realsense_camera 1.4.0 is v0.9.2.

The next package sync of ROS for indigo will include realsense_camera 1.5.0 and librealsense 1.11.0 -- when these new packages are released you will no longer need to install anything from source -- just the Debian packages. This is already the case of Kinetic, and will be true for Indigo with the next ROS package sync which is pending.

More than likely the problems you are seeing are related the the USB3 hub quality, and not configuration.

closed 2016-10-20 by mdhorn

mdhorn, 2016-10-20:
You could also try lowering the frame rate for the streams on both cameras:

`  <arg name="depth_fps"         default="30" />`
`  <arg name="color_fps"         default="30" />`
`  <param name="$(arg camera)/depth_fps"         type="int"  value="$(arg depth_fps)" />`
`  <param name="$(arg camera)/color_fps"         type="int"  value="$(arg color_fps)" />`