realsenseai/realsense-ros #128: catkin_make failing due to missing variables ??

jacksonkr, 2016-10-12:
```
$ cd ~/catkin_ws/src
$ git clone https://github.com/intel-ros/realsense.git
$ cd ../
$ rosdep install --skip-keys=librealsense --from-paths -i src/realsense
#All required rosdeps installed successfully
$ catkin_make
```

(after getting to ~96%)

```
/home/jackson/catkin_ws/src/realsense/realsense_camera/include/realsense_camera/zr300_nodelet.h:68:24: error: ‘timestamp_data’ is not a member of ‘rs’
     std::function<void(rs::timestamp_data)> timestamp_handler_;
                        ^
/home/jackson/catkin_ws/src/realsense/realsense_camera/include/realsense_camera/zr300_nodelet.h:68:43: error: template argument 1 is invalid
     std::function<void(rs::timestamp_data)> timestamp_handler_;
                                           ^
/home/jackson/catkin_ws/src/realsense/realsense_camera/src/zr300_nodelet.cpp: In member function ‘virtual void realsense_camera::ZR300Nodelet::onInit()’:
/home/jackson/catkin_ws/src/realsense/realsense_camera/src/zr300_nodelet.cpp:62:13: error: ‘RS_STREAM_FISHEYE’ was not declared in this scope
     format_[RS_STREAM_FISHEYE] = RS_FORMAT_RAW8;
             ^
/home/jackson/catkin_ws/src/realsense/realsense_camera/src/zr300_nodelet.cpp:62:34: error: ‘RS_FORMAT_RAW8’ was not declared in this scope
     format_[RS_STREAM_FISHEYE] = RS_FORMAT_RAW8;
                                  ^
/home/jackson/catkin_ws/src/realsense/realsense_camera/src/zr300_nodelet.cpp: In member function ‘virtual void realsense_camera::ZR300Nodelet::getParameters()’:
/home/jackson/catkin_ws/src/realsense/realsense_camera/src/zr300_nodelet.cpp:86:42: error: ‘RS_STREAM_FISHEYE’ was not declared in this scope
     pnh_.param("enable_fisheye", enable_[RS_STREAM_FISHEYE], ENABLE_FISHEYE);
                                          ^
/home/jackson/catkin_ws/src/realsense/realsense_camera/src/zr300_nodelet.cpp: In member function ‘virtual void realsense_camera::ZR300Nodelet::advertiseTopics()’:
/home/jackson/catkin_ws/src/realsense/realsense_camera/src/zr300_nodelet.cpp:109:23: error: ‘RS_STREAM_FISHEYE’ was not declared in this scope
     camera_publisher_[RS_STREAM_FISHEYE] = fisheye_image_transport.advertiseCamera(FISHEYE_TOPIC, 1);
                       ^
/home/jackson/catkin_ws/src/realsense/realsense_camera/src/zr300_nodelet.cpp: In member function ‘bool realsense_camera::ZR300Nodelet::getIMUInfo(realsense_camera::GetIMUInfo::Request&, realsense_camera::GetIMUInfo::Response&)’:
/home/jackson/catkin_ws/src/realsense/realsense_camera/src/zr300_nodelet.cpp:134:5: error: ‘rs_motion_intrinsics’ was not declared in this scope
     rs_motion_intrinsics imu_intrinsics;
     ^
/home/jackson/catkin_ws/src/realsense/realsense_camera/src/zr300_nodelet.cpp:135:43: error: ‘imu_intrinsics’ was not declared in this scope
     rs_get_motion_intrinsics(rs_device_, &imu_intrinsics, &rs_error_);
                                           ^
/home/jackson/catkin_ws/src/realsense/realsense_camera/src/zr300_nodelet.cpp:135:69: error: ‘rs_get_motion_intrinsics’ was not declared in this scope
     rs_get_motion_intrinsics(rs_device_, &imu_intrinsics, &rs_error_);
                                                                     ^
/home/jackson/catkin_ws/src/realsense/realsense_camera/src/zr300_nodelet.cpp: In member function ‘void realsense_camera::ZR300Nodelet::configCallback(realsense_camera::zr300_paramsConfig&, uint32_t)’:
/home/jackson/catkin_ws/src/realsense/realsense_camera/src/zr300_nodelet.cpp:469:38: error: ‘RS_OPTION_FISHEYE_EXPOSURE’ was not declared in this scope
     rs_set_device_option(rs_device_, RS_OPTION_FISHEYE_EXPOSURE,
                                      ^
/home/jackson/catkin_ws/src/realsense/realsense_camera/src/zr300_nodelet.cpp:471:38: error: ‘RS_OPTION_FISHEYE_GAIN’ was not declared in this scope
     rs_set_device_option(rs_device_, RS_OPTION_FISHEYE_GAIN, config.fisheye_gain, 0);
                                      ^
/home/jackson/catkin_ws/src/realsense/realsense_camera/src/zr300_nodelet.cpp:472:38: error: ‘RS_OPTION_FISHEYE_STROBE’ was not declared in this scope
     rs_set_device_option(rs_device_, RS_OPTION_FISHEYE_STROBE, config.fisheye_strobe, 0);
                                      ^
/home/jackson/catkin_ws/src/realsense/realsense_camera/src/zr300_nodelet.cpp:473:38: error: ‘RS_OPTION_FISHEYE_EXTERNAL_TRIGGER’ was not declared in this scope
     rs_set_device_option(rs_device_, RS_OPTION_FISHEYE_EXTERNAL_TRIGGER, config.fisheye_external_trigger, 0);
                                      ^
/home/jackson/catkin_ws/src/realsense/realsense_camera/src/zr300_nodelet.cpp:474:38: error: ‘RS_OPTION_FISHEYE_ENABLE_AUTO_EXPOSURE’ was not declared in this scope
     rs_set_device_option(rs_device_, RS_OPTION_FISHEYE_ENABLE_AUTO_EXPOSURE, config.fisheye_enable_auto_exposure, 0);
                                      ^
/home/jackson/catkin_ws/src/realsense/realsense_camera/src/zr300_nodelet.cpp:475:38: error: ‘RS_OPTION_FISHEYE_AUTO_EXPOSURE_MODE’ was not declared in this scope
     rs_set_device_option(rs_device_, RS_OPTION_FISHEYE_AUTO_EXPOSURE_MODE, config.fisheye_auto_exposure_mode, 0);
                                      ^
/home/jackson/catkin_ws/src/realsense/realsense_camera/src/zr300_nodelet.cpp:476:38: error: ‘RS_OPTION_FISHEYE_AUTO_EXPOSURE_ANTIFLICKER_RATE’ was not declared in this scope
     rs_set_device_option(rs_device_, RS_OPTION_FISHEYE_AUTO_EXPOSURE_ANTIFLICKER_RATE,
                                      ^
/home/jackson/catkin_ws/src/realsense/realsense_camera/src/zr300_nodelet.cpp:478:38: error: ‘RS_OPTION_FISHEYE_AUTO_EXPOSURE_PIXEL_SAMPLE_RATE’ was not declared in this scope
     rs_set_device_option(rs_device_, RS_OPTION_FISHEYE_AUTO_EXPOSURE_PIXEL_SAMPLE_RATE,
                                      ^
/home/jackson/catkin_ws/src/realsense/realsense_camera/src/zr300_nodelet.cpp:480:38: error: ‘RS_OPTION_FISHEYE_AUTO_EXPOSURE_SKIP_FRAMES’ was not declared in this scope
     rs_set_device_option(rs_device_, RS_OPTION_FISHEYE_AUTO_EXPOSURE_SKIP_FRAMES,
[... 252 lines cut ...]
     ^
/home/jackson/catkin_ws/src/realsense/realsense_camera/test/camera_core.cpp:644:41: error: ‘RS_STREAM_FISHEYE’ was not declared in this scope
     EXPECT_EQ(g_caminfo_projection_recv[RS_STREAM_FISHEYE][9], 0.0);
                                         ^
/home/jackson/catkin_ws/src/realsense/realsense_camera/test/camera_core.cpp:644:5: error: template argument 1 is invalid
     EXPECT_EQ(g_caminfo_projection_recv[RS_STREAM_FISHEYE][9], 0.0);
     ^
In file included from /usr/include/gtest/gtest.h:57:0,
                 from /home/jackson/catkin_ws/src/realsense/realsense_camera/test/camera_core.cpp:30:
/home/jackson/catkin_ws/src/realsense/realsense_camera/test/camera_core.cpp:645:43: error: ‘RS_STREAM_FISHEYE’ was not declared in this scope
     EXPECT_TRUE(g_caminfo_projection_recv[RS_STREAM_FISHEYE][10] != 0.0);
                                           ^
In file included from /usr/include/gtest/gtest.h:1807:0,
                 from /home/jackson/catkin_ws/src/realsense/realsense_camera/test/camera_core.cpp:30:
/home/jackson/catkin_ws/src/realsense/realsense_camera/test/camera_core.cpp:646:41: error: ‘RS_STREAM_FISHEYE’ was not declared in this scope
     EXPECT_EQ(g_caminfo_projection_recv[RS_STREAM_FISHEYE][11], 0.0);
                                         ^
/home/jackson/catkin_ws/src/realsense/realsense_camera/test/camera_core.cpp:646:5: error: template argument 1 is invalid
     EXPECT_EQ(g_caminfo_projection_recv[RS_STREAM_FISHEYE][11], 0.0);
     ^
make[2]: *** [realsense/realsense_camera/CMakeFiles/tests_camera_core.dir/test/camera_core.cpp.o] Error 1
make[1]: *** [realsense/realsense_camera/CMakeFiles/tests_camera_core.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
```

ScottJohnson2718, 2016-10-12:
That many errors says that something is very wrong.  Do you have the realsense driver?
https://github.com/IntelRealSense/librealsense.git

You didn't print the standard info for these topics to list what version of Ubuntu you have etc so that's my best guess.  
I have it running on a laptop with Ubuntu 14.04 and ROS Indigo and it compiles fine.
The instructions for installation start at
https://github.com/intel-ros/realsense/tree/indigo-devel/realsense_camera
so there is more than what you listed in your post.

Plus it is a good idea to get the driver samples running in the librealsense/bin before you expect the ROS layer on top to work.

Scott

jacksonkr, 2016-10-12:
Yes, that's the page which has been a guide to where I am now. I downloaded the driver and built the example apps using qt as instructed. All examples work as expected with the sensor.

Yes, I'm on 14.04 and Indigo. Also, it took me a minute but I figured out how to get more lines in terminal. I put the entire output in the OP.

molinadavid, 2016-10-13:
I can confirm the same issue, I installed the camera on one of my pc last month and it worked flawlessly (Ros install) but today I am trying to install again on a new Ubuntu installation and I am getting the same errors, my best guess is that they updated something on the realsense driver but I am just starting the debugging process so I can not give any more information.

molinadavid, 2016-10-13:
As a quick update to this problem, if I revert the commit to "f55acae7d1812a7a0fdbdd372b5e6ea9c010b731" 
`git reset --hard f55acae7d1812a7a0fdbdd372b5e6ea9c010b731
HEAD is now at f55acae Update Changelog for 1.5.0 Release`

it builds without any problem.

jacksonkr, 2016-10-13:
@molinadavid 

```
$ git reset --hard f55acae7d1812a7a0fdbdd372b5e6ea9c010b731
fatal: Could not parse object 'f55acae7d1812a7a0fdbdd372b5e6ea9c010b731'.

```

molinadavid, 2016-10-13:
@jacksonkr  Basically what I just did was to revert the commit to version 1.5.0 to make it work

[https://github.com/intel-ros/realsense/commit/f9eed92441ee0d81663d337ccc1a1b5ee98ca827](https://github.com/intel-ros/realsense/commit/f9eed92441ee0d81663d337ccc1a1b5ee98ca827)

The commit hash I used is the parent commit but you could also just download the files directly from github if you still have troubles with it. I just run it again in a new installation with the same results, the latest source code gives me the same error, I figure it has something to do with the support for the ZR300 camera.

Once I reverted the commit to the version stated before it compiled without any issue.

jacksonkr, 2016-10-13:
@molinadavid Odd. This is the hash I saw when I went to that page:
f9eed92441ee0d81663d337ccc1a1b5ee98ca827

Regardless, it's working like a charm.

molinadavid, 2016-10-14:
@jacksonkr Without knowing much about this, what I can see from your logs is that is failing with an error on rtabmap library, are you also using the that library?

Also just to point out in case you missed it, did you run the `rosdep install realsense_camera ` and I installed the librealsense 0.9.2 version instead of the latest one, I don't know if it has something to do.

jacksonkr, 2016-10-14:
Because of an ubuntu problem I ended up installing 16.04 and Kinetic. That being said I ran into similar issues. Once I made sure all the dependencies were in place everything installed fine.

~/catkin_ws/src/realsense
~/catkin_ws/src/rtabmap_ros

~/github/librealsense

To install dependenceis that were brought up I `$ sudo apt-get install ros-kinetec-<dependency>`

jacksonkr, 2016-10-20:
My Jetson TK1 (armhf) will not upgrade to ubuntu 16.04 so I cannot update to kinetic. On this machine I'm stuck with indigo and therefore stuck with this issue.

I have installed librealsense source, ros-indigo-realsense-camera, ros-indigo-librealsense and rosdep --install --from-paths src but the following still causes this nasty error.

```
$ catkin_make
Base path: /home/jackson/catkin_ws
Source space: /home/jackson/catkin_ws/src
Build space: /home/jackson/catkin_ws/build
Devel space: /home/jackson/catkin_ws/devel
Install space: /home/jackson/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/jackson/catkin_ws/build"
####
####
#### Running command: "make -j8 -l8" in "/home/jackson/catkin_ws/build"
####
[ 12%] [ 12%] Built target realsense_camera_gencfg
Built target std_msgs_generate_messages_py
[ 12%] [ 12%] [ 12%] [ 12%] Built target _realsense_camera_generate_messages_check_deps_SetPower
Built target _realsense_camera_generate_messages_check_deps_ForcePower
Built target std_msgs_generate_messages_lisp
Built target std_msgs_generate_messages_cpp
[ 12%] [ 12%] [ 12%] Built target _realsense_camera_generate_messages_check_deps_CameraConfiguration
Built target _realsense_camera_generate_messages_check_deps_IsPowered
Built target _realsense_camera_generate_messages_check_deps_IMUInfo
[ 12%] Built target _realsense_camera_generate_messages_check_deps_GetIMUInfo
[ 12%] Built target sensor_msgs_generate_messages_py
[ 12%] [ 12%] [ 12%] Built target dynamic_reconfigure_generate_messages_cpp
Built target dynamic_reconfigure_generate_messages_py
[ 12%] Built target nodelet_generate_messages_cpp
Built target dynamic_reconfigure_generate_messages_lisp
[ 12%] [ 12%] [ 12%] [ 12%] Built target dynamic_reconfigure_gencfg
Built target rosgraph_msgs_generate_messages_cpp
Built target rosgraph_msgs_generate_messages_lisp
Built target nodelet_generate_messages_py
[ 12%] [ 12%] Built target roscpp_generate_messages_lisp
Built target roscpp_generate_messages_cpp
[ 12%] [ 12%] [ 12%] Built target nodelet_generate_messages_lisp
Built target bond_generate_messages_cpp
Built target rosgraph_msgs_generate_messages_py
[ 12%] [ 12%] Built target roscpp_generate_messages_py
Built target bond_generate_messages_lisp
[ 12%] [ 12%] Built target bond_generate_messages_py
Built target sensor_msgs_generate_messages_lisp
[ 12%] Built target geometry_msgs_generate_messages_cpp
[ 12%] [ 12%] [ 12%] Built target sensor_msgs_generate_messages_cpp
Built target pcl_msgs_generate_messages_cpp
Built target tf_generate_messages_lisp
[ 12%] Built target tf_generate_messages_py
[ 12%] Built target topic_tools_generate_messages_py
[ 12%] [ 12%] [ 12%] Built target geometry_msgs_generate_messages_lisp
Built target tf_generate_messages_cpp
Built target geometry_msgs_generate_messages_py
[ 12%] [ 12%] [ 12%] [ 12%] Built target actionlib_generate_messages_lisp
Built target actionlib_generate_messages_cpp
Built target pcl_msgs_generate_messages_py
Built target actionlib_generate_messages_py
[ 12%] Built target tf2_msgs_generate_messages_lisp
[ 12%] [ 12%] [ 12%] [ 12%] Built target tf2_msgs_generate_messages_py
Built target actionlib_msgs_generate_messages_cpp
Built target topic_tools_generate_messages_lisp
Built target pcl_ros_gencfg
[ 12%] [ 12%] Built target actionlib_msgs_generate_messages_py
Built target actionlib_msgs_generate_messages_lisp
[ 12%] Built target pcl_msgs_generate_messages_lisp
[ 12%] [ 34%] [ 37%] [ 40%] Built target topic_tools_generate_messages_cpp
Built target realsense_camera_generate_messages_py
Built target gtest
Built target tf2_msgs_generate_messages_cpp
[... 388 lines cut ...]
     ^
/home/jackson/catkin_ws/src/realsense/realsense_camera/test/camera_core.cpp:644:41: error: ‘RS_STREAM_FISHEYE’ was not declared in this scope
     EXPECT_EQ(g_caminfo_projection_recv[RS_STREAM_FISHEYE][9], 0.0);
                                         ^
/home/jackson/catkin_ws/src/realsense/realsense_camera/test/camera_core.cpp:644:5: error: template argument 1 is invalid
     EXPECT_EQ(g_caminfo_projection_recv[RS_STREAM_FISHEYE][9], 0.0);
     ^
In file included from /usr/include/gtest/gtest.h:57:0,
                 from /home/jackson/catkin_ws/src/realsense/realsense_camera/test/camera_core.cpp:30:
/home/jackson/catkin_ws/src/realsense/realsense_camera/test/camera_core.cpp:645:43: error: ‘RS_STREAM_FISHEYE’ was not declared in this scope
     EXPECT_TRUE(g_caminfo_projection_recv[RS_STREAM_FISHEYE][10] != 0.0);
                                           ^
In file included from /usr/include/gtest/gtest.h:1807:0,
                 from /home/jackson/catkin_ws/src/realsense/realsense_camera/test/camera_core.cpp:30:
/home/jackson/catkin_ws/src/realsense/realsense_camera/test/camera_core.cpp:646:41: error: ‘RS_STREAM_FISHEYE’ was not declared in this scope
     EXPECT_EQ(g_caminfo_projection_recv[RS_STREAM_FISHEYE][11], 0.0);
                                         ^
/home/jackson/catkin_ws/src/realsense/realsense_camera/test/camera_core.cpp:646:5: error: template argument 1 is invalid
     EXPECT_EQ(g_caminfo_projection_recv[RS_STREAM_FISHEYE][11], 0.0);
     ^
make[2]: *** [realsense/realsense_camera/CMakeFiles/tests_camera_core.dir/test/camera_core.cpp.o] Error 1
make[1]: *** [realsense/realsense_camera/CMakeFiles/tests_camera_core.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
```

jacksonkr, 2016-10-20:
Also, I'm noticing that many of the "error" lines is complaining about the zr300 where I'm only needing to use an r200.

mdhorn, 2016-10-20:
@jacksonkr Is there a reason you are build from sources instead of using the Debian packages? OSRF should be hosing the armhf packages.

If you are going to build from sources, stick with the released versions by checking out the via tags:

Since you are using R200, I would suggest using librealsense v0.9.2 and realsense_camera 1.4.0.

1. Download and build librealsense first
git clone https://github.com/IntelRealSense/librealsense.git
cd librealsense
git checkout -b b0.9.2 v0.9.2

2. Download and build realsense_camera second
git clone https://github.com/intel-ros/realsense
cd realsense
git checkout 1.4.0

jacksonkr, 2016-10-21:
@mdhorn This is happening on ros-indigo Ubuntu 14.04 on amd64 as well as armhf (armv7), even after the notes you gave.

Output from the amd64

```
$ catkin_make
Base path: /home/jackson/catkin_ws
Source space: /home/jackson/catkin_ws/src
Build space: /home/jackson/catkin_ws/build
Devel space: /home/jackson/catkin_ws/devel
Install space: /home/jackson/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/jackson/catkin_ws/build"
####
####
#### Running command: "make -j8 -l8" in "/home/jackson/catkin_ws/build"
####
[  0%] [  3%] Built target std_msgs_generate_messages_lisp
Built target realsense_camera_gencfg
[  3%] [  3%] [  3%] Built target _realsense_camera_generate_messages_check_deps_IsPowered
[  3%] Built target _realsense_camera_generate_messages_check_deps_CameraConfiguration
Built target std_msgs_generate_messages_py
Built target std_msgs_generate_messages_cpp
[  3%] [  3%] [  3%] Built target _realsense_camera_generate_messages_check_deps_GetIMUInfo
Built target _realsense_camera_generate_messages_check_deps_ForcePower
Built target _realsense_camera_generate_messages_check_deps_IMUInfo
[  3%] [  3%] Built target dynamic_reconfigure_generate_messages_cpp
Built target dynamic_reconfigure_generate_messages_lisp
[  3%] [  3%] Built target dynamic_reconfigure_generate_messages_py
Built target dynamic_reconfigure_gencfg
[  3%] Built target _realsense_camera_generate_messages_check_deps_SetPower
[  3%] [  3%] Built target tf_generate_messages_cpp
Built target nodelet_generate_messages_cpp
[  3%] [  3%] Built target nodelet_generate_messages_lisp
Built target rosgraph_msgs_generate_messages_cpp
[  3%] [  3%] Built target rosgraph_msgs_generate_messages_lisp
Built target rosgraph_msgs_generate_messages_py
[  3%] Built target roscpp_generate_messages_cpp
[  3%] [  3%] [  3%] Built target nodelet_generate_messages_py
Built target tf_generate_messages_lisp
Built target roscpp_generate_messages_lisp
[  3%] [  3%] [  3%] Built target sensor_msgs_generate_messages_cpp
Built target sensor_msgs_generate_messages_lisp
Built target bond_generate_messages_cpp
[  3%] [  3%] Built target roscpp_generate_messages_py
Built target bond_generate_messages_lisp
[  3%] [  3%] Built target sensor_msgs_generate_messages_py
[  3%] [  3%] [  3%] Built target tf_generate_messages_py
Built target bond_generate_messages_py
Built target geometry_msgs_generate_messages_lisp
Built target geometry_msgs_generate_messages_cpp
[  3%] [  3%] [  3%] Built target geometry_msgs_generate_messages_py
Built target actionlib_generate_messages_cpp
Built target actionlib_generate_messages_lisp
[  3%] [  3%] Built target actionlib_generate_messages_py
Built target tf2_msgs_generate_messages_py
[  3%] [  3%] Built target tf2_msgs_generate_messages_lisp
Built target topic_tools_generate_messages_cpp
[  3%] Built target actionlib_msgs_generate_messages_cpp
[  3%] Built target actionlib_msgs_generate_messages_lisp
[  3%] [  3%] Built target pcl_ros_gencfg
Built target tf2_msgs_generate_messages_cpp
[  3%] Built target actionlib_msgs_generate_messages_py
[  3%] Built target pcl_msgs_generate_messages_lisp
[  3%] [  3%] Built target pcl_msgs_generate_messages_py
Built target topic_tools_generate_messages_py
[  3%] [  3%] Built target pcl_msgs_generate_messages_cpp
Built target topic_tools_generate_messages_lisp
[  4%] Built target gtest
[  4%] Built target _rtabmap_ros_generate_messages_check_deps_ResetPose
[... 424 lines cut ...]
     ^
/home/jackson/catkin_ws/src/realsense/realsense_camera/test/camera_core.cpp:644:41: error: ‘RS_STREAM_FISHEYE’ was not declared in this scope
     EXPECT_EQ(g_caminfo_projection_recv[RS_STREAM_FISHEYE][9], 0.0);
                                         ^
/home/jackson/catkin_ws/src/realsense/realsense_camera/test/camera_core.cpp:644:5: error: template argument 1 is invalid
     EXPECT_EQ(g_caminfo_projection_recv[RS_STREAM_FISHEYE][9], 0.0);
     ^
In file included from /usr/include/gtest/gtest.h:57:0,
                 from /home/jackson/catkin_ws/src/realsense/realsense_camera/test/camera_core.cpp:30:
/home/jackson/catkin_ws/src/realsense/realsense_camera/test/camera_core.cpp:645:43: error: ‘RS_STREAM_FISHEYE’ was not declared in this scope
     EXPECT_TRUE(g_caminfo_projection_recv[RS_STREAM_FISHEYE][10] != 0.0);
                                           ^
In file included from /usr/include/gtest/gtest.h:1807:0,
                 from /home/jackson/catkin_ws/src/realsense/realsense_camera/test/camera_core.cpp:30:
/home/jackson/catkin_ws/src/realsense/realsense_camera/test/camera_core.cpp:646:41: error: ‘RS_STREAM_FISHEYE’ was not declared in this scope
     EXPECT_EQ(g_caminfo_projection_recv[RS_STREAM_FISHEYE][11], 0.0);
                                         ^
/home/jackson/catkin_ws/src/realsense/realsense_camera/test/camera_core.cpp:646:5: error: template argument 1 is invalid
     EXPECT_EQ(g_caminfo_projection_recv[RS_STREAM_FISHEYE][11], 0.0);
     ^
make[2]: *** [realsense/realsense_camera/CMakeFiles/tests_camera_core.dir/test/camera_core.cpp.o] Error 1
make[1]: *** [realsense/realsense_camera/CMakeFiles/tests_camera_core.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
```

jacksonkr, 2016-10-21:
What's most odd to me is that this works fine on kinetic. I just installed it again on an ubuntu 16.04 amd64.

I would use kinetic but unfortunately ubuntu 16.04 is not playing nicely with my armhf machine.

mdhorn, 2016-10-21:
@jacksonkr  I still think you have the wrong versions of the sources checked out -- the ZR300 code was added after 1.4.0.

Let's take a step back. Do you need to build from sources? Do the prebuilt Debian packages from the ROS repository for ARM not work?
sudo apt-get install ros-indigo-realsense-camera

Since you are on Indigo, you would still need to still need to complete the steps for installing the udev rules and patching the kernel.

See https://github.com/intel-ros/realsense/tree/indigo-devel/realsense_camera#installing-the-ros-debian-package
> librealsense 0.9.2 and earlier require a few extra, manual steps to enable the use of the library.
> 
> Clone the librealsense GitHub repository:
> `git clone https://github.com/IntelRealSense/librealsense.git`
> `cd librealsense`
> `git checkout -b b0.9.2 v0.9.2`
> Follow the instructions for [Video4Linux backend](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#video4linux-backend) **ONLY**. Only one of the three options for Step 3 should be completed

The there are some issues with the ARM ROS Debian packages, it would be good to know -- but we currently don't have platforms for validating those.

If you do need to build from sources, ensure you are using valid combinations of librealsense and realsense_camera. librealsense has hardcoded version checks, so you have to use a valid combo; like librealsense v0.9.2 and realsense_camera 1.4.0.

But all of those error message appears as if you don't have the librealsense header files in the include path. Check that you sourced your <workspace>/dev/setup.bash after you build the librealsense and before you build realsense_camera.

jacksonkr, 2016-10-21:
same error on ubuntu 14.04 amd64, this error on ubuntu 14.04 armhf

$ catkin_make
Base path: /home/ubuntu/catkin_ws
Source space: /home/ubuntu/catkin_ws/src
Build space: /home/ubuntu/catkin_ws/build
Devel space: /home/ubuntu/catkin_ws/devel
Install space: /home/ubuntu/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/ubuntu/catkin_ws/build"
####
####
#### Running command: "make -j1 -l1" in "/home/ubuntu/catkin_ws/build"
####
[  3%] Built target realsense_camera_gencfg
[  3%] Built target _realsense_camera_generate_messages_check_deps_IMUInfo
[  3%] Built target _realsense_camera_generate_messages_check_deps_ForcePower
[  3%] Built target _realsense_camera_generate_messages_check_deps_CameraConfiguration
[  3%] Built target std_msgs_generate_messages_lisp
[  3%] Built target _realsense_camera_generate_messages_check_deps_IsPowered
[  3%] Built target _realsense_camera_generate_messages_check_deps_SetPower
[  3%] Built target _realsense_camera_generate_messages_check_deps_GetIMUInfo
[  8%] Built target realsense_camera_generate_messages_lisp
[  8%] Built target std_msgs_generate_messages_py
[ 13%] Built target realsense_camera_generate_messages_py
[ 13%] Built target std_msgs_generate_messages_cpp
[ 17%] Built target realsense_camera_generate_messages_cpp
[ 17%] Built target realsense_camera_generate_messages
[ 17%] Built target topic_tools_generate_messages_py
[ 17%] Built target nodelet_generate_messages_py
[ 17%] Built target dynamic_reconfigure_generate_messages_cpp
[ 17%] Built target dynamic_reconfigure_generate_messages_lisp
[ 17%] Built target dynamic_reconfigure_generate_messages_py
[ 17%] Built target nodelet_generate_messages_cpp
[ 17%] Built target dynamic_reconfigure_gencfg
[ 17%] Built target roscpp_generate_messages_lisp
[ 17%] Built target tf_generate_messages_cpp
[ 17%] Built target nodelet_generate_messages_lisp
[ 17%] Built target sensor_msgs_generate_messages_py
[ 17%] Built target rosgraph_msgs_generate_messages_cpp
[ 17%] Built target roscpp_generate_messages_cpp
[ 17%] Built target roscpp_generate_messages_py
[ 17%] Built target sensor_msgs_generate_messages_lisp
[ 17%] Built target rosgraph_msgs_generate_messages_lisp
[ 17%] Built target rosgraph_msgs_generate_messages_py
[ 17%] Built target bond_generate_messages_cpp
[ 17%] Built target bond_generate_messages_lisp
[ 17%] Built target bond_generate_messages_py
[ 17%] Built target sensor_msgs_generate_messages_cpp
[ 17%] Built target tf_generate_messages_lisp
[ 17%] Built target tf_generate_messages_py
[ 17%] Built target pcl_ros_gencfg
[ 17%] Built target actionlib_msgs_generate_messages_cpp
[ 17%] Built target geometry_msgs_generate_messages_cpp
[ 17%] Built target geometry_msgs_generate_messages_lisp
[ 17%] Built target geometry_msgs_generate_messages_py
[ 17%] Built target topic_tools_generate_messages_lisp
[ 17%] Built target pcl_msgs_generate_messages_lisp
[ 17%] Built target topic_tools_generate_messages_cpp
[ 17%] Built target tf2_msgs_generate_messages_cpp
[ 17%] Built target pcl_msgs_generate_messages_py
[ 17%] Built target actionlib_generate_messages_lisp
[ 17%] Built target actionlib_generate_messages_cpp
[ 17%] Built target actionlib_generate_messages_py
[ 17%] Built target tf2_msgs_generate_messages_lisp
[ 17%] Built target actionlib_msgs_generate_messages_lisp
[ 17%] Built target actionlib_msgs_generate_messages_py
[ 17%] Built target tf2_msgs_generate_messages_py
[ 17%] Built target pcl_msgs_generate_messages_cpp
[ 18%] Building CXX object realsense/realsense_camera/CMakeFiles/realsense_camera_nodelet.dir/src/base_nodelet.cpp.o
[... 6 lines cut ...]
                                                                                    ^
/home/ubuntu/catkin_ws/src/realsense/realsense_camera/src/base_nodelet.cpp:243:74: error: 'RS_CAMERA_INFO_ADAPTER_BOARD_FIRMWARE_VERSION' was not declared in this scope
         const char * adapter_fw = rs_get_device_info(rs_detected_device, RS_CAMERA_INFO_ADAPTER_BOARD_FIRMWARE_VERSION,
                                                                          ^
/home/ubuntu/catkin_ws/src/realsense/realsense_camera/src/base_nodelet.cpp:244:23: error: 'rs_get_device_info' was not declared in this scope
             &rs_error_);
                       ^
/home/ubuntu/catkin_ws/src/realsense/realsense_camera/src/base_nodelet.cpp:249:43: error: 'RS_CAPABILITIES_MOTION_EVENTS' was not declared in this scope
       if (rs_supports(rs_detected_device, RS_CAPABILITIES_MOTION_EVENTS, &rs_error_))
                                           ^
/home/ubuntu/catkin_ws/src/realsense/realsense_camera/src/base_nodelet.cpp:249:84: error: 'rs_supports' was not declared in this scope
       if (rs_supports(rs_detected_device, RS_CAPABILITIES_MOTION_EVENTS, &rs_error_))
                                                                                    ^
/home/ubuntu/catkin_ws/src/realsense/realsense_camera/src/base_nodelet.cpp:251:80: error: 'RS_CAMERA_INFO_MOTION_MODULE_FIRMWARE_VERSION' was not declared in this scope
         const char * motion_module_fw = rs_get_device_info(rs_detected_device, RS_CAMERA_INFO_MOTION_MODULE_FIRMWARE_VERSION,
                                                                                ^
/home/ubuntu/catkin_ws/src/realsense/realsense_camera/src/base_nodelet.cpp:252:23: error: 'rs_get_device_info' was not declared in this scope
             &rs_error_);
                       ^
[... cut ...]
make[1]: *** [realsense/realsense_camera/CMakeFiles/realsense_camera_nodelet.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j1 -l1" failed

Is there a chance I'm getting this similar error because all the installs are all muddled together? Is there an easy way to wipe out any trace of *realsense and start over? I'm not having this problem at all with kinetic, but I am having in here. If you're convinced it's an install issue then I'll grind on that until I can come up with new information.

jacksonkr, 2016-10-22:
@mdhorn I appreciate your patience during all of this. I decided this was likely a user error so I doubled down to track it down. Here's the documentation for anyone else who find themselves here (note, this is dodcument elsewhere on github but through human error I didn't get it correctly):

librealsense at version 1.4.0 is what was missing. I was able to track down my current version with 

```
$ git describe --tags
1.5.0
```

Here's what needs to be done from the ground up. Works on both Ubuntu 14.04 amd64 and armhf
```
$ cd ~/github # where I keep my standalone github checkouts
$ git clone https://github.com/IntelRealSense/librealsense
$ cd librealsense
$ git checkout -b b1.4.0 1.4.0 # what I was missing before. git checkout -b [local branch name] [repo branch name]
$ mkdir build
$ cmake .. -DBUILD_EXAMPLES=on
$ make && sudo make install

$ cd ~/catkin_ws/src/ # where I keep my ros github checkouts
$ git clone https://github.com/intel-ros/realsense
$ cd realsense
$ git checkout -b b0.9.2 v0.9.2
$ cd ~/catkin_ws
$ catkin_make
```

mdhorn, 2016-10-24:
@jacksonkr 
Glad you got it resolved. Did you ever try just using the prebuilt ROS Debian packages under ARM? I know they work for Indigo for X86
`sudo apt-get install ros-indigo-realsense-camera`

FYI: I made an update to your previous comment. You had the tag swapped; v0.9.2 is for librealsense, and 1.4.0 is for realsense camera.
Also BUILD_EXAMPLES cmake flag for librealsense was not added until version 1.11.0 so it has no impact on the v0.9.2 release.

Cheers.

jacksonkr, 2016-10-26:
@mdhorn Good eye on the version numbers.

I'm trying to install again with only the apt-get binary but I'm running into trouble. see #139.

jacksonkr, 2016-10-26:
I switched to a different machine which is `4.4.0-46-generic` kernel so I could try with only debian packages.

After completely removing EVERY trace of *realsense* (`sudo find / -name *realsense*` then manual remove of each listing) I retried my realsense-dependent app installation (rtabmap) with strictly the debian packages (ros-indigo-librealsense, ros-indigo-realsense-camera).

- I did a `cmake ..` for my realsense-dependent app and realsense support was not enabled.
- `$ sudo apt-get install ros-indigo-librealsense`
- `cmake ..` shows librealsense support
- `make` works as expected for realsense-dependent app

```
$ rtabmap
rtabmap: error while loading shared libraries: librealsense.so: cannot open shared object file: No such file or directory
```

I ran `sudo apt-get install ros-indigo-realsense-camera` and then recompiled my realsense-dependent app (removed all build files then `cmake .. && make && make install`)

Same error when running rtabmap so I ran 
```
$ sudo find / -name librealsense*
/opt/ros/indigo/include/librealsense
/opt/ros/indigo/share/librealsense
/opt/ros/indigo/share/librealsense/cmake/librealsenseConfig.cmake
/opt/ros/indigo/share/librealsense/cmake/librealsenseConfig-version.cmake
/opt/ros/indigo/lib/pkgconfig/librealsense.pc
/opt/ros/indigo/lib/librealsense_camera_nodelet.so
/opt/ros/indigo/lib/x86_64-linux-gnu/librealsense.so
```

And you can see that the `librealsense.so` is in an obscure location so to test this out, I ran the following:

`export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/opt/ros/indigo/lib/x86_64-linux-gnu/`

And then rtabmap could find `librealsense.so`.

As far as I can tell, this issue falls on librealsense because the `librealsense.so` is not in a generic enough location. What do you think?

mdhorn, 2016-10-26:
@jacksonkr You should not be setting your LD_LIBRARY_PATH variable to include ROS, that is the job of the ROS setup.bash script.

Installing ROS is not enough, you need to source the ROS environment to use it:
http://wiki.ros.org/indigo/Installation/Ubuntu#indigo.2BAC8-Installation.2BAC8-DebEnvironment.Environment_setup

If you are install ros-indigo-librealsense and ros-indigo-realsense-camera, I'm not sure what you are attempting to build with 'cmake'?

jacksonkr, 2016-10-27:
@mdhorn

I always keep two ros sourcings in my `~/.bashrc`:
```
source /opt/ros/[distrobution]/setup.bash
source ~/catkin_ws/devel/setup.bash
```

Note that this issue started out with me making both librealsense v0.9.2 and realsense 1.4.0 manually. During the length of this issue I moved to the debian packages per your suggestion, which is where the `LD_LIBRARY_PATH` portion comes in, because the path was not including the `librealsense.so` from the debian packages.

So the fact that my `LD_LIBRARY_PATH` was not set correctly for this build, is that a mistep of ROS? Do I need to move this issue there?

mdhorn, 2016-10-28:
I would verify that the location where the librealsense.so is installed under /opt is also in your LD_LIBRARY_PATH; if not, re-source the setup.bash.

For example I checked and initially the location was missing:
```
$ find /opt -name \*librealsense\* -print
/opt/ros/kinetic/share/librealsense
/opt/ros/kinetic/share/librealsense/cmake/librealsenseConfig-version.cmake
/opt/ros/kinetic/share/librealsense/cmake/librealsenseConfig.cmake
/opt/ros/kinetic/lib/pkgconfig/librealsense.pc
/opt/ros/kinetic/lib/librealsense_camera_nodelet.so
/opt/ros/kinetic/lib/x86_64-linux-gnu/librealsense.so
/opt/ros/kinetic/include/librealsense

$ echo $LD_LIBRARY_PATH 
/home/mhorn/ROS/devel/lib:/opt/ros/kinetic/lib:/usr/local/lib
```
Notice that the location of the librealsense is NOT in my LD_LIBRARY_PATH. I believe this is because I installed the package after I had source the /opt/ros/kinetic/setup.bash.

Re-sourcing the setup.bash fixes the problem:
```
$ source /opt/ros/kinetic/setup.bash 
$ echo $LD_LIBRARY_PATH 
/opt/ros/kinetic/lib:/opt/ros/kinetic/lib/x86_64-linux-gnu:/usr/local/lib
```

Even if I source my development area afterwards, the correct path is maintained:
```
$ source ~/ROS/devel/setup.bash 
$ echo $LD_LIBRARY_PATH 
/home/mhorn/ROS/devel/lib:/opt/ros/kinetic/lib:/opt/ros/kinetic/lib/x86_64-linux-gnu:/usr/local/lib
```

jacksonkr, 2016-10-28:
Hmm, I know that sourcing and resourcing the setup can resolve issues so I'll have to take your word for it since I'm beyond that point. If I run into this again I'll check back with the results.

Thanks @mdhorn

closed 2016-10-28 by jacksonkr