realsenseai/realsense-ros #129: adding simple demo for april tags into the launch files

SteveMacenski, 2016-10-13:
Hi!

I've been working with april tags with the real sense camera and figured I would share my launch file for getting the real sense camera up and running with the ros wrapper of April Tags for pose estimation utilizing the camera installed by `sudo apt-get install ros-indigo-apriltags-ros`

Feel free to include it in future examples. 

https://github.com/SteveMacenski/2_ABB_irb120_F-T-Visual-Feedback/blob/master/realsenseCamera_aprilTags/launch/realsense_apriltags.launch

kevincwells, 2016-11-01:
Hey Steve, that's great to hear -- glad you were able to get it working with R200.  This is the kind of info that would be great to have on the ROS Wiki.  If you follow the instructions for setting up an AprilTags Tutorials index on http://wiki.ros.org/apriltags_ros/Tutorials, you could then add a tutorials page called "Realsense Camera Integration" (or something similar) to document your results.

closed 2016-11-11 by kevincwells

willchamberlain, 2016-11-28:
Thanks Steve: very useful.

mdhorn, 2016-11-29:
@SteveMacenski You could also put an AprilTags Tutorial under the [realsense_camera Tutorials](http://wiki.ros.org/realsense_camera/Tutorials).