realsenseai/realsense-ros #130: Occasionally can't restart SR300 with /camera/driver/force_power service

mattcorsaro1, 2016-10-17:
| Version               | Your Configuration |
|:--------------------- |:------------ |
| Operating System      |Ubuntu 14.04.5 LTS |
| Kernel                | 4.4.0-38-generic |
| Backend               | video4linux |
| ROS                   | indigo |
| ROS RealSense         | 1.5.0 |
| librealsense          | 0.9.1 |
| Camera Type-Firmware  | `ToDo` |

---

### Expected Behavior
Launch `sr300_nodelet_rgbd.launch` and another node from a launch file. In the other node (Python), initialize some variables, wait for three seconds, then force camera power off with
```
from realsense_camera import srv
camera_forcepower_service = rospy.ServiceProxy('/camera/driver/force_power', srv.ForcePower)
camera_forcepower_service(False)
```
Throughout the script, turn the power back on with camera_forcepower_service(True), then off with camera_forcepower_service(False) after reading a cloud.

### Actual Behavior
Most of the time, it works as expected. Once in a while, I won't be able to restart the camera within the middle of my script, even after it's been started and stopped successfully. I'll get an error like this:
```
[ INFO] [1476720876.616341805]: /camera/driver - Starting camera
[ERROR] [1476720876.621235305]: /camera/driver - Error calling rs_start_device ( device:0x2614538 ): 
VIDIOC_S_PARM error 5, Input/output error 

terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[camera/camera_nodelet_manager-1] process has died [pid 11388, exit code -6, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=camera_nodelet_manager __log:=/home/ur5/.ros/log/b1689f24-9482-11e6-b0d0
-4ccc6a230519/camera-camera_nodelet_manager-1.log].
log file: /home/ur5/.ros/log/b1689f24-9482-11e6-b0d0-4ccc6a230519/camera-camera_nodelet_manager-1*.log
[camera/driver-2] process has finished cleanly
log file: /home/ur5/.ros/log/b1689f24-9482-11e6-b0d0-4ccc6a230519/camera-driver-2*.log

```
I then have to reboot the computer to access the camera again. Before rebooting, lsusb doesn't list the camera as connected, even when it's plugged in.

### Steps to Reproduce

Launch camera nodelet with `roslaunch realsense_camera sr300_nodelet_rgbd.launch`
Shutdown in another node with the following Python code:
```
from realsense_camera import srv
camera_forcepower_service = rospy.ServiceProxy('/camera/driver/force_power', srv.ForcePower)
camera_forcepower_service(False)
```
Later in the same Python script, start the camera again with `camera_forcepower_service(True)`
Shut the camera off again after reading an image.

mattcorsaro1, 2016-10-17:
Another error:
```
[ INFO] [1476724194.379703393]: /camera/driver - Starting camera
[ INFO] [1476724194.394910006]: /camera/driver - Camera Started Succesfully
[ERROR] [1476724199.395296647]: /camera/driver - Error calling rs_wait_for_frames ( device:0x1f97858 ): 
Timeout waiting for frames. 

terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[camera/camera_nodelet_manager-1] process has died [pid 8305, exit code -6, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=camera_nodelet_manager __log:=/home/ur5/.ros/log/d7b130f8-948a-11e6-9915-4ccc6a230519/camera-camera_nodelet_manager-1.log].
log file: /home/ur5/.ros/log/d7b130f8-948a-11e6-9915-4ccc6a230519/camera-camera_nodelet_manager-1*.log
[camera/driver-2] process has finished cleanly
log file: /home/ur5/.ros/log/d7b130f8-948a-11e6-9915-4ccc6a230519/camera-driver-2*.log
```

rjingar, 2016-10-18:
Hi @mattcorsaro1 

I tried to start/stop the camera multiple times using your script and it always works for me. I unable to reproduce the issue. How frequently you are seeing the issue? Also, are you seeing the issue regardless of whether there is a subscriber or not?

mattcorsaro1, 2016-10-20:
It happens about 5% of the time. Here's my script to recreate the issue:

camera_test.launch:
```
<launch>
    <include file="$(find realsense_camera)/launch/sr300_nodelet_rgbd.launch"/>

    <group>
        <node name="camera_test" pkg="pkg_name" type="camera_test.py" output="screen">
        </node>
    </group>

</launch>
```

camera_test.py:
```
#!/usr/bin/env python

'''
Turn the camera on and off repeatedly
'''

import rospy
from realsense_camera import srv

def main():
  camera_forcepower_service = rospy.ServiceProxy('/camera/driver/force_power', srv.ForcePower)
  #camera_ispowered_service = rospy.ServiceProxy('/camera/driver/is_powered', srv.IsPowered)

  rospy.sleep(3.0)
  camera_forcepower_service(False)

  while True:
    camera_forcepower_service(True)
    rospy.sleep(0.01)
    camera_forcepower_service(False)

if __name__ == '__main__':
  main()
```

And here's the output:

```
ur5@UR5:~/ur5_ws$ roslaunch ur_grasping camera_test.launch 
... logging to /home/ur5/.ros/log/8c6d86ce-96e6-11e6-96fb-4ccc6a230519/roslaunch-UR5-28089.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://UR5:36803/

SUMMARY
========

PARAMETERS
 * /camera/camera_nodelet_manager/num_worker_threads: 4
 * /camera/depth_rectify_depth/interpolation: 0
 * /camera/disparity_depth/max_range: 4.0
 * /camera/disparity_depth/min_range: 0.5
 * /camera/disparity_registered_sw/max_range: 4.0
 * /camera/disparity_registered_sw/min_range: 0.5
 * /camera/driver/base_frame_id: camera_link
 * /camera/driver/camera_type: SR300
 * /camera/driver/color_fps: 30
 * /camera/driver/color_frame_id: camera_rgb_frame
 * /camera/driver/color_height: 480
 * /camera/driver/color_optical_frame_id: camera_rgb_optica...
 * /camera/driver/color_width: 640
[... 151 lines cut ...]
[ INFO] [1476987946.133510992]: /camera/driver - Camera Stopped Succesfully
[ INFO] [1476987946.134155557]: /camera/driver - Camera is already Stopped
[ INFO] [1476987946.135913951]: /camera/driver - Starting camera
[ INFO] [1476987946.154492640]: /camera/driver - Camera Started Succesfully
[ERROR] [1476987951.154716202]: /camera/driver - Error calling rs_wait_for_frames ( device:0xc2b478 ): 
Timeout waiting for frames. 

Traceback (most recent call last):
  File "/home/ur5/ur5_ws/src/ur_grasping/scripts/camera_test.py", line 23, in <module>
    main()
  File "/home/ur5/ur5_ws/src/ur_grasping/scripts/camera_test.py", line 20, in main
    camera_forcepower_service(False)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 495, in call
    service_uri = self._get_service_uri(request)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 463, in _get_service_uri
    raise ServiceException("service [%s] unavailable"%self.resolved_name)
rospy.service.ServiceException: service [/camera/driver/force_power] unavailable
[camera/camera_nodelet_manager-1] process has died [pid 28107, exit code 1, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=camera_nodelet_manager __log:=/home/ur5/.ros/log/8c6d86ce-96e6-11e6-96fb-
4ccc6a230519/camera-camera_nodelet_manager-1.log].
log file: /home/ur5/.ros/log/8c6d86ce-96e6-11e6-96fb-4ccc6a230519/camera-camera_nodelet_manager-1*.log
[camera_test-16] process has died [pid 28231, exit code 1, cmd /home/ur5/ur5_ws/src/ur_grasping/scripts/camera_test.py __name:=camera_test __log:=/home/ur5/.ros/log/8c6d86ce-96e6-11e6-96fb-4ccc6a230519/cam
era_test-16.log].
```

mdhorn, 2016-10-21:
Starting the camera and then only  having a 1/10 of second before turning it off, and then no wait before turning it back on again doesn't seem like a reasonable test case.

Can you repeat this problem if you have a more reasonable use-case sleep times between the starts and stops?

`  while True:`
`    rospy.sleep(10.0)`
`    camera_forcepower_service(True)`
`    rospy.sleep(10.0)`
`    camera_forcepower_service(False)`

mattcorsaro1, 2016-10-26:
It does work if I add a bigger delay.

closed 2016-10-26 by mattcorsaro1